wolfSSL/wolfssl #753: build error:  wolfcrypt/src/tfm.c:322:14: error: array subscript is below array bounds [-Werror=array-bounds]

muelli, 2017-02-16:
```
✗>make V=1
make -j5  all-am
make[1]: Entering directory '/home/muelli/vcs/wolfssl'
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.  -DBUILDING_WOLFSSL -fvisibility=hidden -fvisibility=hidden -DBUILDING_WOLFSSL -D_POSIX_THREADS  -DHAVE_THREAD_LS -DNDEBUG -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DHAVE_AESGCM -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DHAVE_CURVE25519 -DHAVE_ED25519 -DWOLFSSL_BASE64_ENCODE -DNO_RC4 -DNO_HC128 -DNO_RABBIT -DWOLFSSL_SHA224 -DHAVE_POLY1305 -DHAVE_ONE_TIME_AUTH -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_EXTENDED_MASTER -DNO_PSK -DNO_MD4 -DNO_PWDBASED -DUSE_FAST_MATH -DWOLFSSL_X86_64_BUILD -DNO_DES3 -Wall -Wno-unused -O2 -fomit-frame-pointer -DHAVE___UINT128_T -O3 -g0  -Werror -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv  -MT wolfcrypt/src/src_libwolfssl_la-tfm.lo -MD -MP -MF wolfcrypt/src/.deps/src_libwolfssl_la-tfm.Tpo -c -o wolfcrypt/src/src_libwolfssl_la-tfm.lo `test -f 'wolfcrypt/src/tfm.c' || echo './'`wolfcrypt/src/tfm.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -DBUILDING_WOLFSSL -fvisibility=hidden -fvisibility=hidden -DBUILDING_WOLFSSL -D_POSIX_THREADS -DHAVE_THREAD_LS -DNDEBUG -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DHAVE_AESGCM -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DHAVE_CURVE25519 -DHAVE_ED25519 -DWOLFSSL_BASE64_ENCODE -DNO_RC4 -DNO_HC128 -DNO_RABBIT -DWOLFSSL_SHA224 -DHAVE_POLY1305 -DHAVE_ONE_TIME_AUTH -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_EXTENDED_MASTER -DNO_PSK -DNO_MD4 -DNO_PWDBASED -DUSE_FAST_MATH -DWOLFSSL_X86_64_BUILD -DNO_DES3 -Wall -Wno-unused -O2 -fomit-frame-pointer -DHAVE___UINT128_T -O3 -g0 -Werror -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -MT wolfcrypt/src/src_libwolfssl_la-tfm.lo -MD -MP -MF wolfcrypt/src/.deps/src_libwolfssl_la-tfm.Tpo -c wolfcrypt/src/tfm.c  -fPIC -DPIC -o wolfcrypt/src/.libs/src_libwolfssl_la-tfm.o
wolfcrypt/src/tfm.c: In function ‘fp_mulmod’:
wolfcrypt/src/tfm.c:322:14: error: array subscript is below array bounds [-Werror=array-bounds]
         C->dp[y] = 0;
              ^
wolfcrypt/src/tfm.c: In function ‘fp_sqrmod’:
wolfcrypt/src/tfm.c:1514:10: error: array subscript is below array bounds [-Werror=array-bounds]
     B->dp[y] = 0;
          ^
cc1: all warnings being treated as errors
Makefile:3129: recipe for target 'wolfcrypt/src/src_libwolfssl_la-tfm.lo' failed
make[1]: *** [wolfcrypt/src/src_libwolfssl_la-tfm.lo] Error 1
make[1]: Leaving directory '/home/muelli/vcs/wolfssl'
Makefile:1995: recipe for target 'all' failed
make: *** [all] Error 2
```

kaleb-himes, 2017-03-01:
Thanks for the report muelli. We will be looking into this as soon as possible!

toddouska, 2017-03-01:
Which version of gcc are you using?  That should help with reproducing and a fix.

muelli, 2017-03-02:
```
>gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

>apt-cache policy gcc
gcc:
  Installed: 4:5.3.1-1ubuntu1
  Candidate: 4:5.3.1-1ubuntu1
  Version table:
 *** 4:5.3.1-1ubuntu1 500
        500 http://de.archive.ubuntu.com/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status

```

dgarske, 2017-11-03:
Hi Muelli,

This was fixed on July 21, 2017 in commit:
https://github.com/wolfSSL/wolfssl/pull/1049/commits/26ac5e1ab7a710b8194c78e7d658cf0cd87dfab8

This is included since release v3.12.0-stable.

Thanks,
David Garske, wolfSSL

closed 2017-11-03 by dgarske