wolfSSL/wolfssl #779: erros when enabling session ticket on a server build only without old TLS support

hajjihraf, 2017-03-06:
Hi,

I encountered a build errors on server only build when using TLS.

attached the user_settings.h I use in my testing.

Thanks,
Achraf

[user_settings.zip](https://github.com/wolfSSL/wolfssl/files/820890/user_settings.zip)

kaleb-himes, 2017-03-08:
Hi hajjihraf,

What build environment are you working with? What compiler/toolchain, how is wolfSSL being compiled and linked (statically or as a shared library)?

Please provide much more detail that we might assist you!

- Kaleb

hajjihraf, 2017-03-15:
Hi Kaleb,

I'm building a static server (not a DLL) on windows using MinGW toolchain.
The problems come from:
1- struct WOLFSSL: the member "expect_session_ticket" is defined when wolfSSL client support is enabled and Session ticket is enabled
2- in internal.c: the function c32toa is not defined when HAVE_TLS_EXTENSIONS is not defined.

Regards,
Achraf

JacobBarthelmeh, 2017-04-07:
Hello Achraf,

After pull request 842 is merged the issue should be resolved.  Looking through the history of issues reported I see there was a couple in the past and want to thank you again for opening these. I'm curious if you would be able to provide us with some information about the project being worked on?

Regards,
Jacob

hajjihraf, 2017-04-07:
Hi Jacob,

Thanks for your response.
All I'm authorized to Tell is that I'm using wolfSSL for an embedded software project and the company  I'm working for has already  bougth the wolfSSL commercial license. 

Thanks Again,
Achraf

JacobBarthelmeh, 2017-04-11:
Hi Achraf,

Thank you for the feedback, I think this is resolved now and am going to go ahead and close the issue. As a side note we also have a support email channel at support@wolfssl.com that can provide faster turn around times for technical questions and build issues.

Regards,
Jacob

closed 2017-04-11 by JacobBarthelmeh