wolfSSL/wolfssl #811: API/ABI changes review for wolfSSL

lvc, 2017-03-21:
The review of API/ABI changes for wolfSSL since 2.0.2 version: https://abi-laboratory.pro/tracker/timeline/wolfssl/

The report is updated 3 times a week (RSS: https://abi-laboratory.pro/tracker/rss/wolfssl/feed.rss). Hope it will be helpful for users and maintainers of the library.

Created with the help of open-source abi-tracker tool: https://github.com/lvc/abi-tracker

The tool checks _all_ API symbols declared in header files (doesn't take docs into account), so there may be some false positives.

Thank you.

[image]

[image]

## Environment
* Linux x86_64
* GCC 4.9
* wolfSSL 2.0.2 and higher

cconlon, 2017-03-23:
Hi!

Thanks for letting us know wolfSSL is being tracked on the abi-laboratory.pro site, and for setting it up!  We use the abi-tracker tool internally during our release cycle, but haven't had time to automate the process yet.  This will be very helpful!

When looking over the generated report (https://abi-laboratory.pro/tracker/timeline/wolfssl/), it looks like several of our GitHub releases/tags are skipped.  For example between the listed "3.9.3" and "3.10.3" tags, all of the following are missing from the report:

v3.9.10-stable
v3.9.10b
v3.10.0-stable
v3.10.0a
v3.10.2-stable

From the pattern, it seems that all tags ending in non-decimal numbers might currently be skipped?

As a thanks for getting this set up for us, I'd like to send some wolfSSL swag your way.  Can you email me your mailing address so we can ship a few things out to you?  You can reach me at chris (at) wolfssl (dot) com.

Thanks again!
Chris

lvc, 2017-03-25:
Hi,

The report has been fixed: https://abi-laboratory.pro/tracker/timeline/wolfssl/

Thank you.

[image]

cconlon, 2017-03-27:
Thanks for the fixes!

closed 2017-03-27 by cconlon

dgarske, 2021-01-20:
@lvc, I am not sure if you are the right contact still, but the ABI tracker for wolfSSL does not have our Dec 23, 2020 release v4.6.0 listed. Is there a way we can fix that? Thanks, David Garske, wolfSSL

linuxhw, 2021-01-21:
Fixed: https://abi-laboratory.pro/?view=timeline&l=wolfssl

Thank you.