wolfSSL/wolfssl #849: Incompatable with crypto_stream_chacha20_ietf_xor_ic funtion from libsodium

zhou0, 2017-04-12:
Chacha20 of wolfssl use 96 bits nonce, so is libsodium. But when i use wolfssl to decrypt stream encrypted by crypto_stream_chacha20_ietf_xor_ic, i only get the correct plain text with stream that are very shot. with long stream, i get most of the stream chunks decrypted wrong and only a few chunks correct. 

to demo the problem:

1. pull the mysocks project ( which use wolfssl for chacha20 )from  https://github.com/zhou0/mysocks.git , it a cmake based project, build the project and run the following command

ssclient-wolfssl -k fuckgfw -m chacha20-ietf -p 2194 -s localhost -l 1895

2. pull the shadowsocks  (which uses libsodium for chacha20 )from https://github.com/shadowsocks/shadowsocks-libev.git or download a release later than v2.5.0 https://github.com/shadowsocks/shadowsocks-libev/archive/v2.5.0.tar.gz , build the project and run the following command in a second terminal

./ss-server -k fuckgfw -m chacha20-ietf -p 2194 -s localhost -l 1895 -v

3. now in a 3rd terminal run the following command , you will observer the problem.

curl -v -x socks5h://localhost:1895 cn.bing.com

if you change the url to bing.com, such as

curl -v -x socks5h://localhost:1895 bing.com

which is correct because the HTTP reply is just a short HTTP redirection message.

JacobBarthelmeh, 2017-04-13:
Hi zhou0,

I looked at the version of wolfSSL used with mysocks and it appears to be old. Does this happen also with using a more recent version of wolfSSL and are you seeing the problems persist when connecting libraries other than libsodium? There has been RFC draft updates to how ChaCha20-Poly1305 is used with TLS connections since wolfSSL version 3.2.0.

One thing that can be done for interoperability testing with ChaCha20-Poly1305 suites is the following.

Start OpenSSL Server
./apps/openssl s_server -accept 11111 -cert ~/wolfssl/certs/server-cert.pem -key ~/wolfssl/certs/server-key.pem

Connect to Server with wolfSSL
./examples/client/client -l ECDHE-RSA-CHACHA20-POLY1305 -i

zhou0, 2017-04-14:
This is what i get running these 2 commands:

 ./apps/openssl s_server -accept 11111 -cert ../wolfssl-3.2.0/certs/server-cert.pem -key ../wolfssl-3.2.0/certs/server-key.pem
Using default temp DH parameters
ACCEPT
ERROR
34389124096:error:1417A0C1:SSL routines:tls_post_process_client_hello:no shared cipher:ssl/statem/statem_srvr.c:1422:
shutting down SSL
CONNECTION CLOSED
ACCEPT

./client -l ECDHE-RSA-CHACHA20-POLY1305 
err = -313, revcd alert fatal error
yassl error: SSL_connect failed

so chacha20 of wolfssl-3.2.0 does not work with chacha20 of openssl-1.1.0.   As far as i know, chacha20 of openssl uses 64 bits nonce, so this result is expected.

zhou0, 2017-04-14:
by the way, i have run the test for all wolfssl versions from 3.2.0 up to the latest. All tests fail. But from version 3.9.0 up, the error message changes to:

err = -151, ASN date error, current date after
wolfSSL error: wolfSSL_connect failed

zhou0, 2017-04-14:
I have fixed this in mysocks and proved that wolfssl chacha20  is compatible with libsodium.  It is not compatible with openssl howerver.

closed 2017-04-14 by zhou0