wolfSSL/wolfssl #859: Compiling for Arduino fails

pasko-zh, 2017-04-13:
Hello wolfSSL!

I am trying to compile a simple sketch with wolfcrypt (SHA 256 HMAC) for an esp8266 with the Arduino tool chain. I followed the steps [here](https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO) and read the thread [here](https://www.wolfssl.com/forums/topic861-solved-getting-started-wolfcrypt-on-arduino.html).

I am using the following versions

- wolfSSL, 3.10.2-stable
- Arduino 1.8.2
- Arduino esp core 2.3.0

When I am working with Visual Studio (together with the visual micro plugin), the following code 
```c 
// This sketch is saved as wolfhmactest.ino

#include <wolfssl.h>
#include <wolfcrypt\sha256.h>
#include <wolfcrypt\hmac.h>

void hmac_sha256_test() {
	Hmac hmac_256;
	int ret;
	// "Jefe", or { 74, 101, 102, 101 }
	const char *key_1[] = { "Jefe" };
	ret = wc_HmacSetKey(&hmac_256, 2, (byte *)key_1, (word32) 4);
}

void setup() {
	Serial.begin(115200);
	delay(1000);
}

void loop() {
	hmac_sha256_test();
	delay(5000);
}

```  
compiles with the following errors:

````
Compiling 'wolfhmactest' for 'Adafruit HUZZAH ESP8266'
wolfhmactest.cpp.o*: (.text._Z16hmac_sha256_testv+0x4): undefined reference to wc_HmacSetKey

Error linking for board Adafruit HUZZAH ESP8266
Build failed for project 'wolfhmactest'
 
wolfhmactest.cpp.o*: In function hmac_sha256_test()
wolfhmactest.ino:11: undefined reference to wc_HmacSetKey
 
collect2.exe*: error: ld returned 1 exit status
````

When I try to compile with the (native) Arduino IDE, I need to change some entries in the header files first, otherwise the IDE won't find them. For instance I need to change `#include <wolfssl/wolfcrypt/types.h>` to `#include <wolfcrypt/types.h>`. When having done so, the error message of the compiler is the same as above.

Note: Also, the [example ](https://github.com/wolfSSL/wolfssl/blob/master/IDE/ARDUINO/sketches/wolfssl_client/wolfssl_client.ino) does not compile. I tried to compile it for standard Arduino Boards but it fails. It seems that calls to any functions of wolfSSL do not work, the output is:
```
C:\Users\pesce\AppData\Local\Temp\ccifgqBu.ltrans3.ltrans.o: In function `setup':

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:43: undefined reference to `wolfTLSv1_2_client_method'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:48: undefined reference to `wolfSSL_CTX_new'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:54: undefined reference to `wolfSSL_CTX_set_verify'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:55: undefined reference to `wolfSSL_SetIOSend'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:56: undefined reference to `wolfSSL_SetIORecv'

C:\Users\pesce\AppData\Local\Temp\ccifgqBu.ltrans3.ltrans.o: In function `loop':

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:121: undefined reference to `wolfSSL_get_error'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:122: undefined reference to `wolfSSL_ERR_error_string'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:114: undefined reference to `wolfSSL_pending'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:134: undefined reference to `wolfSSL_free'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:96: undefined reference to `wolfSSL_new'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:98: undefined reference to `wolfSSL_get_error'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:99: undefined reference to `wolfSSL_ERR_error_string'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:105: undefined reference to `wolfSSL_get_version'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:109: undefined reference to `wolfSSL_write'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:110: undefined reference to `wolfSSL_get_current_cipher'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:112: undefined reference to `wolfSSL_CIPHER_get_name'

C:\Users\pesce\Documents\Arduino\sketch_apr13a/sketch_apr13a.ino:115: undefined reference to `wolfSSL_read'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Ethernet.
```

Thanks for your feedback!
paško

kaleb-himes, 2017-04-19:
Hi @pasko-zh,

Thanks for this report! The issue is stemming from two new .c files that are included inline in ssl.c and should not be compiled directly. I have updated the IDE/ARDUINO/wolfssl-arduino.sh script to remove these files from the root directory when building an arduino sketch.

See: https://github.com/wolfSSL/wolfssl/pull/868

Warm Regards,

Kaleb

closed 2017-04-19 by kaleb-himes

pasko-zh, 2017-04-20:
@kaleb-himes : Thanks a lot for this quick reply and the fix! Now it compiles without any problems :-)