wolfSSL/wolfssl #892: C# Wrapper DTLS-Server Example Does Not Work

dustycodes, 2017-05-03:
Hey,

I just cloned the latest on the master branch (1dc5a0fba268cac8de15a17bd95a3bc4a7975772) and cannot get this portion of the code to work in the wolfSSL-DTLS-Server project:
```
Console.WriteLine("Calling ctx Init from wolfSSL");     
  
ctx = wolfssl.CTX_dtls_new(wolfssl.useDTLSv1_2_server());

if (ctx == IntPtr.Zero)
{
    Console.WriteLine("Error creating ctx structure");
    wolfssl.CTX_free(ctx);
    return;
}
```
I receive this error whenever making a call to useDTLSv* methods:

> "System.EntryPointNotFoundException: Unable to find an entry point named 'wolfDTLSv1_2_server_method' in DLL 'wolfssl.dll'.\r\n   at wolfSSL.CSharp.wolfssl.wolfDTLSv1_2_server_method()\r\n   at wolfSSL.CSharp.wolfssl.useDTLSv1_2_server() in C:\\Users\\dustycodes\\Documents\\Repositories\\wolfssl\\wrapper\\CSharp\\wolfSSL_CSharp\\wolfSSL.cs:line 1349"

I followed the guide [here ](https://www.wolfssl.com/wolfSSL/Docs-wolfssl-csharp-manual.html )to get setup using the wrapper. It still could be a bad setup on my part but I have run through it 3 times and have arrived at the same checkpoint.

cconlon, 2017-05-03:
Hi,

When you are compiling wolfSSL and the C# wrapper, have you defined **WOLFSSL_DTLS**?  This will enable DTLS functionality.  You should be able to add this either to your list of preprocessor defines in the project properties, or in the <user_settings.h> file found in the <wolfssl_root>/IDE/WIN directory.

Best Regards,
Chris

dustycodes, 2017-05-03:
Adding it to the preprocessor defines in the project properties did not work for me. I added it to user_settings.h and it worked! Thank you!

closed 2017-05-03 by dustycodes