wolfSSL/wolfssl #898: AesCtr could not handle arbitury plaintext/ciphertext length

zhou0, 2017-05-08:
the wc_AesCtrEncrypt function , prototype void wc_AesCtrEncrypt(Aes* aes, byte* out, const byte* in, word32 sz) does not work correctly if the last parameter is not a multiple of 16.

I

kaleb-himes, 2017-05-08:
Hi zhou0,

That is correct. We tried to make that apparent via the comment in our documentation example which can be reviewed [HERE](https://www.wolfssl.com/wolfSSL/Docs-wolfssl-manual-18-1-wolfcrypt-api-aes.html). I've included the noted comment below:

```
wc_AesCtrEncrypt

...

byte msg[AES_BLOCK_SIZE * n]; //n being a positive integer making msg some multiple of 16 bytes
```

If you wish to encrypt something that is not a multiple of 16, you would do the following:

1. Create a buffer that is ```AES_BLOCK_SIZE * n``` as noted in the example.
2. XMEMSET the buffer to all 0's: ```XMEMSET(your_buffer, 0, AES_BLOCK_SIZE*n);```
3. Copy your msg into the buffer and decide on a padding scheme: ```XMEMCPY(your_buffer, your_msg, XSTRLEN(your_msg));```
4. Then let's say you decide to pad with the byte ```0xAB```, fill the remainder of ```your_buffer``` with 0xAB bytes before calling wc_AesCtrEncrypt, or just leave them as 0x00 (result from XMEMSET).
5. Then when you decrypt your application will need to be aware of the same padding scheme. 

wolfSSL does not implement padding by default, we leave this up to the user, so any buffer passed to the aes encrypt method will be ignored or skipped if it is not a multiple of AES_BLOCK_SIZE (IE 16).

Warm  Regards,

Kaleb

kaleb-himes, 2017-05-08:
Hi zhou0,

Since this is expected behavior and covered in our documentation I am going to close this issue for now. I will take the action to see if there is a better way to document this for our users.

Best Regards,

Kaleb

kaleb-himes, 2017-05-08:
Hi zhou0,

I apologize my previous statement is inaccurate since you are using the stream cipher. The AesCtrEncrypt API should work with a non-standard length input and I just tested this with the attached simple program (change the .txt to .c to compile).

Could you send me a simple test case like the one I attached to reproduce the issue you are seeing?

Warmest Regards,

Kaleb

[aesctr-nonstandard-blocksize-test.txt](https://github.com/wolfSSL/wolfssl/files/984639/aesctr-nonstandard-blocksize-test.txt)

kaleb-himes, 2017-05-08:
Hi zhou0,

Have you tried running the wolfcrypt test application (wolf-root)/wolfcrypt/test/test.c to make sure the crypto is working properly? There is a test case in that application that tests the wc_AesCtrEncrypt API with a non-standard input size.

Warm Regards,

Kaleb

zhou0, 2017-05-09:
@kaleb-himes  In my application,  i do an AesCtr encryption and a decryption, but the result plaint texts dont match.  the test program does not cover this.  first, the test case only tests for encryption but not for decryption.  second,  you should use AES_DECRYPTION as the fifth parameter in the 2nd   wc_AesSetKeyDirect call.

kaleb-himes, 2017-05-09:
Hi zhou0,

I see the confusion now. 

AES CTR MODE does the same operation for encryption and decryption. If you pass in encrypted data it gets decrypted. If you pass in plain text data it gets encrypted. The example is not incorrect, you should NOT set AES_DECRYPTION for the 5th argument when using AES CTR MODE. AES_DECRYPTION is only used with BLOCK ENCRYPTION MODES. [See comment in test.c here](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/test/test.c#L3901)

I have updated the example from yesterday to show how this should work and included a case of using AES_DECRYPTION for the fifth argument to show that the result is indeed incorrect when AES CTR MODE is used in this way.

[test.txt](https://github.com/wolfSSL/wolfssl/files/987100/test.txt)

kaleb-himes, 2017-05-09:
Hi zhou0,

Thank you again for opening this issue as it gave us the opportunity to update our documentation.

I have updated the documentation section covering the ```wc_AesSetKeyDirect``` API to include a note about using this API in AES COUNTER MODE:

>  wc_AesSetKeyDirect
> 
> ...
> 
> dir - Cipher direction. Set AES_ENCRYPTION to encrypt,  or AES_DECRYPTION to decrypt. (See enum in wolfssl/wolfcrypt/aes.h) ```(NOTE: If using wc_AesSetKeyDirect with Aes Counter mode (Stream cipher) only use AES_ENCRYPTION)```
>
> ...
> 

This will be updated the on the website the next time we upload our documentation. Thanks again for your question and let us know if anything else comes up!

Warmest Regards,

Kaleb

closed 2017-05-09 by kaleb-himes

zhou0, 2017-05-10:
@kaleb-himes  Problem solved. Thank you!