wolfSSL/wolfssl #908: Unable to build with --disable-harden

gonzzor, 2017-05-11:
```
$ ./configure --disable-harden && make -j
....
wolfcrypt/src/ecc.c:1935:31: error: ‘wc_off_on_addr’ defined but not used [-Werror=unused-const-variable=]
     static const wolfssl_word wc_off_on_addr[2] =
                               ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:3224: recipe for target 'wolfcrypt/src/src_libwolfssl_la-ecc.lo' failed

```
The reason is the commit c1640e8a3dcb418d9f3f609fe16323c3629cc4de which did this
```
@@ -1907,7 +1930,7 @@ done:
     !defined(__cplusplus)
     /* let's use the one we already have */
     extern const wolfssl_word wc_off_on_addr[2];
-#elif defined(ECC_TIMING_RESISTANT)
+#else
     static const wolfssl_word wc_off_on_addr[2] =
     {
     #if defined(WC_64BIT_CPU)
```

By reverting that change it's possible to build again.

gonzzor, 2017-05-11:
Thanks, that was quick :)

dgarske, 2017-05-12:
You are welcome. Let us know if you find anything else.

closed 2017-05-12 by dgarske