wolfSSL/wolfssl #910: How to compile nginx with wolfSSL?

macedigital, 2017-05-11:
What is the official method of using wolfSSL together with nginx?

The [3.11.0 release notes](https://github.com/wolfSSL/wolfssl/releases/tag/v3.11.0-stable) mention nginx (and haproxy) support, but I fail to see how it would be possible to compile nginx with wolfSSL.

AFAIK, the safest way of embedding a SSL/TLS-library in nginx is by using the `--with-openssl="path/to/extracted/sources"`flag. But this doesn't work as wolfSSL has a directory structure that isn't exactly "compatible" with OpenSSL (nor with what the nginx build scripts would expect). Symlinking `wolfssl-config` -> `config` is nowhere near enough ...

Alternatively, should I first compile wolfSSL as a library (+ development files), and then add linker options, etc., so nginx will then use those at compile & runtime?

kaleb-himes, 2017-07-07:
Hi macedigital,

wolfSSL retains "patch files" that must be applied to specific versions of nginx for compatibility purposes. Please contact rod@wolfssl.com or rich@wolfssl.com or support@wolfssl.com for further details.

Best Regards,

Kaleb

closed 2017-07-07 by kaleb-himes

macedigital, 2017-07-11:
@kaleb-himes Thanks for the suggestion, email is underway 👍