wolfSSL/wolfssl #961: Certificate CA path length check broken for server chains with included trusted self signed CA

dr-duplo, 2017-06-12:
It seems like the check for the CA path length is broken for the case when the peer sends a complete chain including a self signed root certificate. The code checks the path length of the cert against it's issuer which is the cert itself and fails. The check should not be done if the cert is a trusted CA.

Setup:
- wolfssl 3.11.0
- TLS client loads self signed trusted root CA into ctx
- server sends complete cert chain including the trusted root CA
Error:
- pathlen check fails for the trusted root CA (tested against itself)
Expected:
- pathlen check is not executed for the trusted self signed root CA

dgarske, 2017-06-13:
Hi Dr Duplo, is it possible you can provide a certificate chain to reproduce it. If you would prefer to not post here you can email it to us at support@wolfssl.com. Thanks, David Garske.

kaleb-himes, 2018-02-22:
dr-duplo,

we had another user encounter a similar issue which they were able to resolve by compiling with WOLFSSL_ALT_CERT_CHAINS. Have you tried this solution?

If you have and that does not resolve the issue could you provide a certificate chain for us to test with in order to reproduce and fix this?

Thanks!

- Kaleb

dr-duplo, 2018-03-05:
@kaleb-himes,

thanks for your info. Sounds like it would solve it.
I will try to provide a chain to you

Thanks a lot.

closed 2018-05-24 by kaleb-himes