wolfSSL/wolfssl #974: make error  on mingw with  version 3.11.0

zhou0, 2017-06-16:
Li ZHOU@LiZHOU-PC MINGW32 /d/mysocks/wolfssl-3.11.0-stable
$ uname -a
MINGW32_NT-6.1 LiZHOU-PC 2.6.0(0.304/5/3) 2016-09-07 21:23 i686 Msys

Li ZHOU@LiZHOU-PC MINGW32 /d/mysocks/wolfssl-3.11.0-stable
$ make
make -j2  all-am
make[1]: 进入目录“/d/mysocks/wolfssl-3.11.0-stable”
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
wolfcrypt/src/aes.c: In function 'AES_GCM_encrypt':
wolfcrypt/src/aes.c:3410:16: warning: implicit declaration of function '_mm_insert_epi64' [-Wimplicit-function-declaration]
         tmp1 = _mm_insert_epi64(tmp1, ibytes*8, 0);
                ^~~~~~~~~~~~~~~~
wolfcrypt/src/aes.c:3410:9: warning: nested extern declaration of '_mm_insert_epi64' [-Wnested-externs]
         tmp1 = _mm_insert_epi64(tmp1, ibytes*8, 0);
         ^~~~
wolfcrypt/src/aes.c:3410:14: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
         tmp1 = _mm_insert_epi64(tmp1, ibytes*8, 0);
              ^
wolfcrypt/src/aes.c:3411:14: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
         tmp1 = _mm_insert_epi64(tmp1, 0, 1);
              ^
wolfcrypt/src/aes.c:3534:10: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
     tmp1 = _mm_insert_epi64(tmp1, nbytes*8, 0);
          ^
wolfcrypt/src/aes.c:3535:10: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
     tmp1 = _mm_insert_epi64(tmp1, abytes*8, 1);
          ^
wolfcrypt/src/aes.c: In function 'AES_GCM_decrypt':
wolfcrypt/src/aes.c:3610:14: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
         tmp1 = _mm_insert_epi64(tmp1, ibytes*8, 0);
              ^
wolfcrypt/src/aes.c:3611:14: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
         tmp1 = _mm_insert_epi64(tmp1, 0, 1);
              ^
wolfcrypt/src/aes.c:3656:10: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
     tmp1 = _mm_insert_epi64(tmp1, nbytes * 8, 0);
          ^
wolfcrypt/src/aes.c:3657:10: error: incompatible types when assigning to type '__m128i {aka __vector(2) long long int}' from type 'int'
     tmp1 = _mm_insert_epi64(tmp1, abytes * 8, 1);
          ^
make[1]: *** [Makefile:2986：wolfcrypt/src/src_libwolfssl_la-aes.lo] 错误 1
make[1]: *** 正在等待未完成的任务....
make[1]: 离开目录“/d/mysocks/wolfssl-3.11.0-stable”
make: *** [Makefile:2053：all] 错误 2

JacobBarthelmeh, 2017-06-27:
Hi zhou0,

May we get the configure used with wolfSSL i.e. "./configure ....." and the architecture this is being ran on (32 bit vs 64 bit).

zhou0, 2017-06-29:
the error is caused by thg --enable-aesni flag on i686 platform, I can use --disable-aesni to fix this error.

closed 2017-06-29 by zhou0