wolfSSL/wolfssl #1000: Stack smashing

thomasknauth, 2017-06-27:
I'm trying to identify who's to blame for the the following MWE to fail:

```
#include <wolfssl/ssl.h>
#include <wolfssl/wolfcrypt/asn.h>

void cb(void) {
    DecodedCert d;
    InitSignatureCtx(&d.sigCtx, NULL, INVALID_DEVID);                                                            
}

int main(int argc, char** argv)
{
    cb();
    return 0;
}
```

Initially I thought gcc might be to blame, but clang also informs me of the stack smashing. I've built wolfssl from source with

`CC=clang ./configure --prefix=$HOME/local --enable-static CFLAGS="-fstack-protector" && make V=1 install`

The MWE is based on client-tls.c from wolfssl-examples/tls built with `make clean && make CC=clang minimal`. Clang is 3.8.0-2ubuntu4 (tags/RELEASE_380/final), but the problem is also present when compiling with Ubuntu's version of gcc-5/6/7.

Stepping through the assembly, I indeed see the stack canary being overwritten. It looks to me like the canary is misplaced, but then again it would be very weird if two different compilers did the same mistake.

JacobBarthelmeh, 2017-06-27:
Hi thomasknauth,

Often with applications linking to wolfSSL and seeing stack issues the adjustment needed is having #include <wolfssl/options.h> before any other inclusion of wolfSSL headers. Can you add this and see if that resolves the issue. In this case would be changed to look like

```
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <wolfssl/wolfcrypt/asn.h>
```

thomasknauth, 2017-06-27:
Well, that indeed fixed it. Makes me wonder why this header is not included in the examples, if it is important.

closed 2017-06-27 by thomasknauth