wolfSSL/wolfssl #1001: Building outside source tree

davidgfnet, 2017-06-28:
I'm building wolfssl for several archs and running configure from outside the source dir. This worked well with cyassl but since I upgraded from 3.0 now I'm getting this:

`make[1]: Entering directory '/home/david/projects/telbot/build_wolfssl/mips-linux'`

If I run it from the source dir it just works which is wrong I guess
Thanks!

JacobBarthelmeh, 2017-06-28:
Hi davidgfnet,

Thank you for the report. I'd like to pin down a little more about how it is built and what is being ran to try and replicate the issue. Is it a process like this

```
git clone https://github.com/wolfssl/wolfssl
cd wolfssl
./autogen.sh
cd ../
mkdir build
cd build
../wolfssl/configure
make
```

After getting 

> make[1]: Entering directory '/home/david/projects/telbot/build_wolfssl/mips-linux'

what happens? Does it hang here?

davidgfnet, 2017-06-29:
Shit I copied the wrong bit :D
Well it's gcc complaining about not being able to find options.h IIRC.
Building in the source dir (./configure) just works, so for now I modified
my automation to copy the source dir to every platform and that works (just
uses more space of course).

JacobBarthelmeh, 2017-06-30:
I can confirm that with the above steps for building outside of the wolfSSL root directory their is a build error.

```
  CC       examples/server/tests_unit_test-server.o
make[1]: *** No rule to make target `cyassl/options.h', needed by `all-am'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

Have logged it as a known issue.

closed 2017-07-02 by davidgfnet

ejohnstown, 2018-06-25:
Have you tried this recently? In v3.15.3, we updated the configure to handle the cyassl/options.h the same way wolfssl/options.h is handled.