wolfSSL/wolfssl #1013: ECC pubkey encryption

davidgfnet, 2017-07-09:
Hello,
I see that the ecc encryption routine takes a private and a public key. Digging into the function I see that the algorithm is based in generating a shared secret between two parties (priv key from sender, pub key from receiver) and then using AES128.
Now my question is: can't we have the RSA counterpart for wc_RsaPublicEncrypt? Where we just take a public ECC key and encrypt some small amount of data? I'm implementing a custom protocol and while I could generate a key pair in each client and use this function to exchange an encrypted shared secret, I'd rather not use key generation functions on the client.

Something along the lines of a simple wc_ecc_mulmod_ex that takes the data (provided it's shorter than the key size) and multiplies it by the public key. Then the other party should be able to recover the data by just multiplying again by the private key. Am I missing something?

Thanks!

kaleb-himes, 2017-07-14:
Hi davidgfnet,

Clarifying some things first:

> (priv key from sender, pub key from receiver) and then using AES128.

You should never ever send your private key to anyone or it would no longer be private. Let's instead state that line as:

"(pub key from sender, priv key from receiver) and then using AES128."

--------

> can't we have the RSA counterpart for wc_RsaPublicEncrypt?

I believe you meant:

"can't we have the ECC counterpart for wc_RsaPublicEncrypt?"

--------

If I have that correct and you are looking for an ECC counterpart to wc_RsaPublicEncrypt then the unfortunate answer is NO. The reason RSA can encrypt things directly is due to a natural by-product of the math involved. ECC does not work this way.

See more on this topic:

https://crypto.stackexchange.com/questions/31617/can-elliptic-curve-25519-be-used-to-encrypt-file
https://security.stackexchange.com/questions/63506/using-ecc-like-rsa-to-encrypt-decrypt
https://crypto.stackexchange.com/questions/12823/ecdsa-vs-ecies-vs-ecdh

Let us know if you have any other questions.

Warm Regards,

Kaleb

kaleb-himes, 2017-07-14:
Hi davidgfnet,

Clarifying some things first:

--------

> can't we have the RSA counterpart for wc_RsaPublicEncrypt?

I believe you meant:

"can't we have the ECC counterpart for wc_RsaPublicEncrypt?"

---------

If I have that correct and you are looking for an ECC counterpart to wc_RsaPublicEncrypt then the unfortunate answer is NO. The reason RSA can encrypt things directly is due to a natural by-product of the math involved. ECC does not work this way.See more on this topic:
https://crypto.stackexchange.com/questions/31617/can-elliptic-curve-25519-be-used-to-encrypt-file
https://security.stackexchange.com/questions/63506/using-ecc-like-rsa-to-encrypt-decrypt
https://crypto.stackexchange.com/questions/12823/ecdsa-vs-ecies-vs-ecdh

Let us know if you have any other questions.

Warm Regards,

Kaleb

kaleb-himes, 2017-07-14:
Hi davidgfnet,

Just as a followup I know you stated you wish to avoid the key generation functionality which we do use in the API wc_ecc_encrypt (ESDSA + AES) to do something very similar to what is described in the Elliptic Curve Integrated Encryption Scheme (ECIES). If you are willing to consider generating a shared secret you will be able to achieve encryption with ECC based algorithms.

If you would like to engage us on a consulting basis to discuss this more in depth we would be happy to hear from you via our support channel support@wolfssl.com

davidgfnet, 2017-07-14:
Hey sorry for that, my question was a bit sloppy.
But back to the topic, it is indeed possible to encrypt/decrypt with ECC, it's just it is *desirable* to use a key exchange algorithm, isn't it? So in theory you could just multiply a point (which is your data) by your public key, so that the other end can use the privatekey to undo that multiplication and get the original point, aka value.

davidgfnet, 2017-07-14:
Er yeah, ECIES is what I was looking for if I'm not mistaken.
Do you accept pull requests if I decide to implement it?

kaleb-himes, 2017-07-14:
We do accept Pull Requests as long as the requesting party has signed a contributor agreement with wolfSSL Inc. I can get you a copy to review if you would like, just submit a request for a contributor agreement to info@wolfssl.com

If you do decide to implement ECIES based on one of the following standards:
ANSI X9.63
IEEE 1363a
ISO/IEC 18033-2
SECG SEC-1.

We would be happy to review it pending a signed contributor agreement. I would recommend taking a look at our ```wc_ecc_encrypt``` API as a reference before taking on that project as well!

Warm Regards,

Kaleb

closed 2017-12-21 by kaleb-himes