wolfSSL/wolfssl #1025: Unable to load private that was generated by openssl command.

NamTe, 2017-07-14:
Hi.
I generated rsa key by openssl using command 

```
ubuntu@ubuntu-01:~$ openssl genrsa -out private.pem 3072
ubuntu@ubuntu-01:~$ openssl pkcs8 -topk8 -inform PEM -outform DER -in private.pem -out test.p8 -nocrypt

```

After that, I use wolfssl to load private key file (test.p8) but the function wc_RsaPrivateKeyDecode return the error -143 please help me to resolve this issue. If i have wrong code please point out the problem. Thank in advance.

```
int rsa_load_private_file(const char* derFile, RsaKey *rsaKey)
{
    int ret = EXIT_FAILURE;
    FILE *file;
    byte *buffer = NULL;
    word32 bytes = 0;
    word32 idx = 0;
    //byte *buffer_cpy = NULL;
    file = fopen(derFile, "rb");
    //fseek(file, 0, SEEK_SET);
    //fseek(file, 22, SEEK_SET);
    if (file)
    {
        buffer = (byte*) malloc(RSA_KEY_SIZE);
        if (buffer)
        {
            bytes = fread(buffer, 1, RSA_KEY_SIZE, file);
            fclose(file);
        }
    }
    printf("Load RSA Public Key: Len %d\n", bytes);
    hexdump(buffer, bytes, 16);
    //strncp
    if (buffer != NULL && bytes > 0)
    {
        ret = wc_RsaPrivateKeyDecode(buffer, &idx, rsaKey, (word32) bytes);
    }

    if (buffer)
    {
        free(buffer);
    }
    return ret;
}
```

kaleb-himes, 2017-07-14:
Hi NamTe,

As you found, removing the ASN1 formatting for the pkcs8 object allowed you to successfully decode the key. To achieve this programatically please use the ```wc_GetPkcs8TraditionalOffset``` API as see in the updated function below:

```
int rsa_load_private_file(const char* derFile, RsaKey *rsaKey)
{
    int ret = EXIT_FAILURE;
    FILE *file;
    byte *buffer = NULL;
    word32 bytes = 0;
    word32 idx = 0;
    file = fopen(derFile, "rb");

    if (file)
    {
        buffer = (byte*) malloc(RSA_KEY_SIZE);
        if (buffer)
        {
            bytes = (word32) fread(buffer, 1, RSA_KEY_SIZE, file);
            fclose(file);
        }
    }

    if (buffer != NULL && bytes > 0)
    {
        ret = wc_GetPkcs8TraditionalOffset(buffer, &idx, bytes);
        ret = wc_RsaPrivateKeyDecode(buffer, &idx, rsaKey, (word32) bytes);
    }

    if (buffer)
    {
        free(buffer);
    }

    return ret;
}
```

NamTe, 2017-07-14:
Dear Kaleb.
That's great. Thank you so much. That help me a lot. I have spent the whole day looking for a solution :D

closed 2017-07-14 by NamTe

kaleb-himes, 2017-07-14:
NamTe,

Not a problem, we are always happy to help!

One final mention, I forgot to include the ret check on the new API, it's always good practice to ensure everything is running correctly:

```
        ret = wc_GetPkcs8TraditionalOffset(buffer, &idx, bytes);
        if (ret != 0) {
            printf("Failed to get traditional offset\n");
            return ret;
        }
        ret = wc_RsaPrivateKeyDecode(buffer, &idx, rsaKey, (word32) bytes)
```

NamTe, 2017-07-14:
Hi. Kaleb.

Thank for your support at the issue that i faced.
Now I have another problem. I have a certificate file

That after run the command

> openssl x509 -in test.cer -text -inform der

it show the ouput

> 
> Certificate:
>     Data:
>         Version: 1 (0x0)
>         Serial Number: 1727 (0x6bf)
>     Signature Algorithm: rsassaPss
>          Hash Algorithm: sha256
>          Mask Algorithm: mgf1 with sha256
>          Salt Length: 0x20
>          Trailer Field: 0xBC (default)
>         Issuer: C=..., O=..., OU=..., CN=...
>         Validity
>             Not Before: .... GMT
>             Not After : ..... GMT
>         Subject: C=..., O=..., OU=..., CN=...
>         Subject Public Key Info:
>             Public Key Algorithm: rsaEncryption
>                 Public-Key: (3072 bit)
>                 Modulus:
>                     00:9a:83:86:...............
>                 Exponent: 17 (0x11)
>     Signature Algorithm: rsassaPss
>          Hash Algorithm: sha256
>          Mask Algorithm: mgf1 with sha256
>          Salt Length: 0x20
>          Trailer Field: 0xBC (default)
> 
>          0c:0a:a2:59:b7...............
> -----BEGIN CERTIFICATE-----
> MIIEljCCAsoCAga/MEEGCSqGSIb3DQEBCjA0oA8wDQYJYIZIAWUDBAIBBQChHDAa
> ...............
> -----END CERTIFICATE-----

I try to extract public key by loading test.cer file. by using the code.

```
int rsa_load_public_key(const char* derFile, RsaKey *rsaKey)
{
    int ret;
    int derCertSz;
    byte derCert[4096];

    FILE* file;
    RsaKey pubKey;
    word32 idx = 0;
    DecodedCert cert;

    /* open and read DER-formatted cert into buffer */
    file = fopen(derFile, "rb");
    if (!file)
    {
        printf("%s\n", "fopen");
        return -1;
    }

    derCertSz = fread(derCert, 1, sizeof(derCert), file);
    fclose(file);

    /* initialize DecodedCert with DER cert */
    InitDecodedCert(&cert, derCert, derCertSz, 0);
    ret = ParseCert(&cert, CERT_TYPE, NO_VERIFY, 0);
    if (ret != 0)
    {
        printf("%s\n", "ParseCert");
        return -1;
        // ParseCert failed
    }

    /* extract the public key from the cert */
    wc_FreeRsaKey(rsaKey);
    wc_InitRsaKey(rsaKey, NULL);
    idx = 0;
    ret = wc_RsaPublicKeyDecode(cert.publicKey, &idx, rsaKey, cert.pubKeySize);
    if (ret != 0)
    {
        printf("%s\n", "RsaPublicKeyDecode");
        return -1;
    }
    return 0;
}

```
But the function ParseCert return -1. I also try example from wolfssl-examples/certfields/main.c but it Failed to convert DER to WOLFSSL_X509. So please give me a hint.