wolfSSL/wolfssl #1037: POINT_CONVERSION_COMPRESSED

zero-rp, 2017-07-19:
Can you implement  POINT_CONVERSION_COMPRESSED in ECC

dgarske, 2017-07-31:
Hi zero-rp,

We support compressed ECC key import/export by defining `HAVE_COMP_KEY`. However it sounds like you are looking to convert points only. Can you provide some additional detail on what it is that you are trying to do?

Thanks,
David Garske, wolfSSL

zero-rp, 2017-07-31:
Yes,Let me give you an example.This is what I use to exchange keys on the network.
    
    static EC_KEY *ECDHtemppubkey(unsigned char *pubkey)
    {
    EC_KEY *ecdh = EC_KEY_new();
    ecdh = EC_KEY_new_by_curve_name(NID_X9_62_prime256v1);//NID_secp521r1  
    EC_KEY_generate_key(ecdh);
    const EC_POINT *point = EC_KEY_get0_public_key(ecdh);
    const EC_GROUP *group = EC_KEY_get0_group(ecdh);
    EC_POINT_point2oct(group, point, POINT_CONVERSION_COMPRESSED, pubkey, 33, NULL);
    return ecdh;
    }
In this example, the points on the ECC need to be compressed into 33 bytes of data for network transmission

JacobBarthelmeh, 2017-08-23:
Hi zero-rp,

The function to do the same as with the use case in the code snippet is wc_ecc_export_x963_ex. This can take an argument to compress a key and is used to export the compressed public key.

Example sudo code would be

```
ecc_key key;
int ret;
unsigned char buf[100];
unsigned int sz = sizeof(buf);
wc_ecc_init(&key);

// init rng to use

ret = wc_ecc_make_key(&rng, 32, &key);
//check ret
ret = wc_ecc_export_x963_ex(&key, buf, &sz, 1);
//check ret
```

Which gives a 33 byte compressed public key stored in "buf". I guess to understand the feature request better is this a request to expand the compatibility layer to include EC_POINT_point2oct with POINT_CONVERSION_COMPRESSED? If able to use wolfSSL API then wc_ecc_export_x963_ex will work.

JacobBarthelmeh, 2017-12-21:
Hi zero-rp,

Am going to go ahead and close this report. Please let us know of any feature gaps that could be addressed or work towards setting up a call if needed to go over the project. A good email address to send feature requests to would be info@wolfssl.com.

Regards,
Jacob

closed 2017-12-21 by JacobBarthelmeh