wolfSSL/wolfssl #1125: No rule to make target `cyassl/options.h', needed by `all-am'.  Stop.

DengkeDu, 2017-08-29:
version: 3.12.0
In Makefile, I find:

> am__append_2 = cyassl/options.h

> nobase_include_HEADERS = ...
   ...
   $(am__append_2)

Why this failed?

DengkeDu, 2017-08-29:
wolfssl/options.h have its rule:

> wolfssl/options.h: $(top_builddir)/config.status $(top_srcdir)/wolfssl/options.h.in
cd $(top_builddir) && $(SHELL) ./config.status $@

so can we add the same?

ejohnstown, 2017-12-20:
I am unable to recreate this issue with GitHub **tag v3.12.0-stable** on macOS or Ubuntu. Are you using a specific GitHub commit other than **tag v3.12.0-stable**? With a new clone from the GitHub repository, I am using the command lines:

    $ git clone https://github.com/wolfSSL/wolfssl.git
    $ cd wolfssl
    $ git co v3.12.0-stable
    $ ./autogen.sh && ./configure --enable-distro && make
    $ make check

and it builds cleanly. I am not getting an error on **cyassl/options.h**. How are you trying to build?

DengkeDu, 2017-12-21:
It works now, thanks!

closed 2017-12-21 by DengkeDu