wolfSSL/wolfssl #1127: AES CTR mode with STM32F2 Stdperih is actually CBC

gonzzor, 2017-08-30:
The CTR mode for STM32F2 doesn't work when using stdperih lib, it doesn't keep track of the counter but rather use the output from the last block hinting that this is copied from CBC and hasn't been properly changed nor tested.

The following line should increment the counter, not memcpy the previous block.

https://github.com/wolfSSL/wolfssl/blob/34dc4a493d64160d6d87a5cfbd7c0ca803fce68a/wolfcrypt/src/aes.c#L3140

dgarske, 2017-08-31:
Hi Gonzzor,

Thanks for the these AES STM32F2 StdPeriLib reports. I'm working on fixes these and should have a PR released soon.

Thanks, David Garske

dgarske, 2017-09-18:
Fixes for STM32F2/F4 StdPeriLib have been pushed to PR #1116 and passed our wolfCrypt tests. 

Please let me know if you have any issues. Thanks, David Garske, wolfSSL.

closed 2017-10-10 by dgarske