wolfSSL/wolfssl #1131: Unable to handshake with validec.entrust.net received fatal error(40) and -313. Using full build

ifree, 2017-09-01:
I build wolfssl with --enable-all option, still get the fatal error, but I can successfully connect with openssl client.

kaleb-himes, 2017-09-01:
Hi ifree,

Yes it looks like that server requires the ```ec_point_formats``` extension in order to honor a Client connection attempt. Without that extension it will outright reject the Client Hello packet. 

wolfSSL is in the process of adding this extension as we speak. If you checkout this Pull Request https://github.com/wolfSSL/wolfssl/pull/1034 you can successfully test connecting to that domain (See result below).

```
khimes$ ./examples/client/client -h validec.entrust.net -p 443 -d -g
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
SSL curve name is SECP256R1
SSL connect ok, sending GET...
HTTP/1.1 404 Not Found
Date: Fri, 01 Sep 2017 15:58:35 GMT
Server: Apache
Co
ntent-Length: 208
Connection: close
Content-Type: text/html; charset=iso-8859
```

I did the following:

```
git clone https://github.com/wolfssl/wolfssl.git
cd wolfssl
git pull https://github.com/moisesguimaraes/wolfssl.git ec-point-format
./autogen.sh
./configure && make
./examples/client/client -h validec.entrust.net -p 443 -d -g
```

Once that PR is merged you will be able to test with master. It is currently under review and has a few minor fixes to be addressed with some of our other build environments. @moisesguimaraes is actively working the issue!

Warmest Regards,

KH

ifree, 2017-09-01:
Thank you very much!

closed 2017-09-01 by ifree