wolfSSL/wolfssl #1133: name space violation

bagder, 2017-09-06:
WolfSSL 3.12 has a global symbol `SHA256_DIGEST_SIZE` set to an enum in its header files. It is using the `SHA256_`  prefix (added in [this commit](https://github.com/wolfSSL/wolfssl/commit/ac6635593b0fa3ffbf4cbfd75f2e71db1671be15)) and I argue that's not suitable to do as it has a high risk of colliding with other libs' or programs' used names.

This exact symbol causes [build failures in curl](https://github.com/curl/curl/issues/1865). While we will work around this issue, I think this is a flaw in WolfSSL.

dgarske, 2017-09-08:
Hi Bagder,

We appreciate your feedback and are always happy to see open source projects keeping us updated.

To rename the `SHA256_DIGEST_SIZE` enum cause issues with anyone who uses it. Another possibility would be for us to change it to a define. This would allow you to do an `#ifndef SHA256_DIGEST_SIZE` type thing to make sure its defined. I've made note of this and will discuss it with the wolf team.

I looked at the CURL issue you mentioned and it appears to be resolved via PR (https://github.com/curl/curl/pull/1867), which renames the CURL enum to `CURL_SHA256_DIGEST_SIZE`. That seems like a good solution.

Thanks,
David Garske, wolfSSL

bagder, 2017-09-08:
Thanks. Yes other TLS/crypto libs have provided that as a `#define` already which then has worked great for us since that's easily checked. enums are not as easy.

We decided that the cleanest work-around for the compile error we get now was to change the symbol we use. That was fairly easy for us to do here, but you should also be aware that it means that all existing curl versions (and possibly other projects using similar tricks) will get a compiler error with WolfSSL 3.12, and only 7.56.0 (due to be released on October 4, 2017) and later versions will build cleanly with it.

(I realize you may not be able to change this very easily now when already shipped in a release.)

dgarske, 2017-09-08:
Hi Bagder,

I did want to clarify that wolfSSL has used `SHA256_DIGEST_SIZE` in sha256.h as an enum as far back as I can trace in our current wolfSSL repo (12/17/14).

You could also do an ifdef check for the `WOLF_CRYPT_SHA256_H` header define to determine if wolf is included.

Thanks,
David Garske, wolfSSL

dgarske, 2017-11-03:
Hi Bagder,

The latest v3.12.2 release has a refactor of all hash types and SSL_ types to allow better compatibility when used along side other libraries. You can disable the old name support using `./configure --disable-oldnames` or defining `NO_OLD_WC_NAMES`. With this defined you can have `SHA256_DIGEST_SIZE` defined and it will not conflict. Our library uses `WC_SHA256_DIGEST_SIZE`.

Thanks,
David Garske, wolfSSL

closed 2017-11-03 by dgarske