wolfSSL/wolfssl #1139: wc_AesDecrypt doesn't work on STM32 with stdperihlib

gonzzor, 2017-09-12:
When using stdperihlib on STM32F2 the wc_AesDecrypt() function doesn't work properly.

The decryption key isn't properly set in the hardware, comparing the code with the one provided in the stdperihlib there is a section missing.

The missing part should be here.
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/aes.c#L465

But why is the code copied at all to begin with instead of using the CRYP_AES_ECB() from the stdperihlib?

dgarske, 2017-09-12:
Hi gonzzor,

Have you tried with the latest in PR #1116 ? There have been a few fixes thanks to your reports. I will have time later this week to finalize those fixes. There is still an issue with AES CTR when using the remainder logic to handle unaligned blocks, but other AES tests pass.

Thanks,
David Garske, wolfSSL

gonzzor, 2017-09-13:
Hi,

That sounds really promising. I haven't had time to check test it, but by looking at the changes it should work.

Comparing with CRYP_AES_ECB() I can see that it uses CRYP_DataType_32b instead of CRYP_DataType_8b when setting the key. It also flushes the FIFO before setting the key.

Not sure if any of them actually matters.

Looking forward to have all the test pass :)

dgarske, 2017-09-18:
Fixes for STM32F2/F4 StdPeriLib have been pushed to PR #1116 and passed our wolfCrypt tests. 

The CRYP_DataType_8b is the one we want to use. The FIFO clear order in relation to the key is not important to the operation.

Thanks, David Garske, wolfSSL.

closed 2017-10-10 by dgarske