wolfSSL/wolfssl #1147: Misleading path in porting guide

tobhe, 2017-09-18:
The porting guide mentions:
> In the `./wolfssl/ctaocrypt/settings.h` file, there are several defines specific to different operating systems, TCP/IP stacks, and chipsets (MBED, FREESCALE_MQX, MICROCHIP_PIC32, MICRIUM, EBSNET, etc.). 

I believe the correct path would be `wolfssl/wolfcrypt/settings.h`

kaleb-himes, 2017-09-18:
@tobhe 

Thanks for noting this I have added the update to our porting guide and the next time the website gets updated this change will be reflected.

Warm Regards,

Kaleb

closed 2017-09-18 by kaleb-himes

kaleb-himes, 2017-09-18:
[wolfssl_porting_guide.pdf](https://github.com/wolfSSL/wolfssl/files/1311598/wolfssl_porting_guide.pdf)

Attaching an updated version for your reference here!

Regards,

Kaleb