wolfSSL/wolfssl #1173: Boost.Asio and WolfSSL

Krakean, 2017-10-06:
Hello
Just curious, is there any plans about to add support for WolfSSL in Boost.Asio? 
Seems like Boost.Asio author is probably quite busy nowadays with C++'s networking-ts implementation, maybe WolfSSL authors can make a pull request to Boost.Asio github repo with WolfSSL support?

JacobBarthelmeh, 2017-10-10:
Hi Krakean,

Thanks for the post, this has been added to the desired feature request list for evaluation. Currently it is not on the road map for wolfSSL but if a set deadline is encountered for needing the feature contact us.

Regards,
Jacob

Krakean, 2017-12-03:
@JacobBarthelmeh hi, any progress on this ticket?

Krakean, 2017-12-11:
@JacobBarthelmeh Jacob?

ejohnstown, 2017-12-12:
This is on our feature request list. Unfortunately at this time we are busy with paid work. It is still on the list, and we'll look at it at some point. Do you have a deadline where you think our adding this support would help you?

JacobBarthelmeh, 2018-07-02:
Hi Krakean,

We have made some progress on adding wolfSSL support with Boost.Asio. The pull request under review currently is located here https://github.com/wolfSSL/wolfssl/pull/1655. Would you be interested in being a beta tester?

Regards,
Jacob

Krakean, 2018-07-02:
Hi

Sadly, but no, because I'm no longer wolfssl user, had to stay with openssl due boost.asio compatibility.
Will revisit wolfssl for my next project.