wolfSSL/wolfssl #1255: RSA and ECDSA much slower than openssl

WuBingzheng, 2017-11-29:
I test the performance of wolfssl (3.12.2) and openssl (1.0.1e-fips).
They are almost same at AES with AESNI enabled.
However wolfssl works much slower than openssl at RSA and ECDSA.

The results in ops/s of sign/verify:

```
|         | RSA2048   | ECDSA256  |
| wolfssl | 228/2905  |  964/1443 |
| openssl | 814/27499 | 9465/2519 |
```

CPU: Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz

Is this right? or is there anything wrong?

Thank,
Wu

kaleb-himes, 2017-11-30:
Hi WuBingzheng,

Could you try running your tests with these settings:

```
./configure --disable-harden
make
./wolfcrypt/benchmark/benchmark
```

Then also try:

```
./configure --disable-harden --enable-intelasm
make
./wolfcrypt/benchmark/benchmark
```

Could you also let us know what your openssl settings were at the time of the test for example you should see output like this:

```
$ openssl speed rsa 2048
Error: bad option or value

Available values:
md4      md5      hmac     sha1     sha256   sha512   whirlpoolrmd160
seed-cbc rc2-cbc  bf-cbc
des-cbc  des-ede3 aes-128-cbc aes-192-cbc aes-256-cbc aes-128-ige aes-192-ige aes-256-ige 
camellia-128-cbc camellia-192-cbc camellia-256-cbc rc4
rsa512   rsa1024  rsa2048  rsa4096
dsa512   dsa1024  dsa2048
ecdsap160 ecdsap192 ecdsap224 ecdsap256 ecdsap384 ecdsap521
ecdsak163 ecdsak233 ecdsak283 ecdsak409 ecdsak571
ecdsab163 ecdsab233 ecdsab283 ecdsab409 ecdsab571
ecdsa
ecdhp160  ecdhp192  ecdhp224 ecdhp256  ecdhp384  ecdhp521
ecdhk163  ecdhk233  ecdhk283  ecdhk409  ecdhk571
ecdhb163  ecdhb233  ecdhb283  ecdhb409  ecdhb571
ecdh
seed     rc2      des      aes      camellia rsa      blowfish

Available options:
-engine e       use engine e, possibly a hardware device.
-evp e          use EVP e.
-decrypt        time decryption instead of encryption (only EVP).
-mr             produce machine readable output.
-multi n        run n benchmarks in parallel.
cissfips@ciss:~/wolfssl$ openssl speed rsa2048
Doing 2048 bit private rsa's for 10s: 12522 2048 bit private RSA's in 10.00s
Doing 2048 bit public rsa's for 10s: 413922 2048 bit public RSA's in 10.00s
OpenSSL 1.0.1t  3 May 2016
built on: Thu Nov  2 14:01:13 2017
options:bn(64,64) rc4(16x,int) des(idx,cisc,16,int) aes(partial) blowfish(idx) 
compiler: gcc -I. -I.. -I../include  -fPIC -DOPENSSL_PIC -DOPENSSL_THREADS -D_REENTRANT -DDSO_DLFCN -DHAVE_DLFCN_H -m64 -DL_ENDIAN -DTERMIO -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,relro -Wa,--noexecstack -Wall -DMD32_REG_T=int -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DMD5_ASM -DAES_ASM -DVPAES_ASM -DBSAES_ASM -DWHIRLPOOL_ASM -DGHASH_ASM
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.000799s 0.000024s   1252.2  41392.2
```

Could you share those details along with the results of the above tests and let us know your results?

Warm Regards,

Kaleb

kaleb-himes, 2017-11-30:
The reason ```--disable-harden``` will speed things up is it removes additional security protections against side-channel attacks offering weaker security but faster performance.

We always have the strong protection enabled by default as we consider our customers safety paramount to performance!

kaleb-himes, 2017-12-01:
Here are my results with ```--enable-intelasm --enable-sp --disable-harden``` in wolfSSL. I ran these tests on an Intel(R) Xeon(R) CPU E3-1270 v6 @ 3.80GHz

```
RSA   2048 public      61900 ops took 1.000 sec, avg 0.016 ms, 61876.868 ops/sec
RSA   2048 private      2100 ops took 1.015 sec, avg 0.483 ms, 2069.875 ops/sec
ECDSA  256 sign        62500 ops took 1.000 sec, avg 0.016 ms, 62489.377 ops/sec
ECDSA  256 verify      20500 ops took 1.000 sec, avg 0.049 ms, 20496.741 ops/sec
```

```
openssl speed rsa2048
Doing 2048 bit private rsa's for 10s: 12533 2048 bit private RSA's in 10.00s
Doing 2048 bit public rsa's for 10s: 413850 2048 bit public RSA's in 10.00s
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.000798s 0.000024s   1253.3  41385.0

openssl speed ecdsap256
Doing 256 bit sign ecdsa's for 10s: 118034 256 bit ECDSA signs in 9.99s 
Doing 256 bit verify ecdsa's for 10s: 51143 256 bit ECDSA verify in 10.00s
                              sign    verify    sign/s verify/s
 256 bit ecdsa (nistp256)   0.0001s   0.0002s  11815.2   5114.3
```

Looking forward to your results!

kaleb-himes, 2017-12-01:
Lastly here are my results on the same machine WITHOUT disabling the side-channel security:

```./configure --enable-intelasm --enable-sp```

```
RSA   2048 public      61800 ops took 1.001 sec, avg 0.016 ms, 61761.273 ops/sec
RSA   2048 private      2100 ops took 1.013 sec, avg 0.482 ms, 2072.631 ops/sec
ECDSA  256 sign        62400 ops took 1.000 sec, avg 0.016 ms, 62382.658 ops/sec
ECDSA  256 verify      20600 ops took 1.002 sec, avg 0.049 ms, 20550.348 ops/sec
```

Slightly slower but still very performant.

WuBingzheng, 2017-12-01:
Thanks for your reply.

Here are the results with some configure flags:

```
| wolfssl configure flags                        | RSA2048   | ECDSA256  |
| --disable-harden                               | 502/14213 | 1936/1494 |
| --disable-harden --enable-intelasm             | 493/13876 | 1731/1431 |
| --enable-sp --disable-harden --enable-intelasm | 550/19098 | 8724/2611 |
```

Here are the old results:

```
|         | RSA2048   | ECDSA256  |
| wolfssl | 228/2905  |  964/1443 |
| openssl | 814/27499 | 9465/2519 |
```

The `--disable-harden` and `--enable-sp` flags indeed speed up the benchmark, while ` --enable-intelasm` does not has much effect.
However, while ECDSA catches openssl, RSA is still slower than openssl, which is default installed in my CentOS 7.2.

Here is the openssl output:

```
#openssl speed rsa2048
Doing 2048 bit private rsa's for 10s: 8105 2048 bit private RSA's in 10.01s
Doing 2048 bit public rsa's for 10s: 271840 2048 bit public RSA's in 10.01s
OpenSSL 1.0.1e-fips 11 Feb 2013
built on: Tue Sep 27 13:37:25 UTC 2016
options:bn(64,64) md2(int) rc4(16x,int) des(idx,cisc,16,int) aes(partial) idea(int) blowfish(idx)
compiler: gcc -fPIC -DOPENSSL_PIC -DZLIB -DOPENSSL_THREADS -D_REENTRANT -DDSO_DLFCN -DHAVE_DLFCN_H -DKRB5_MIT -m64 -DL_ENDIAN -DTERMIO -Wall -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches   -m64 -mtune=generic -Wa,--noexecstack -DPURIFY -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DMD5_ASM -DAES_ASM -DVPAES_ASM -DBSAES_ASM -DWHIRLPOOL_ASM -DGHASH_ASM
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.001235s 0.000037s    809.7  27156.8
```

WuBingzheng, 2017-12-01:
```
| --disable-harden --enable-sp                   | 934/28471 | 31534/10508 |
```

Add another test. It seems than `--enable-intelasm` slow down the benchmark. When drop this flag, RSA catches up openssl, and ECDSA  get faster than openssl!

kaleb-himes, 2017-12-01:
Hi WuBingzheng,

That is a very interesting result indeed! Thanks for sharing your findings.
We'll look into the results with --enable-intelasm actually slowing things down on your chipset:

CPU: Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz
RESULT: enabled intelasm actually slows things down when combined with single point precision.

as opposed to the tests I ran on:

Intel(R) Xeon(R) CPU E3-1270 v6 @ 3.80GHz
RESULT: enabled intelasm greatly increases performance when combined with single point precision.

I'll let you know what we find out.

WuBingzheng, 2017-12-02:
I test flag `--enable-intelasm` again at the same machine after getting your reply.
Now this flag really speeds up the benchmask.
I try hard to reproduce my origin result, but failed.
It seems that I made a big mistake, by recording the two results (with and without the flag) wrong!
I am very sorry for wasting your time.

Sorry again,
Wu

closed 2017-12-02 by WuBingzheng

WuBingzheng, 2017-12-02:
After digging my shell history, I found part reason of my mistake.
I typed `--enable-intelasm` as `--enable-interasm` by mistake.
The `./configrue` gave a warning which I missed, and continued.
So this flag did not work at all.
So in my post of 3-test-results, the second one had the same benchmark with the first one, and the third one was not as good as openssl too.
I still do not know the reason of the mistake in my latter post.

Sorry again.

kaleb-himes, 2017-12-07:
No need to apologize! We are happy to hear you were able to achieve the same results we were seeing. Good to know the wolfSSL solution is as or more performant than OpenSSL!

Thanks for your followup and let us know if anything else comes up. We are always happy to help in any way we can.

- KH