wolfSSL/wolfssl #1266: Support WICED Studio

master-q, 2017-12-12:
```
1)  Please create a list of areas in the WICED SDK that currently use mbedtls
(ex: TLS server, wpa_supplicant, etc), along with your time estimate for porting
into each area.
We can then use that list to prioritize the areas we should work on first.

2)  Please update your existing time estimates based on new research
```

master-q, 2017-12-12:
The mbedtls is located at `/home/User/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/mbedtls_open`. Then, we can pick up number of calling mbedtls API as following:

```
$ pwd
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi
$ grep mbedtls_ -r | grep -v BESL/mbedtls_open | grep -v "^Binary file" | wc -l
379
```

master-q, 2017-12-12:
The API callings are categorized into following:

* `apps/snip/crypto` application using AES/MD5/SHA1/SHA256.
* `apps/snip/rsa_pkcs1` application using PKCS#1/RSA/SHA256.

master-q, 2017-12-12:
`WICED/security/BESL/host/WICED/wiced_supplicant.c` library is called by WIFI_COMMANDS macro:

```c
#define WIFI_COMMANDS \
    ALL_COMMANDS_ENTERPRISE_SECURITY_COMMANDS \
    WIFI_COMMANDS_LIMITED_SET \
// --snip--
    { (char*) "test_join",                          test_join,                          2, NULL, NULL, (char*) "<ssid> <open|wep|wpa_aes|wpa_tkip|wpa2|wpa2_tkip> [key] [ip netmask gateway] <iterations>"ESCAPE_SPACE_PROMPT, (char*) "Test joining an AP. DHCP assumed if no IP address provided"}, \
```

And above commands is called by following applications:

* apps/demo/apollo
* apps/demo/audio_player
* apps/demo/headset
* apps/snip/flac
* apps/snip/ota2_example
* apps/snip/ssdp
* apps/test/duktape
* apps/test/nsh

master-q, 2017-12-12:
Planed. But there are no good test for `WICED/*` library.
I believe `apps/test/console` is a choice, but I have no concrete command sequences for better coverage of testing...

master-q, 2017-12-12:
Choose 1st build target:

```
$ pwd
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi
$ ./make.sh test.console-CYW943907AEVAL1F-FreeRTOS-LwIP
$ find build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/ -name "*.o" | grep mbedtls | head
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/version_features.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/dhm.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/pk.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/ecdh.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/pkparse.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/havege.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/ssl_ticket.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/x509_csr.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/entropy.o
build/test.console-CYW943907AEVAL1F-FreeRTOS-LwIP/Modules/WICED/security/BESL/mbedtls_open/library/cmac.o
```

master-q, 2017-12-13:
@cconlon Do you have some manual to map mbed TLS API into wolfSSL?
If no, may I create it at GItHub Wiki?

master-q, 2017-12-13:
Added wolfSSL as COMPONENTS which can be built on WICED-Studio-6.0.

* https://github.com/master-q/wolfssl/commit/62c23c3c87beabe71ccf9818ea93c3167f08feb5
* https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/commit/495b21fea12fdabdb6524dd56ccb6e323ae0be6d

But I didn't choose the correct compile options which are following:

* https://github.com/master-q/wolfssl/commit/62c23c3c87beabe71ccf9818ea93c3167f08feb5#diff-e4c1909fbc5860dba7555327a5f8bdbeR38

After port `wolfcrypt/test/test.c` onto WICED-Studio-6.0, I should review the options.

master-q, 2017-12-13:
Explained how to build wolfSSL on WICED-Studio-6.0:

* https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/commit/0a3b831b94769d24b71a8d44b638f1480f6c28ff#fda3484edf8db0684440157ce0b110d784d42704_3_2

master-q, 2017-12-13:
Get following error, while build test.c:

```
Making test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP.elf
build/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP/libraries/wolfSSL.a(test.o): In function `wolfcrypt_test':
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/test/test.c:415: undefined reference to `CheckRunTimeSettings'
build/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP/libraries/wolfSSL.a(rsa.o): In function `wc_InitRsaKey_ex':
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:247: undefined reference to `mp_init_multi'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:251: undefined reference to `mp_init_multi'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:253: undefined reference to `mp_clear'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:254: undefined reference to `mp_clear'
build/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP/libraries/wolfSSL.a(rsa.o): In function `wc_FreeRsaKey':
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:362: undefined reference to `mp_forcezero'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:363: undefined reference to `mp_forcezero'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:364: undefined reference to `mp_forcezero'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:365: undefined reference to `mp_forcezero'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:366: undefined reference to `mp_forcezero'
build/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP/libraries/wolfSSL.a(rsa.o):/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:367: more undefined references to `mp_forcezero' follow
build/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP/libraries/wolfSSL.a(rsa.o): In function `wc_FreeRsaKey':
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:370: undefined reference to `mp_clear'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:371: undefined reference to `mp_clear'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:372: undefined reference to `mp_clear'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:373: undefined reference to `mp_clear'
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL/wolfssl/IDE/WICED-Studio/../../wolfcrypt/src/rsa.c:374: undefined reference to `mp_clear'
--snip--
```

master-q, 2017-12-13:
> Get following error, while build test.c:

Fix them adding `-DUSE_FAST_MATH`:

```
$ ./make_test_wolfssl_test_freertos.sh
--snip--
----------------------------------|---------|---------|
                                  |         |  Static |
              Module              |  Flash  |   RAM   |
----------------------------------+---------+---------|
App                               |       0 |     104 |
crc                               |       0 |    1060 |
FreeRTOS                          |       0 |    6625 |
Interrupt Vectors                 |       0 |     288 |
libc                              |       0 |   30818 |
LwIP                              |       0 |   39023 |
Networking                        |       0 |    1483 |
NVRam                             |       0 |    2210 |
Other                             |       0 |  154751 |
Packet Buffers                    |       0 |  110892 |
platform                          |       0 |     478 |
RAM Initialisation                |      32 |       0 |
resources                         |       0 |      32 |
Ring_Buffer                       |       0 |     140 |
Startup Stack & Link Script fill  |       0 |     275 |
WICED                             |       0 |    4035 |
Wiced_RO_FS                       |       0 |     570 |
WWD                               |       0 |     956 |
----------------------------------+---------+---------|
TOTAL (bytes)                     |       0 |  353740 |
----------------------------------|---------|---------|
--snip--
$ nm build/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP/binary/test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP.elf | grep wolfcrypt_test
004ad350 T wolfcrypt_test
```

master-q, 2017-12-13:
Used GLOBAL_DEFINES to remove duplicated compile options:

https://github.com/master-q/wolfssl/commit/3d42f00642ee86a0da74d77e17fdfaca372582ee

master-q, 2017-12-13:
Just clean up compile options:

* FREERTOS: Needed. To use FreeRTOS.
* HAVE_LWIP_NATIVE: NoNeeded. WOLFSSL_LWIP is better.
* NO_FILESYSTEM: Needed. WICED doesn't support dirent.h.
* BUILDING_WOLFSSL: NoNeeded. It just decorates function scopes.
* NDEBUG: NoNeeded. wolfSSL code doesn't menion it.
* TFM_TIMING_RESISTANT: Needed. It gets better security.
* ECC_TIMING_RESISTANT: Needed. It gets better security.
* WC_RSA_BLINDING: Needed. To avoid "For timing resistance / side-channel attack..." warning.
* WOLFSSL_SHA512: Needed. The mbed TLS also supports SHA512.
* WOLFSSL_SHA384: Needed. The mbed TLS also supports SHA384.
* NO_DSA: Needed. DSA is an old algorithm.
* HAVE_ECC: Needed. To use Elliptic-curve cryptography.
* TFM_ECC256: Needed. To get speed on ECC.
* ECC_SHAMIR: Needed. To get speed on ECC.
* WOLFSSL_BASE64_ENCODE: Needed. mbed TLS also provide base64 encode.
* NO_RC4: Needed. mbed TLS doesn't provide RC4.
* NO_HC128: Needed. mbed TLS doesn't provide HC128.
* NO_RABBIT: Needed. mbed TLS doesn't provide RABBIT.
* WOLFSSL_SHA224: Needed. mbed TLS doesn't provide SHA224.
* WOLFSSL_SHA3: Needed. It's a latest hash.
* HAVE_POLY1305: Needed. It's a latest mac.
* HAVE_ONE_TIME_AUTH: Needed.
* HAVE_CHACHA: Needed. It's a latest mac.
* HAVE_HASHDRBG: Needed.
* HAVE_TLS_EXTENSIONS: Needed.
* HAVE_SUPPORTED_CURVES: Needed. To support Elliptic-curve cryptography.
* HAVE_EXTENDED_MASTER: Needed.
* NO_PSK: NoNeeded. mbed TLS supports PSK.
* NO_MD4: NoNeeded. mbed TLS supports MD4.
* NO_PWDBASED: Needed.
* USE_FAST_MATH: Needed.
* WC_NO_ASYNC_THREADING: Needed. If disable, it depends on pthread.
* NO_DES3: NoNeeded. mbed TLS supports DES3.
* NO_MAIN_DRIVER: Needed. To remove main function in test.c.

master-q, 2017-12-14:
The mbed TLS's API is used from following functions in WICED-Studio-6.0 middleware:

* WICED/internal/wiced_cooee.c: wiced_wifi_cooee()
* WICED/network/wiced_tcpip_common.c: wiced_tcp_send_buffer()
* WICED/network/wiced_tcpip_common.c: wiced_tcp_stream_write()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_check_psk_identity()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_check_retransmission()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_event_thread()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_free_peer()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_network_receive()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_network_send()
* WICED/security/BESL/host/WICED/wiced_dtls.c: dtls_server_event_processing_thread()
* WICED/security/BESL/host/WICED/wiced_dtls.c: mbedtls_reset_context()
* WICED/security/BESL/host/WICED/wiced_dtls.c: mbedtls_ssl_copy_context()
* WICED/security/BESL/host/WICED/wiced_dtls.c: mbedtls_timing_get_delay()
* WICED/security/BESL/host/WICED/wiced_dtls.c: mbedtls_timing_set_delay()
* WICED/security/BESL/host/WICED/wiced_dtls.c: wiced_dtls_calculate_overhead()
* WICED/security/BESL/host/WICED/wiced_dtls.c: wiced_dtls_deinit_identity()
* WICED/security/BESL/host/WICED/wiced_dtls.c: wiced_dtls_encrypt_packet()
* WICED/security/BESL/host/WICED/wiced_dtls.c: wiced_dtls_init_identity()
* WICED/security/BESL/host/WICED/wiced_dtls.c: wiced_generic_start_dtls_with_ciphers()
* WICED/security/BESL/host/WICED/wiced_supplicant.c: besl_supplicant_init()
* WICED/security/BESL/host/WICED/wiced_supplicant.c: supplicant_peap_thread()
* WICED/security/BESL/host/WICED/wiced_tls.c: eap_ssl_flush_output()
* WICED/security/BESL/host/WICED/wiced_tls.c: eap_ssl_receive_packet()
* WICED/security/BESL/host/WICED/wiced_tls.c: mbedtls_debug()
* WICED/security/BESL/host/WICED/wiced_tls.c: mbedtls_reset_context()
* WICED/security/BESL/host/WICED/wiced_tls.c: my_verify()
* WICED/security/BESL/host/WICED/wiced_tls.c: ssl_flush_output()
* WICED/security/BESL/host/WICED/wiced_tls.c: ssl_receive_packet()
* WICED/security/BESL/host/WICED/wiced_tls.c: tls_add_alpn_extension()
* WICED/security/BESL/host/WICED/wiced_tls.c: tls_calculate_encrypt_buffer_length()
* WICED/security/BESL/host/WICED/wiced_tls.c: tls_host_create_buffer()
* WICED/security/BESL/host/WICED/wiced_tls.c: tls_host_get_packet_data()
* WICED/security/BESL/host/WICED/wiced_tls.c: tls_host_receive_packet()
* WICED/security/BESL/host/WICED/wiced_tls.c: tls_packetize_buffered_data()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_generic_start_tls_with_ciphers()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_calculate_overhead()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_close_notify()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_deinit_context()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_deinit_identity()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_deinit_root_ca_certificates()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_encrypt_packet()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_init_context()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_init_identity()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_init_root_ca_certificates()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_is_encryption_enabled()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_receive_eap_packet()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_receive_packet()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_reset_context()
* WICED/security/BESL/host/WICED/wiced_tls.c: wiced_tls_set_extension()
* WICED/security/PostgreSQL/fortuna.c: ciph_encrypt()
* WICED/security/PostgreSQL/fortuna.c: ciph_init()
* WICED/security/PostgreSQL/fortuna.c: md_init()
* WICED/security/PostgreSQL/fortuna.c: md_result()
* WICED/security/PostgreSQL/fortuna.c: md_update()
* libraries/protocols/websocket/websocket_handshake.c: wiced_generate_server_websocket_key()
* libraries/utilities/command_console/fs/fs_test.c: sw_calc_file_sha1()

master-q, 2017-12-14:
Above function used by following applications:

* wiced_wifi_cooee() needs new test application.
* wiced_tcp_send_buffer() called by apps/snip/websocket_server.
* wiced_tcp_stream_write() called by apps/snip/websocket_client.
* dtls_check_psk_identity() called by apps/snip/coap_app.
* dtls_check_retransmission() called by apps/snip/coap_app.
* dtls_event_thread() called by apps/snip/coap_app.
* dtls_free_peer() called by apps/snip/coap_app.
* dtls_network_receive() called by apps/snip/coap_app.
* dtls_network_send() called by apps/snip/coap_app.
* dtls_server_event_processing_thread() called by apps/snip/coap_app.
* mbedtls_reset_context() called by apps/snip/coap_app.
* mbedtls_ssl_copy_context() called by apps/snip/coap_app.
* mbedtls_timing_get_delay() called by apps/snip/coap_app.
* mbedtls_timing_set_delay() called by apps/snip/coap_app.
* wiced_dtls_calculate_overhead() called by apps/snip/coap_app.
* wiced_dtls_deinit_identity() called by apps/snip/coap_app.
* wiced_dtls_encrypt_packet() called by apps/snip/coap_app.
* wiced_dtls_init_identity() called by apps/snip/coap_app.
* wiced_generic_start_dtls_with_ciphers() called by apps/snip/coap_app.
* besl_supplicant_init() called by apps/snip/rfmon join_ent command.
* supplicant_peap_thread() needs new test application.
* eap_ssl_flush_output() called by apps/snip/tcp_server.
* eap_ssl_receive_packet() called by apps/snip/tcp_server.
* mbedtls_debug() used on mbed TLS, which is changed by WICED.
* mbedtls_reset_context() called by apps/snip/tcp_server.
* my_verify() called by apps/snip/tcp_server.
* ssl_flush_output() called by apps/snip/tcp_server.
* ssl_receive_packet() called by apps/snip/tcp_server.
* tls_add_alpn_extension() called by apps/test/http2.
* tls_calculate_encrypt_buffer_length() called by apps/snip/websocket_server.
* tls_host_create_buffer() used on mbed TLS, which is changed by WICED.
* tls_host_get_packet_data() called by apps/snip/tcp_server.
* tls_host_receive_packet() called by apps/snip/tcp_server.
* tls_packetize_buffered_data() called by apps/snip/tcp_server.
* wiced_generic_start_tls_with_ciphers() called by apps/snip/tcp_server.
* wiced_tls_calculate_overhead() called by apps/snip/tcp_server.
* wiced_tls_close_notify() called by apps/snip/tcp_server.
* wiced_tls_deinit_context() called by apps/snip/https_server.
* wiced_tls_deinit_identity() called by apps/test/http2.
* wiced_tls_deinit_root_ca_certificates() called by apps/test/http2.
* wiced_tls_encrypt_packet() called by apps/snip/tcp_server.
* wiced_tls_init_context() called by apps/snip/httpbin_org.
* wiced_tls_init_identity() called by apps/snip/https_server and apps/snip/websocket_server.
* wiced_tls_init_root_ca_certificates() called by apps/snip/httpbin_org.
* wiced_tls_receive_eap_packet() needs new test application.
* wiced_tls_receive_packet() called by apps/test/http2.
* wiced_tls_reset_context() called by apps/test/http2.
* wiced_tls_set_extension() called by apps/snip/httpbin_org.
* ciph_encrypt()/ciph_init()/md_init()/md_result()/md_update() not used yet.
* wiced_generate_server_websocket_key() called by apps/snip/websocket_server and apps/snip/websocket_client.
* sw_calc_file_sha1() needs new test application.

master-q, 2017-12-15:
Before translate mbed TLS API onto wolfSSL, I think I shoud read https://tls.mbed.org/high-level-design.

cconlon, 2017-12-15:
Thanks for the update

master-q, 2017-12-16:
Next, I should read https://tls.mbed.org/kb/how-to/mbedtls-tutorial.

master-q, 2017-12-18:
Build and run following test application:

https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/blob/master/43xxx_Wi-Fi/apps/test/wolfssl_test/wolfssl_test.c

```
$ pwd
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi
$ ./make.sh test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP download run
$ picocom -b 115200 /dev/ttyUSB1
Crypt Test
error    test passed!
base64   test passed!
base64   test passed!
MD5      test passed!
MD4      test passed!
SHA      test passed!
SHA-224  test passed!
SHA-256  test passed!
SHA-384  test passed!
SHA-512  test passed!
SHA-3    test passed!
Hash     test passed!
HMAC-MD5 test passed!
HMAC-SHA test passed!
HMAC-SHA224 test passed!
HMAC-SHA256 test passed!
HMAC-SHA384 test passed!
HMAC-SHA512 test passed!
Chacha   test passed!
POLY1305 test passed!
ChaCha20-Poly1305 AEAD test passed!
DES      test passed!
DES3     test passed!
AES      test passed!
AES192   test passed!
AES256   test passed!
RANDOM   test failed!
 error = -5004
Crypt Test: Return code -1
```

Why the RANDOM test failed?

master-q, 2017-12-18:
Error code `-5004` means `random_rng_test()` returns a non-zero value.

master-q, 2017-12-18:
There are many `int wc_GenerateSeed(OS_Seed* os, byte* output, word32 sz)`. Which should I use?

master-q, 2017-12-18:
My build option uses `/dev/urandom`, which is not found at WICED platform of course.
Let's search where mbed TLS in WICED gets entropy from.

master-q, 2017-12-18:
Umm... Is it O.K. for production? > Cypress

```c
// WICED/security/BESL/mbedtls_open/include/mbedtls/config.h
--snip--
#define MBEDTLS_ENTROPY_MAX_SOURCES 2

/* extra added by Cypress */
#define MBEDTLS_NO_PLATFORM_ENTROPY
#define MBEDTLS_TEST_NULL_ENTROPY
#define MBEDTLS_ENTROPY_C
#define MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES
--snip--
// WICED/security/BESL/mbedtls_open/library/entropy.c
--snip--
#if defined(MBEDTLS_ENTROPY_C)

#if defined(MBEDTLS_TEST_NULL_ENTROPY)
//#warning "**** WARNING!  MBEDTLS_TEST_NULL_ENTROPY defined! "
//#warning "**** THIS BUILD HAS NO DEFINED ENTROPY SOURCES "
//#warning "**** THIS BUILD IS *NOT* SUITABLE FOR PRODUCTION USE "
#endif
```

master-q, 2017-12-18:
I found an test application creating random numbers with WICED API:

```
$ ./make.sh test.entropy-CYW943907AEVAL1F-FreeRTOS-LwIP download run
$ picocom -b 115200 /dev/ttyUSB1
Starting WICED vWiced_006.000.001.0005
Platform CYW943907AEVAL1F initialised
Started FreeRTOS v9.0.0
Initialising LwIP v2.0.2
DHCP CLIENT hostname WICED IP
WLAN MAC Address : A4:08:EA:D9:BC:A8
WLAN Firmware    : wl0: Oct 23 2017 03:40:42 version 7.15.168.101 (r674438) FWID 01-13cae12
WLAN CLM         : API: 12.2 Data: 9.10.74 Compiler: 1.31.3 ClmImport: 1.36.3 Creation: 2017-10-23 03:36:41 
-------
Default PRNG:
--------

RANDOM TEST (64 bytes)
Entropy = 5.812500 bits per byte.

Optimum compression would reduce the size
of this 64 byte file by 27 percent.

Chi square distribution for 64 samples is 240.00, and randomly
would exceed this value less than 0.01 percent of the times.

Arithmetic mean value of data bytes is 146.5312 (127.5 = random).
Monte Carlo value for Pi is 2.400000000 (error 23.61 percent).
Serial correlation coefficient is 0.083711 (totally uncorrelated = 0.0).
Run succeeded

-------
Secure PRNG:
--------

RANDOM TEST (64 bytes)
Entropy = 5.706955 bits per byte.

Optimum compression would reduce the size
of this 64 byte file by 28 percent.

Chi square distribution for 64 samples is 272.00, and randomly
would exceed this value less than 0.01 percent of the times.

Arithmetic mean value of data bytes is 125.8594 (127.5 = random).
Monte Carlo value for Pi is 3.200000000 (error 1.86 percent).
Serial correlation coefficient is 0.045149 (totally uncorrelated = 0.0).
Run succeeded

Done - press reset on board to continue!
```

master-q, 2017-12-18:
There is `wiced_result_t wiced_crypto_get_random( void* buffer, uint16_t buffer_length )` to get list of random numbers.
The API has two mode. One is regular PRNG. The other is secure PRNG which needs around 6K of heap.

master-q, 2017-12-18:
Added the `wiced_crypto_get_random()` function using WICED API.

But following errors occur. Their sequence:

1. Boot WICED platform.
2. Run test.
3. The base64 test failed.
4. Wait 3 seconds.
5. Run test.
6. Reboot WICED platform after RANDOM test.

```
Starting WICED vWiced_006.000.001.0005
Platform CYW943907AEVAL1F initialised
Started FreeRTOS v9.0.0
Initialising LwIP v2.0.2
DHCP CLIENT hostname WICED IP
WLAN MAC Address : A4:08:EA:D9:BC:A8
WLAN Firmware    : wl0: Oct 23 2017 03:40:42 version 7.15.168.101 (r674438) FWID 01-13cae12
WLAN CLM         : API: 12.2 Data: 9.10.74 Compiler: 1.31.3 ClmImport: 1.36.3 Creation: 2017-10-23 03:36:41 

Crypt Test
error    test passed!
base64   test passed!
RTC/Time not set!
base64   test failed!
 error = -1303
Crypt Test: Return code -1
Sleep 3 seconds to restart...

Crypt Test
error    test passed!
base64   test passed!
base64   test passed!
MD5      test passed!
MD4      test passed!
SHA      test passed!
SHA-224  test passed!
SHA-256  test passed!
SHA-384  test passed!
SHA-512  test passed!
SHA-3    test passed!
Hash     test passed!
HMAC-MD5 test passed!
HMAC-SHA test passed!
HMAC-SHA224 test passed!
HMAC-SHA256 test passed!
HMAC-SHA384 test passed!
HMAC-SHA512 test passed!
Chacha   test passed!
POLY1305 test passed!
ChaCha20-Poly1305 AEAD test passed!
DES      test passed!
DES3     test passed!
AES      test passed!
AES192   test passed!
AES256   test passed!
RANDOM   test passed!

Starting WICED vWiced_006.000.001.0005
Platform CYW943907AEVAL1F initialised
Started FreeRTOS v9.0.0
Initialising LwIP v2.0.2
DHCP CLIENT hostname WICED IP
WLAN MAC Address : A4:08:EA:D9:BC:A8
WLAN Firmware    : wl0: Oct 23 2017 03:40:42 version 7.15.168.101 (r674438) FWID 01-13cae12
WLAN CLM         : API: 12.2 Data: 9.10.74 Compiler: 1.31.3 ClmImport: 1.36.3 Creation: 2017-10-23 03:36:41 

Crypt Test
error    test passed!
base64   test passed!
RTC/Time not set!
base64   test failed!
 error = -1303
Crypt Test: Return code -1
Sleep 3 seconds to restart...
```

master-q, 2017-12-19:
Following two define macros are needed to fix `RTC/Time not set!` error:

```c
#elif defined(MICROCHIP_TCPIP_V5) || defined(MICROCHIP_TCPIP)
    #include <time.h>
    #define XTIME(t1)       pic32_time((t1))
    #define XGMTIME(c, t)   gmtime((c))
--snip--
time_t pic32_time(time_t* timer)
--snip--
struct tm* gmtime(const time_t* timer)
```

master-q, 2017-12-19:
I believe we can't implement `XTIME()` which returns Y2K...

> The RTC block has been deprecated from this datasheet in revision *A and later. This block is used by Cypress for internal testing/validation/verification and is not intended for customers to use.

http://www.cypress.com/documentation/datasheets/cyw43907-wiced-ieee-80211-abgn-soc-embedded-applications-processor

master-q, 2017-12-19:
I think a data abort exception at RSA test:

```
$ ./make.sh test.wolfssl_test-CYW943907AEVAL1F-FreeRTOS-LwIP-debug download run
$ picocom -b 115200 /dev/ttyUSB1
--snip--
AES192   test passed!
AES256   test passed!
RANDOM   test passed!
=== EXCEPTION ===
data_abort_handler
DFSR : 0x00001C06
DFAR : 0x00000000
IFSR : 0x00000000
IFAR : 0x00000000
CPSR : 0x00070197
R0   : 0x00000000
R1   : 0x00000020
R2   : 0xFFFFFFFF
R3   : 0x00000000
R4   : 0x005376CC
R5   : 0x005014B0
R6   : 0x06060606
R7   : 0x0004DEBD
R8   : 0x08080808
R9   : 0x09090909
R10  : 0x10101010
R11  : 0x11111111
R12  : 0x005089AC
LR   : 0x004DE9C4
=================
```

cconlon, 2017-12-19:
Thanks for the updates.

We have several ways to plug an entropy source into wolfSSL.

1)  Write a new GenerateSeed() function specific to the WICED SDK, in ./wolfcrypt/src/random.c.  This function should fill the "output" buffer with "sz" random bytes.  This is the most common method we use for new platform ports.

2)  Define CUSTOM_RAND_GENERATE to a custom random function that matches the following prototype:

```
word32 rand_gen(void);
```
If the function mapped to CUSTOM_RAND_GENERATE returns a type different than a word32, CUSTOM_RAND_TYPE should be defined to that data type.

> I believe we can't implement XTIME() which returns Y2K...

A new code block should be added in ./wolfssl/wolfcrypt/wc_port.h which defines XTIME to a function in the WICED SDK that returns the number of seconds since the Unix epoch.  Other platform examples of how time was handled can be seen in wc_port.h.  If this is not possible, we will need to force the application to define XTIME in their own <user_settings.h> file.  Our UTasker port does this, if you look in ./wolfssl/wolfcrypt/settings.h:

```
/* user needs to define XTIME to function that provides
 * seconds since Unix epoch */
#ifndef XTIME
    #error XTIME must be defined in wolfSSL settings.h
    /* #define XTIME fnSecondsSinceEpoch */
#endif
```

> I think a data abort exception at RSA test:

The most common reasons for a crash in RSA are:

1.  The stack size of the application is not set high enough.

2.  The application has been compiled with different preprocessor defines than the wolfSSL library / sources have been.

master-q, 2017-12-20:
> 1. Write a new GenerateSeed() function specific to the WICED SDK, in ./wolfcrypt/src/random.c. This function should fill the "output" buffer with "sz" random bytes. This is the most common method we use for new platform ports.

I choose this method, which is following:

https://github.com/master-q/wolfssl/commit/9e6a2744813c36d842c4ea96867060e2a491859f

master-q, 2017-12-20:
>  If this is not possible, we will need to force the application to define XTIME in their own <user_settings.h> file. Our UTasker port does this, if you look in ./wolfssl/wolfcrypt/settings.h:

Sorry. I can't understand it.

The uTasker configuration needs XTIME macro out of wolfSSL library. However of course, such XTIME can't implemented, because there are no RTC on hardware...

master-q, 2017-12-20:
> Sorry. I can't understand it.
> 
> The uTasker configuration needs XTIME macro out of wolfSSL library. However of course, such XTIME can't be implemented, because there are no RTC on hardware...

Or should I the UTC time from NTP?

master-q, 2017-12-20:
@cconlon

> Sorry. I can't understand it.
> 
> The uTasker configuration needs XTIME macro out of wolfSSL library. However of course, such XTIME can't be implemented, because there are no RTC on hardware...

I have an idea from Kojo. I'll try to implement pseudo Unix epoch time.

master-q, 2017-12-22:
https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/commit/7a77e27e4261969587511f6958ff2ef656d480d5

Pass all of tests with 32kB stack.

master-q, 2017-12-22:
> I have an idea from Kojo. I'll try to implement pseudo Unix epoch time.

There are some plans:

1. Calculate Unix epoch time using the time value at building environment.
2. Calculate Unix epoch time using a constant time value.

1 is better than 2. But 1 needs portable Unix epoch time function/command between Windows/MacOS/Linux. As first step, I should choose 2.

master-q, 2017-12-22:
The Unix epoch time can be gotten by following command:

```shell
$ date +'%s'
```

master-q, 2017-12-22:
> 1. Calculate Unix epoch time using the time value at building environment.

I found better way for 1 with Perl.

https://github.com/master-q/wolfssl/commit/1c3facf0bf9780d2c0a9908457b817378c045076

master-q, 2017-12-25:
Publish first version of HTTPS client: https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/commit/c663ff228fefcf49fb5791ac6650d106a757e54a which includes Cypress Copyright. Because I copied many code from original WICED source code.

master-q, 2017-12-27:
Introduced a make target to switch wolfSSL and mbed TLS:

https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/commit/8edd5121edc669d5a18262149887869ad2f45053

master-q, 2017-12-29:
Umm... WICED strongly customizes mbed TLS...

```
$ git grep -i wiced WICED/security/BESL/mbedtls_open/ | wc -l
101
```

master-q, 2018-01-04:
I believe `aes_decrypt_ccm()` function is hidden in Cypress binary:

```
$ pwd
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi
$ ./make.sh clean && ./make.sh QUIET= snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP
$ vi WICED/internal/wiced_cooee.c
--snip--
    /* Decrypt the data */
    {
        uint8_t* cooee_key;
        wiced_dct_read_lock( (void**) &cooee_key, WICED_FALSE, DCT_SECURITY_SECTION, offsetof(platform_dct_security_t, cooee_key), COOEE_KEY_SIZE );

        mbedtls_aes_setkey_dec( &aes_ctx, cooee_key, 128 );

        wiced_dct_read_unlock( cooee_key, WICED_FALSE );
    }

    if (aes_decrypt_ccm( &aes_ctx, content_length, 10, cooee_nonce, sizeof(cooee_nonce),(uint8_t*) cooee_header, &workspace->received_cooee_data[10], &workspace->received_cooee_data[10] ) != 0)
    {
        WPRINT_WICED_INFO( ("Cooee payload decryption failed\n") );
        goto return_with_error;
    }
$ vi WICED/security/BESL/crypto_internal/wiced_security_internal.h
--snip--
/**
 * @brief                   AES-CCM decryption
 *
 * @param[in]  ctx              : AES context
 * @param[in]  length           : Length of the input data
 * @param[in]  aad_length       : Length of the additional associated data
 * @param[in]  nonce            : The nonce to use
 * @param[in]  nonce_length     : Length of nonce.
 * @param[in]  aad_input        : The buffer containing the additional associated data
 * @param[in]  ciphertext_input : Buffer holding the input data
 * @param[out] plaintext_output : Buffer which receives the output plaintext
 */
int aes_decrypt_ccm( mbedtls_aes_context *ctx, uint32_t length, uint32_t aad_length,  const unsigned char *nonce, uint8_t nonce_length, const unsigned char *aad_input, const unsigned char *ciphertext_input, unsigned char *plaintext_output );
$ grep aes_decrypt_ccm ./build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP/binary/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP.map
 .text.aes_decrypt_ccm
aes_decrypt_ccm                                   ././WICED/security/BESL/BESL_generic.ARM_CR4.release.a(aes-ctr-ccm.o)
```

Above means the `aes_decrypt_ccm` is derived from the `BESL_generic.ARM_CR4.release.a` file. And source codes of the `BESL_generic.ARM_CR4.release.a` is not published at the internet.

Tentatively, I choose a way to overwrite above internal API using wolfSSL API.

master-q, 2018-01-04:
There is a semantic gap between `aes_decrypt_ccm` and `wc_AesCcmDecrypt`:

```c
 WOLFSSL_API int  wc_AesCcmDecrypt(Aes* aes, byte* out,
                                   const byte* in, word32 inSz,
                                   const byte* nonce, word32 nonceSz,
                                   const byte* authTag, word32 authTagSz,
                                   const byte* authIn, word32 authInSz);
```

What does `aad_input` means? Is it equivalent to `authTag`?

master-q, 2018-01-04:
I believe `struct WOLFSSL_SESSION` is equivalent to `mbedtls_ssl_session`.

master-q, 2018-01-04:
I think `WOLFSSL_X509_CHAIN` is equivalent to `mbedtls_x509_crt`.

master-q, 2018-01-06:
After my long reading code, I believe we should create `WOLFSSL_CTX` before reading PEM and key, to store them on the structure.
It means we should change following structure in original WICED code:

```c
typedef struct
{
    union
    {
        wiced_tls_psk_key_t psk_key;
        mbedtls_pk_context  private_key;
    };
    mbedtls_x509_crt        certificate;
    wiced_tls_sign_certificate_verify custom_sign;
} wiced_tls_identity_t;
```

as following:

```c
typedef struct
{
    WOLFSSL_CTX* ctx;
    wiced_tls_sign_certificate_verify custom_sign; // How to use with wolfSSL?
} wiced_tls_identity_t;
```

It seems like a bit strange name. Why do we call `WOLFSSL_CTX` as `identity`? It's caused by following initialization flow at WICED:

```
typedef struct mbedtls_ssl_context  wiced_tls_workspace_t;
typedef struct mbedtls_ssl_session  wiced_tls_session_t;
typedef struct mbedtls_x509_crt wiced_tls_certificate_t;

typedef int (*wiced_tls_sign_certificate_verify)(  void* key ,rsa_hash_id_t hash_id, int32_t hashlen, const unsigned char *hash, unsigned char *sign, uint32_t* key_length, wiced_tls_key_type_t type );

typedef struct
{
    union
    {
        wiced_tls_psk_key_t psk_key;
        mbedtls_pk_context  private_key;
    };
    mbedtls_x509_crt        certificate;
    wiced_tls_sign_certificate_verify custom_sign;
} wiced_tls_identity_t;

typedef struct
{
    uint32_t                 context_id;
    wiced_tls_workspace_t    context;
    wiced_tls_session_t*     session;
    wiced_tls_identity_t*    identity;
} wiced_tls_context_t;

wiced_https_get()
=> wiced_tls_context_t *context;
   wiced_tls_identity_t *tls_identity = NULL;
   tls_identity = ( wiced_tls_identity_t* ) malloc_named ("tls_identity", sizeof( wiced_tls_identity_t ));
   wiced_tls_init_identity( tls_identity, dct_security->private_key, strlen( dct_security->private_key ), (uint8_t*) dct_security->certificate, strlen( dct_security->certificate ) );
   => mbedtls_x509_crt_init( &identity->certificate );
      mbedtls_x509_crt_parse( &identity->certificate, (const unsigned char *) certificate_data, certificate_length );
      mbedtls_pk_init( &identity->private_key );
      mbedtls_pk_parse_key( &identity->private_key, (const unsigned char *) private_key, key_length, NULL, 0 );
   wiced_tcp_create_socket( &socket, WICED_STA_INTERFACE ) ) != WICED_SUCCESS );
   context = MALLOC_OBJECT( "HTTP Client TLS Context", wiced_tls_context_t );
   socket.context_malloced = WICED_TRUE;
   wiced_tls_init_context( context, tls_identity, peer_cn );
   => context->context.conf = tls_host_malloc ("tls", sizeof(mbedtls_ssl_config));
      context->context_id = WICED_TLS_CONTEXT_ID;
      context->identity = identity;
      ((mbedtls_ssl_config*)context->context.conf)->peer_cn = peer_cn;
   wiced_tcp_enable_tls( &socket, context );
   => socket->tls_context = context;
   wiced_tcp_connect( &socket, address, 443, 20000 );
   => wiced_tcp_start_tls( socket, WICED_TLS_AS_CLIENT, WICED_TLS_DEFAULT_VERIFICATION );
      => wiced_generic_start_tls_with_ciphers( socket->tls_context, socket, type, verification, NULL, TLS_TCP_TRANSPORT );
         => mbedtls_ssl_config* conf = (mbedtls_ssl_config*) tls_context->context.conf;
```

As above, WICED call `mbedtls_x509_crt_parse` and `mbedtls_pk_parse_key` before calling `wiced_tls_init_context`.

Totally, breaking the name convention is better than destroying WICED external API.

cconlon, 2018-01-09:
> What does aad_input means? Is it equivalent to authTag?

My guess is that "aad_input" is referring to AES-CCM's optional additional authenticated data (AAD).  For wc_AesCcmDecrypt() this would be authIn.

> After my long reading code, I believe we should create WOLFSSL_CTX before reading PEM and key, to store them on the structure.

A user needs to create a WOLFSSL_CTX with wolfSSL_CTX_new( ) before any keys or certificates (or other CTX-level settings can be loaded into the context.

> It seems like a bit strange name. Why do we call WOLFSSL_CTX as identity?

When I read your code section above, it looks like WICED calls that structure an "identity" because it is representing the identity of the client.  That identity looks like it could be represented through a PSK (pre-shared key) or X.509 certificate.

It seems like mbedTLS first loads a certificate and key into separate structures - mbedtls_x509_crt for the certificate, and either wiced_tls_psk_key_t (in the case PSK's are being used) or mbedtls_pk_context (for a normal RSA/ECC key).  wolfSSL doesn't require users to create individual structures first.  Users/apps just load a certificate file into a WOLFSSL_CTX with wolfSSL_CTX_use_certificate_file() or wolfSSL_CTX_use_certificate_buffer() and a private key into the WOLFSSL_CTX with wolfSSL_CTX_use_PrivateKey_file() or wolfSSL_CTX_use_PrivateKey_buffer().  The wolfSSL examples that ship with our download are good examples of doing this.  You can also look in our examples repository: https://github.com/wolfSSL/wolfssl-examples/tree/master/tls

In the wolfSSL case, wiced_tls_init_identity() should probably call wolfSSL_CTX_new(), wolfSSL_CTX_use_certificate_file/wolfSSL_CTX_use_certificate_buffer() and wolfSSL_CTX_use_PrivateKey_file/wolfSSL_CTX_use_PrivateKey_buffer().

master-q, 2018-01-12:
> My guess is that "aad_input" is referring to AES-CCM's optional additional authenticated data (AAD). For wc_AesCcmDecrypt() this would be authIn.

I think you are right. Thanks a lot.

```c
/**
 * @brief                   AES-CCM encryption
 *
 * @param[in]  ctx               : AES context
 * @param[in]  length            : Length of the input data
 * @param[in]  aad_length        : Length of the additional associated data
 * @param[in]  nonce             : The nonce to use
 * @param[in]  nonce_length      : Length of nonce.
 * @param[in]  aad_input         : The buffer containing the additional associated data
 * @param[in]  plaintext_input   : Buffer holding the input data
 * @param[out] ciphertext_output : Buffer which receives the output ciphertext
 * @param[out] mac_output        : Buffer which recieves the output MAC
 */
int aes_encrypt_ccm( mbedtls_aes_context *ctx, uint32_t length, uint32_t aad_length, const unsigned char *nonce, uint8_t nonce_length, const unsigned char *aad_input, const unsigned char *plaintext_input, unsigned char *ciphertext_output, unsigned char mac_output[8] );
```

master-q, 2018-01-12:
> A user needs to create a WOLFSSL_CTX with wolfSSL_CTX_new( ) before any keys or certificates (or other CTX-level settings can be loaded into the context.

Umm... It means we should cache the keys and certificates into `wiced_tls_identity_t`.

master-q, 2018-01-12:
> When I read your code section above, it looks like WICED calls that structure an "identity" because it is representing the identity of the client. That identity looks like it could be represented through a PSK (pre-shared key) or X.509 certificate.

Yes.

> It seems like mbedTLS first loads a certificate and key into separate structures - mbedtls_x509_crt for the certificate, and either wiced_tls_psk_key_t (in the case PSK's are being used) or [...]

I understand above, now.

> In the wolfSSL case, wiced_tls_init_identity() should probably call wolfSSL_CTX_new(), wolfSSL_CTX_use_certificate_file/wolfSSL_CTX_use_certificate_buffer() and wolfSSL_CTX_use_PrivateKey_file/wolfSSL_CTX_use_PrivateKey_buffer().

I think it means:

```c
typedef struct
{
    WOLFSSL_CTX*                      ctx;
    wiced_tls_sign_certificate_verify custom_sign; // Not used
} wiced_tls_identity_t;

typedef struct
{
    uint32_t                 context_id;
    WOLFSSL*                 ssl;
    wiced_tls_identity_t*    identity;
} wiced_tls_context_t;
```

master-q, 2018-01-15:
We can compile snip.https_client application with wolfSSL using following patch:

https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL/commit/07b8a388574ad443ee658ee123d3f067b7d80fdc

But the above application causes some errors:

```
$ sudo picocom -b 115200 /dev/ttyUSB1
Starting WICED vWiced_006.000.001.0005
Platform CYW943907AEVAL1F initialised
Started FreeRTOS v9.0.0
Initialising LwIP v2.0.2
DHCP CLIENT hostname WICED IP
WLAN MAC Address : A4:08:EA:D9:BC:A8
WLAN Firmware    : wl0: Oct 23 2017 03:40:42 version 7.15.168.101 (r674438) FWID 01-13cae12
WLAN CLM         : API: 12.2 Data: 9.10.74 Compiler: 1.31.3 ClmImport: 1.36.3 Creation: 2017-10-23 03:36:41 
Connect to the network
Joining : foo-bar
Successfully joined : foo-bar
Setting IPv6 link-local address
Obtaining IPv4 address via DHCP
IPv6 Network ready IP: XXXX::XXXX:XXXX:XXXX:XXXX
IPv4 Network ready IP: XXX.XXX.XXX.XXX
Resolving IP address of https server...
Resolved Server IP: XXX.XXX.XXX.XXX

Read the certificate Key from DCT
xxx TODO: support TLS on wolfSSL at WICED/security/BESL/host/WICED/wiced_tls.c, line 1302.
xxx TODO: support TLS on wolfSSL at WICED/security/BESL/host/WICED/wiced_tls.c, line 1314.
xxx TODO: support TLS on wolfSSL at WICED/security/BESL/host/WICED/wiced_tls.c, line 1343.

http_get: Get failed: 4
```

because it has many unimplemented functions with wolfSSL:

```
$ pwd
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi
$ git grep "xxx TODO: support"
WICED/WICED.mk:# xxx TODO: support COOEE with wolfSSL
WICED/network/LwIP/WICED/tcpip.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/LwIP/WICED/tcpip.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/LwIP/WICED/tcpip.c:           printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/LwIP/WICED/tcpip.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/LwIP/WICED/tcpip.c:        printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/LwIP/WICED/wiced_network.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/wiced_tcpip_common.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/wiced_tcpip_common.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/wiced_tcpip_common.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/wiced_tcpip_common.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/network/wiced_tcpip_common.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/network/wiced_tcpip_common.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/security/BESL/BESL.mk:# xxx TODO: support TLS with wolfSSL
WICED/security/BESL/host/WICED/dtls_types.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/dtls_types.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/dtls_types.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:        printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
[... 20 lines cut ...]
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/BESL/host/WICED/wiced_tls.c:    printf("xxx TODO: support TLS on wolfSSL at %s, line %d.\n", __FILE__, __LINE__);
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
[... cut ...]
WICED/security/BESL/host/WICED/wiced_tls.c:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
WICED/security/PostgreSQL/PostgreSQL.mk:# xxx TODO: support Fortuna with wolfSSL
apps/test/entropy/entropy.mk:# xxx TODO: support Fortuna on wolfSSL
include/wiced_tcpip.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
include/wiced_tls.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
include/wiced_tls.h:#ifdef WICED_MBEDTLS /* xxx TODO: support TLS on wolfSSL */
```

We will get the application running after implementing above codes.

master-q, 2018-01-16:
I get `Memory exhausted` error when calling `wolfSSL_connect()` function from WICED.

```
$ ./make clean && ./make snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl download run
--snip--
Making build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/libraries/USB_Device_43909_Library_CYW943907AEVAL1F.a
Making snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl.elf
/home/kiwamu/Documents/WICED-Studio-6.0/43xxx_Wi-Fi/tools/ARM_GNU/bin/Linux64/arm-none-eabi-ld: final link failed: Memory exhausted
make[1]: *** [build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/binary/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl.elf] Error 1
```

wolfSSL is so large for WICED platform?

master-q, 2018-01-16:
Without calling `wolfSSL_connect()`, Makefile show following:

```
----------------------------------|---------|---------|
                                  |         |  Static |
              Module              |  Flash  |   RAM   |
----------------------------------+---------+---------|
App                               |       0 |    1408 |
crc                               |       0 |    1060 |
DHCP_Server                       |       0 |    1566 |
DNS                               |       0 |    1780 |
FreeRTOS                          |       0 |    6625 |
HTTP_client                       |       0 |     659 |
Interrupt Vectors                 |       0 |     288 |
libc                              |       0 |   28902 |
LwIP                              |       0 |   51142 |
Networking                        |       0 |    4876 |
NVRam                             |       0 |    2210 |
Other                             |       0 |   57468 |
Packet Buffers                    |       0 |  110892 |
platform                          |       0 |     470 |
RAM Initialisation                |      32 |       0 |
resources                         |       0 |      32 |
Ring_Buffer                       |       0 |     140 |
Startup Stack & Link Script fill  |       0 |     272 |
Supplicant - BESL                 |       0 |     500 |
WICED                             |       0 |    4902 |
Wiced_RO_FS                       |       0 |     568 |
WWD                               |       0 |    1184 |
----------------------------------+---------+---------|
TOTAL (bytes)                     |       0 |  276944 |
----------------------------------|---------|---------|
```

master-q, 2018-01-16:
```perl
sub process_section( $area, $section, $size, $modulefn, $decaddress, $hexaddress )
{
# --snip--
    elsif ( ( $section =~ m/.*tx_buffer_pool_memory/ ) ||
         ( $section =~ m/.*rx_buffer_pool_memory/ ) ||
         ( $section =~ m/.*buffer_pool_memory_common/ ) ||
         ( $section =~ m/.*buffer_pool_memory_128/ ) ||
         ( $section =~ m/.*memp_memory/ ) )
    {
        $module = "Packet Buffers";
    }
```

master-q, 2018-01-16:
Umm...

```shell
$ size `find build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/ -name "*.o"`| grep -v "  text" | sort -n
      0       0       0       0       0 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/cpuid.o
      0       0       0       0       0 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/dh.o
     14       0       0      14       e build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/logging.o
    138       0       0     138      8a build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/wolfmath.o
    140       0      12     152      98 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/memory.o
    172       0       4     176      b0 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/wc_port.o
    226       0       0     226      e2 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/src/wolfio.o
    396       0       0     396     18c build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/wc_encrypt.o
    566       0       0     566     236 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/chacha20_poly1305.o
    688       0       0     688     2b0 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/signature.o
    733       0       0     733     2dd build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/hash.o
    842       0       0     842     34a build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/chacha.o
   1216       0       0    1216     4c0 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/hmac.o
   1262       0       0    1262     4ee build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/coding.o
   1422       0       0    1422     58e build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/md4.o
   1494       0       0    1494     5d6 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/poly1305.o
   1950       0       0    1950     79e build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/random.o
   2124       0       0    2124     84c build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/md5.o
   2128       0       0    2128     850 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/sha256.o
   3404       0       0    3404     d4c build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/src/keys.o
   3478       0       0    3478     d96 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/rsa.o
   3884       0       0    3884     f2c build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/des3.o
   3982       0       0    3982     f8e build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/sha.o
   4942       0       0    4942    134e build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/error.o
   5238       0       0    5238    1476 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/src/tls.o
   5620       0       0    5620    15f4 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/sha3.o
   9321       0       0    9321    2469 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/tfm.o
   9594       0   12860   22454    57b6 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/src/ssl.o
  12278      96       0   12374    3056 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/asn.o
  12538       0       0   12538    30fa build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/sha512.o
  12944       0       0   12944    3290 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/aes.o
  15110       0       0   15110    3b06 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/src/ecc.o
  33491       0       0   33491    82d3 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/wolfcrypt/test/test.o
  39240       0       0   39240    9948 build/snip.https_client-CYW943907AEVAL1F-FreeRTOS-LwIP-wolfssl/Modules/WICED/security/BESL/wolfssl/src/internal.o
```

master-q, 2018-01-16:
I found following forum:

https://www.wolfssl.com/forums/topic813-ld-link-failed-memory-exhausted.html

The `Memory exhausted` error is fixed by `-g1` option on my board.

master-q, 2018-01-17:
Ah, following APIs have semantic gap:

```c
wiced_result_t network_tcp_receive( wiced_tcp_socket_t* socket, wiced_packet_t** packet, uint32_t timeout );
static int ssl_receive_packet( WOLFSSL* ssl, char* buf, int sz, void* ctx );

void foobar(void) {
    wolfSSL_SetIORecv( tls_context->context, ssl_receive_packet );
}
```

The `network_tcp_receive()` on WICED allocates with flexible size. But the `ssl_receive_packet()` handler on wolfSSL needs predefined size to be called.
How to know size of the packet before receive from network?

master-q, 2018-01-17:
Umm... The `ssl_receive_packet()` should return just required size...

```c
static int Receive(WOLFSSL* ssl, byte* buf, word32 sz)
{
// --snip--
    recvd = ssl->ctx->CBIORecv(ssl, (char *)buf, (int)sz, ssl->IOCB_ReadCtx)
// --snip--
    return recvd;
}

static int GetInputData(WOLFSSL *ssl, word32 size)
{
// --snip--
        in = Receive(ssl,
                     ssl->buffers.inputBuffer.buffer +
                     ssl->buffers.inputBuffer.length,
                     inSz);
        if (in == -1)
            return SOCKET_ERROR_E;

        if (in == WANT_READ)
            return WANT_READ; // SUCCESS

        if (in > inSz)
            return RECV_OVERFLOW_E; // ERROR!
```

master-q, 2018-01-17:
It means that `ssl_receive_packet()` should split packet into required size and remaining one?

master-q, 2018-01-19:
Maintaining fragments of the packet is complex. How about don't maintain the fragments at first step?

master-q, 2018-01-25:
Unfortunately, I can't continue this issue no more.

You can re-start it with following:

```
$ rm -rf /home/User/Documents/WICED-Studio-6.0
$ cd /home/User/Documents
$ git clone https://gitlab.com/masterq/WICED-Studio-6.0-wolfSSL.git WICED-Studio-6.0
$ cd WICED-Studio-6.0/43xxx_Wi-Fi/WICED/security/BESL
$ git clone -b feature-wiced-studio-6.0 https://github.com/master-q/wolfssl.git
$ cd /home/User/Documents/WICED-Studio-6.0/43xxx_Wi-Fi
$ vi WICED/security/BESL/host/WICED/wiced_tls.c
 803 #elif defined WICED_WOLFSSL
 804 static int ssl_receive_packet( WOLFSSL* ssl, char* buf, int sz, void* ctx )
 805 {
 806     wiced_tcp_socket_t* socket = ( wiced_tcp_socket_t* ) ctx;
 807     UNUSED_PARAMETER(socket);
 808 
 809     printf("xxx TODO: implement ssl_receive_packet() (sz=%d)\n", sz);
 810 
 811     return -1;
 812 }
 813 #endif
```

[1 more reply]

closed 2018-03-16 by master-q