wolfSSL/wolfssl #1273: Can wolfssl host multiple virtual servers by SNI?

WuBingzheng, 2017-12-17:
Hi,

It seems that it can not, from https://www.wolfssl.com/using-server-name-indication-sni-with-wolfssl/ .

So I can not use wolfssl as the SSL engine in a server which must support virtual-server, can I?

Is there any plan to support it?

Thanks,
Wu

dgarske, 2017-12-20:
Hi WuBingzheng,

Actually we do fully support using SNI on the server side. A typical use for the server is to peek at the socket bytes and then call `wolfSSL_SNI_GetFromBuffer`. That function parses the client_hello and extracts the SNI host name from the TLS extension. This then allows you (as the server) to decide which website / certificates you will present.

The blog post states "the wolfSSL server doesn`t host multiple `virtual` servers". This is meant to say we do not have a TLS HTTPS virtual server to demonstrate this, but we do have all the functionality for you to implement this in your own application.

You can find an example for using SNI on the server side to extract the host name here:
https://github.com/wolfSSL/wolfssl/blob/master/src/sniffer.c#L1823

```
int wolfSSL_SNI_GetFromBuffer(const byte* clientHello, word32 helloSz, byte type, byte* sni, word32* inOutSz);
```

For the client side the usage is:

https://github.com/wolfSSL/wolfssl/blob/master/examples/client/client.c#L1678

```
wolfSSL_CTX_UseSNI(ctx, 0, sniHostName, (word16)XSTRLEN(sniHostName));
```

When building the library make sure you use `./configure --enable-sni` OR define `HAVE_SNI` and `HAVE_TLS_EXTENSIONS`.

Thanks,
David Garske, wolfSSL

WuBingzheng, 2017-12-22:
Got it. Thanks.

This API is a little different with others. It does not get a `ssl` parameter, but `clientHello` byte.

closed 2017-12-22 by WuBingzheng

Nax10823, 2021-08-28:
For those that stumbled upon this thread, and don't know how to get `clientHello` for `wolfSSL_SNI_GetFromBuffer` with a connected sockfd, see the following [post](https://www.wolfssl.com/forums/topic1293-server-sni-wolfsslsnigetfrombuffer.html) on the product support forum.