wolfSSL/wolfssl #1295: make check failed (broken pipe)

tx-wolfssl, 2018-01-02:
wolfssl 3.13.0; CentOS 5 (autogen done on CentOS 6)

```
ALL Tests Passed
PASS: scripts/psk.test
WOLFSSL_EXTERNAL_TEST NOT set, won't run
PASS: scripts/external.test
PING www.google.com (108.177.119.99) 56(84) bytes of data.
64 bytes from 108.177.119.99: icmp_seq=1 ttl=47 time=16.8 ms
64 bytes from 108.177.119.99: icmp_seq=2 ttl=47 time=13.6 ms

--- www.google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1003ms
rtt min/avg/max/mdev = 13.685/15.266/16.847/1.581 ms
external test can't be run in this modePASS: scripts/google.test
tr: write error: Broken pipe
tr: write error
tr: write error: Broken pipe
tr: write error
WOLFSSL_OPENSSL_TEST NOT set, won't run
PASS: scripts/openssl.test
==========================================================
1 of 9 tests failed
Please report to https://github.com/wolfssl/wolfssl/issues
==========================================================
make[2]: *** [check-TESTS] Error 1
make[2]: Leaving directory `/usr/src/redhat/BUILD/wolfssl-3.13.0-stable'
make[1]: *** [check-am] Error 2
make[1]: Leaving directory `/usr/src/redhat/BUILD/wolfssl-3.13.0-stable'
make: *** [check] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.3576 (%check)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.3576 (%check)
```

Which additional details do you need?

JacobBarthelmeh, 2018-01-03:
Hi tx-wolfssl,

- Just to clarify, what is the default configuration being tested? Is it still with setting the build, host, and target?

- I'm not sure if this would cause the issue being seen but with cross compiling a library the host and target should be the same.

- What endineaness is the system being built for (Big or Little)? 

- Does the tests pass with defining the macro TIME_T_NOT_LONG? 

- And if possible may we get the what the variable date_str holds right before the comparison on line 10685 of tests/api.c.

Kind Regards,
Jacob

tx-wolfssl, 2018-01-03:
Hi Jacob,

1.) Host and target are the same. Build system is a stock CentOS 5 on i386. Build is done via rpmbuild. Excerpt of log:

```
+ ./configure --build=i686-redhat-linux-gnu --host=i686-redhat-linux-gnu --target=i386-redhat-linux-gnu --program-prefix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --datadir=/usr/share --includedir=/usr/include --libdir=/usr/lib --libexecdir=/usr/libexec --localstatedir=/var --sharedstatedir=/usr/com --mandir=/usr/share/man --infodir=/usr/share/info --disable-static --enable-shared --enable-psk --enable-opensslextra --enable-fastmath --enable-bigcache --enable-aesgcm --enable-aesccm --enable-camellia --enable-ripemd --enable-blake2 --enable-sha512 --enable-sessioncerts --enable-keygen --enable-certgen --enable-certreq --enable-certext --enable-ecc --enable-curve25519 --enable-ed25519 --enable-eccencrypt --enable-oldtls --enable-rsa --enable-dh --enable-asn --enable-aes --disable-md5 --disable-sha --enable-poly1305 --enable-chacha --enable-hashdrbg --enable-filesystem --enable-inline --enable-ocsp --enable-sni --enable-supportedcurves --enable-session-ticket --enable-examples --enable-savesession --enable-hkdf
[...]
Configuration summary for wolfssl version 3.13.0

   * Installation prefix:        /usr
   * System type:                redhat-linux-gnu
   * Host CPU:                   i686
   * C Compiler:                 gcc
   * C Flags:                    -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m32 -march=i386 -mtune=generic -fasynchronous-unwind-tables  -Wno-pragmas -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wswitch-enum -Wundef -Wunused -Wunused-variable -Wwrite-strings -fwrapv -DTFM_NO_ASM
   * C++ Compiler:               g++
   * C++ Flags:                  -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m32 -march=i386 -mtune=generic -fasynchronous-unwind-tables
   * CPP Flags:                   -fvisibility=hidden
   * CCAS Flags:                 -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=4 -m32 -march=i386 -mtune=generic -fasynchronous-unwind-tables
   * LIB Flags:                   -pie -z relro -z now 
   * Debug enabled:              no
   * Coverage enabled:           
   * Warnings as failure:        no
   * make -j:                    2
   * VCS checkout:               no
```

2.) No crosscompile :-)

3.) i386/LE

4.) Yes, with defining TIME_T_NOT_LONG the rpm builds successfully even with my own configuration.

5.) Sure, here it is:

```
   wolfSSL_BIO(): passed
   wolfSSL_ASN1_TIME_adj():line 10685: date_str=XXX19190205024144ZXXX

ERROR - tests/api.c line 10686 failed with:
    expected: 0 == memcmp((date_str),("20550313091000Z"),(15))
    result:   0 != -1
```

sincerely,
Frank

JacobBarthelmeh, 2018-01-03:
Hi Frank,

Thanks, that info helps. I can reproduce the same value of date_str if forcing time_t to be an int type. Currently what TIME_T_NOT_LONG is doing is just compiling out tests that would cause an overflow if time_t is of type int. For the time being the macro TIME_T_NOT_LONG can be used when compiling wolfSSL in this case, and I'll bring up adding a fix for automatically detecting that time_t is an int instead of long on CentOS 5 internally.

Regards,
Jacob

tx-wolfssl, 2018-01-03:
Great! Thank you!

closed 2018-05-24 by JacobBarthelmeh