wolfSSL/wolfssl #1335: How to use wolfssl for haproxy

saga1015, 2018-01-23:
Hi, when i use wolfssl for haproxy 1.7.10 ,I  use 
make TARGET=linux26 \
USE_PCRE=1 \
USE_PCRE_JIT=1 \
USE_OPENSSL=1 \
USE_POLL="" \
USE_PRIVATE_CACHE=1 \
USE_THREAD=1 \
USE_PTHREAD_PSHARED=1 \
USE_REGPARM=1 \
USE_VSYSCALL=1 \
CPU=native \
USE_LINUX_TPROXY="" \
USE_NETFILTER="" \
USE_TPROXY="" \
USE_TFO=1 \
SSL_INC="/opt/websuite/wolfssl/include/wolfssl/openssl" \
SSL_LIB="/opt/websuite/wolfssl/lib" \
PCRE_LIB="/opt/websuite/common/lib" \
PCRE_INC="/opt/websuite/common/include" \
ADDLIB=-ldl  \
USE_CPU_AFFINITY=1 \
USE_LIBCRYPT="" \
TARGET_CFLAGS="-m64 -m128bit-long-double" 

and modify haproxy Makefile  -lssl --> -lwolfssl ,but show me info:

src/ssl_sock.o: In function `smp_fetch_ssl_fc_is_resumed':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4811: undefined reference to `SSL_ctrl'
src/ssl_sock.o: In function `ssl_sock_ocsp_stapling_cbk':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:654: undefined reference to `SSL_get_privatekey'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:686: undefined reference to `SSL_ctrl'
src/ssl_sock.o: In function `ssl_sock_infocbk':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1023: undefined reference to `SSL_get_ex_data'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1042: undefined reference to `SSL_get_wbio'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1043: undefined reference to `SSL_get_rbio'
src/ssl_sock.o: In function `ssl_sock_bind_verifycbk':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1060: undefined reference to `SSL_get_ex_data_X509_STORE_CTX_idx'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1061: undefined reference to `SSL_get_ex_data'
src/ssl_sock.o: In function `ssl_sock_do_create_cert':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1201: undefined reference to `SSL_get_privatekey'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1286: undefined reference to `SSLv23_server_method'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1286: undefined reference to `SSL_CTX_new'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1288: undefined reference to `SSL_CTX_use_PrivateKey'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1316: undefined reference to `SSL_CTX_free'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1290: undefined reference to `SSL_CTX_use_certificate'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1292: undefined reference to `SSL_CTX_check_private_key'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1308: undefined reference to `SSL_CTX_ctrl'
src/ssl_sock.o: In function `ssl_get_tmp_dh':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1636: undefined reference to `SSL_get_privatekey'
src/ssl_sock.o: In function `ssl_sock_close':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3909: undefined reference to `SSL_free'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_sni':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:5008: undefined reference to `SSL_get_servername'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_has_sni':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4795: undefined reference to `SSL_get_servername'
src/ssl_sock.o: In function `ssl_sock_advertise_alpn_protos':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:1171: undefined reference to `SSL_select_next_proto'
src/ssl_sock.o: In function `ssl_tlsext_ticket_key_cb':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:511: undefined reference to `SSL_get_ex_data'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_cipher':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4829: undefined reference to `SSL_get_current_cipher'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4829: undefined reference to `SSL_CIPHER_get_name'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_protocol':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4951: undefined reference to `SSL_get_version'
src/ssl_sock.o: In function `smp_fetch_ssl_c_used':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4625: undefined reference to `SSL_get_peer_certificate'
src/ssl_sock.o: In function `smp_fetch_ssl_c_verify':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:5142: undefined reference to `SSL_get_verify_result'
src/ssl_sock.o: In function `smp_fetch_ssl_x_serial':
[... 221 lines cut ...]
src/ssl_sock.o: In function `ssl_sock_set_servername':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4176: undefined reference to `SSL_ctrl'
src/ssl_sock.o: In function `__ssl_sock_init':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:6457: undefined reference to `SSL_library_init'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:6461: undefined reference to `SSL_CTX_get_ex_new_index'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:6474: undefined reference to `SSL_CTX_get_ex_new_index'
src/shctx.o: In function `shctx_get_cb':
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:507: undefined reference to `d2i_SSL_SESSION'
src/shctx.o: In function `shctx_new_cb':
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:403: undefined reference to `SSL_SESSION_get_id'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:409: undefined reference to `i2d_SSL_SESSION'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:415: undefined reference to `i2d_SSL_SESSION'
src/shctx.o: In function `shctx_remove_cb':
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:526: undefined reference to `SSL_SESSION_get_id'
src/shctx.o: In function `shared_context_set_cache':
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:647: undefined reference to `SSL_CTX_set_session_id_context'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:654: undefined reference to `SSL_CTX_ctrl'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:659: undefined reference to `SSL_CTX_sess_set_new_cb'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:660: undefined reference to `SSL_CTX_sess_set_get_cb'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:661: undefined reference to `SSL_CTX_sess_set_remove_cb'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:650: undefined reference to `SSL_CTX_ctrl'
collect2: error: ld returned 1 exit status
make: *** [haproxy] Error 1

how to do complie sucess?

dgarske, 2018-01-23:
Hi saga1015,

It looks like you are 90% there with getting this working. The build errors indicates that our wolfSSL library isn't enabling the compatibility layer.

The wolfSSL library must be built using `./configure --enable-haproxy` option, which enables the openssl compatibility layer. The defines that get set for this are OPENSSL_EXTRA and WOLFSSL_HAPROXY.

Make sure you put these defines into your CFLAGS here:
TARGET_CFLAGS="-DOPENSSL_EXTRA -DWOLFSSL_HAPROXY".

The ./configure generates a file wolfssl/options.h which contains the generated options. It would be ideal to include this `#include <wolfssl/options.h>` before including any of our openssl compatibility layer files. This will ensure the included headers match the configuration defines used when building wolfSSL.

Thanks,
David Garske, wolfSSL

saga1015, 2018-01-24:
@dgarske thank you very much. 
wolfssl complie option is 
./configure --prefix=/opt/websuite/wolfssl \
--enable-all \
--enable-aesni \
--enable-intelasm \
--enable-sni \
--enable-alpn
make && make install
and wolfssl/options.h is exists ,OPENSSL_EXTRA and WOLFSSL_HAPROXY in options.h.
haproxy complie options add ：
TARGET_CFLAGS="-m64 -m128bit-long-double -DOPENSSL_EXTRA -DWOLFSSL_HAPROXY"
but complie not sucess.
What should I do？

dgarske, 2018-01-24:
Hi saga1015,

Can you provide some of the build errors? If you do include options.h, you will not need to modify your TARGET_CFLAGS. Are you using our latest GitHub sources? We've recently merged in a large number of openssl compatibility additions.

We do have some additional HAProxy information we can send you. Would you mind sending us an email directly to support@wolfssl.com?

Thanks,
David Garske, wolfSSL

saga1015, 2018-01-24:
1、build errors 
src/ssl_sock.o: In function `ssl_sock_close':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3909: undefined reference to `SSL_free'
src/ssl_sock.o: In function `ssl_sock_init':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3377: undefined reference to `SSL_new'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3388: undefined reference to `SSL_set_fd'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3389: undefined reference to `SSL_free'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3431: undefined reference to `SSL_new'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3442: undefined reference to `SSL_set_fd'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3443: undefined reference to `SSL_free'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3400: undefined reference to `SSL_set_ex_data'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3401: undefined reference to `SSL_free'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3454: undefined reference to `SSL_set_ex_data'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3455: undefined reference to `SSL_free'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3411: undefined reference to `SSL_set_connect_state'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3413: undefined reference to `SSL_set_session'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3414: undefined reference to `SSL_SESSION_free'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3465: undefined reference to `SSL_set_accept_state'
src/ssl_sock.o: In function `ssl_sock_shutw':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3923: undefined reference to `SSL_shutdown'
src/ssl_sock.o: In function `ssl_sock_to_buf':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3746: undefined reference to `SSL_read'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3759: undefined reference to `SSL_get_error'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3773: undefined reference to `SSL_get_error'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:3781: undefined reference to `SSL_renegotiate_pending'
src/ssl_sock.o: In function `__ssl_sock_init':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:6457: undefined reference to `SSL_library_init'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:6461: undefined reference to `SSL_CTX_get_ex_new_index'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:6474: undefined reference to `SSL_CTX_get_ex_new_index'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_sni':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:5008: undefined reference to `SSL_get_servername'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_has_sni':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4795: undefined reference to `SSL_get_servername'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_alpn':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4927: undefined reference to `SSL_get0_alpn_selected'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_npn':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4903: undefined reference to `SSL_get0_next_proto_negotiated'
src/ssl_sock.o: In function `smp_fetch_ssl_fc_is_resumed':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4811: undefined reference to `SSL_ctrl'
src/ssl_sock.o: In function `ssl_sock_set_servername':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4171: undefined reference to `SSL_get_servername'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4174: undefined reference to `SSL_set_session'
src/ssl_sock.o: In function `smp_fetch_ssl_x_version':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4656: undefined reference to `SSL_get_peer_certificate'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4658: undefined reference to `SSL_get_certificate'
src/ssl_sock.o: In function `smp_fetch_ssl_c_used':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4625: undefined reference to `SSL_get_peer_certificate'
src/ssl_sock.o: In function `smp_fetch_ssl_x_notbefore':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4531: undefined reference to `SSL_get_peer_certificate'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4533: undefined reference to `SSL_get_certificate'
src/ssl_sock.o: In function `smp_fetch_ssl_x_notafter':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4431: undefined reference to `SSL_get_peer_certificate'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4433: undefined reference to `SSL_get_certificate'
src/ssl_sock.o: In function `ssl_sock_get_cert_used_sess':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4226: undefined reference to `SSL_get_peer_certificate'
src/ssl_sock.o: In function `smp_fetch_ssl_c_verify':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:5142: undefined reference to `SSL_get_verify_result'
src/ssl_sock.o: In function `ssl_sock_get_verify_result':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4249: undefined reference to `SSL_get_verify_result'
src/ssl_sock.o: In function `smp_fetch_ssl_x_sha1':
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4388: undefined reference to `SSL_get_peer_certificate'
/root/source/haproxy/src/haproxy-1.7.10/src/ssl_sock.c:4390: undefined reference to `SSL_get_certificate'
src/ssl_sock.o: In function `smp_fetch_ssl_x_serial':
[... cut ...]
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:409: undefined reference to `i2d_SSL_SESSION'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:415: undefined reference to `i2d_SSL_SESSION'
src/shctx.o: In function `shared_context_set_cache':
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:661: undefined reference to `SSL_CTX_sess_set_remove_cb'
/root/source/haproxy/src/haproxy-1.7.10/src/shctx.c:650: undefined reference to `SSL_CTX_ctrl'

2、wolfssl version is 3.13.0，not github sources
3、how can i set options.h in haproxy source code ?

saga1015, 2018-01-24:
@dgarske 
i modify haproxy Makefile
OPTIONS_LDFLAGS += $(if $(SSL_LIB),-L$(SSL_LIB)) -lssl -lcrypto to 
OPTIONS_LDFLAGS += $(if $(SSL_LIB),-L$(SSL_LIB)) -lwolfssl -lcrypto

saga1015, 2018-01-30:
@dgarske I send email to support@wolfssl.com

kaleb-himes, 2018-05-24:
@saga1015,

Closing this out as it was handled via Zendesk (support@wolfssl.com).

closed 2018-05-24 by kaleb-himes

git001, 2019-02-02:
@kaleb-himes was you able to build and run haproxy with wolfssl?