wolfSSL/wolfssl #1339: curve X25519 for ECDHE in TLS 1.3

julesvanthoor, 2018-01-29:
Does the server in the examples directory also support curve X25519 for ECDHE in TLS 1.3? When testing the server with this curve, there seems to be an error.

SparkiDev, 2018-01-30:
Hi Jules,
The server example does support X25519 in TLS 1.3.
I have been able to run a test case using master:
  ./examples/sever/server -v 4
  ./examples/client/client -v 4 -t
I see: 'SSL curve name is X25519' on the client and server.

The server does not need any command line options to negotiate X25519.

When configuring wolfSSL you must include the option: --enable-curve25519.

I have also been able to run the client against Cloudflare and successfully handshake as both support Draft22.

Which client are you testing against the wolfSSL server?

Sean

kaleb-himes, 2018-05-24:
@julesvanthoor,

Were you able to resolve this using Sean's feedback? Can we close this issue?

julesvanthoor, 2018-05-24:
Yes, the feedback was great, and all is solved! Thanks!

kaleb-himes, 2018-05-24:
@julesvanthoor,

That is great to hear, thank you for confirming! I will go ahead and close this issue.

Cheers,

- KH

closed 2018-05-24 by kaleb-himes