wolfSSL/wolfssl #1350: make fails when configured with --enable-ipv6

GaryGapinski, 2018-02-05:
When configured with `./configure --enable-ipv6` the following error is encountered:
```
….
  CC       wolfcrypt/test/test.o
  CC       examples/client/client.o
  CC       examples/echoclient/echoclient.o
In file included from examples/client/client.c:40:0:
./wolfssl/test.h: In function ‘build_addr’:
./wolfssl/test.h:698:14: error: comparison between pointer and zero character constant [-Werror=pointer-compare]
     if (peer == INADDR_ANY)
              ^~
./wolfssl/test.h:698:9: note: did you mean to dereference the pointer?
     if (peer == INADDR_ANY)
         ^
In file included from ./cyassl/test.h:3:0,
                 from examples/echoclient/echoclient.c:44:
./wolfssl/test.h: In function ‘build_addr’:
./wolfssl/test.h:698:14: error: comparison between pointer and zero character constant [-Werror=pointer-compare]
     if (peer == INADDR_ANY)
              ^~
./wolfssl/test.h:698:9: note: did you mean to dereference the pointer?
     if (peer == INADDR_ANY)
         ^
cc1: all warnings being treated as errors
Makefile:3039: recipe for target 'examples/echoclient/echoclient.o' failed
make[1]: *** [examples/echoclient/echoclient.o] Error 1
make[1]: *** Waiting for unfinished jobs....
cc1: all warnings being treated as errors
Makefile:3039: recipe for target 'examples/client/client.o' failed
make[1]: *** [examples/client/client.o] Error 1
make[1]: Leaving directory '/home/gapinski/Projects/wolfssl'
Makefile:2266: recipe for target 'all' failed
make: *** [all] Error 2
```
(The error does not occur if examples are disabled with --disable-examples.)

dgarske, 2018-02-05:
Hi Gary,

Thanks for your report. A fix has been pushed to pull request 1351 here:
https://github.com/wolfSSL/wolfssl/pull/1351

Let us know if you have any other issues or questions. You can also email us a support@wolfssl.com for free pre-sales or open source support.

Thanks,
David Garske, wolfSSL

dgarske, 2018-02-06:
Hi Gary,

Thanks for spotting that. I had a typo with the IPV6 any addr logic. The `!=` should be `==`. I've pushed a fix.

Thanks,
David Garske, wolfSSL

GaryGapinski, 2018-02-07:
Hello, David:

Thanks! That appears to work properly.

Packet capture attached: [wolfssl-ipv6.pcapng.gz](https://github.com/wolfSSL/wolfssl/files/1702587/wolfssl-ipv6.pcapng.gz). Two sessions using `examples/client/client`. The first was with `-W 1` (which includes both status request types 5 and 17: probably should be just type 5). The second was with `-W 2` — type 17 only, which induced an OCSP request (over IPv6) by the client as the server did not staple (no support for type 17). The first session is an example of how a server will typically ignore the status request extensions (in this case a nonce), as it will have previously procured an OCSP response for subsequent stapling.

Regards,
Gary

closed 2018-02-07 by GaryGapinski