wolfSSL/wolfssl #1354: compile error when using suse, how to fix it ?

cloudyfly, 2018-02-07:
0x00:  Error followed:
make[1]: Entering directory `/home/agent/lib/wolfssl-3.13.0-stable'
make -j2  all-am
make[2]: Entering directory `/home/agent/lib/wolfssl-3.13.0-stable'
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
cc1: error: unrecognized command line option "-Wmaybe-uninitialized"
cc1: error: unrecognized command line option "-Wunused-result"
make[2]: *** [wolfcrypt/src/src_libwolfssl_la-hash.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
**cc1: error: unrecognized command line option "-Wmaybe-uninitialized"
cc1: error: unrecognized command line option "-Wunused-result"**
make[2]: *** [wolfcrypt/src/src_libwolfssl_la-hmac.lo] Error 1
make[2]: Leaving directory `/home/agent/lib/wolfssl-3.13.0-stable'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/agent/lib/wolfssl-3.13.0-stable'
make: *** [agent] Error 2

0x01: suse version
SZX1000381038:/home/agent # uname -a
Linux SZX1000381038 3.0.76-0.11-default #1 SMP Fri Jun 14 08:21:43 UTC 2013 (ccab990) x86_64 x86_64 x86_64 GNU/Linux

0x02: gcc version
SZX1000381038:/home/agent # gcc -v
Using built-in specs.
Target: x86_64-suse-linux
Configured with: ../configure --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib64 --libexecdir=/usr/lib64 --enable-languages=c,c++,objc,fortran,obj-c++,java,ada --enable-checking=release --with-gxx-include-dir=/usr/include/c++/4.3 --enable-ssp --disable-libssp --with-bugurl=http://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --disable-libgcj --disable-libmudflap --with-slibdir=/lib64 --with-system-zlib --enable-__cxa_atexit --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-version-specific-runtime-libs --program-suffix=-4.3 --enable-linux-futex --without-system-libunwind --with-cpu=generic --build=x86_64-suse-linux
Thread model: posix
gcc version 4.3.4 [gcc-4_3-branch revision 152973] (SUSE Linux)

cloudyfly, 2018-02-07:
I delete -Wmaybe-uninitialized and -Wunused-result from wolfssl-3.13.0-stable/Makefile +1559, and have benn passed.

dgarske, 2018-02-07:
Hi Cloudyfly,

It looks like the `x86_64-suse-linux` doesn't understand the CFLAGS for `-Wmaybe-uninitialized -Wunused-result`. I'll see about adding an exclusion for those flags on that platform in our build scripts.

Thanks, David Garske, wolfSSL

dgarske, 2018-03-06:
Hi Cloudyfly,

If you run the `./configure` script it runs a check to see if `-Wunused-result` and `-Wmaybe-uninitialized` are supported. 

```
configure:28604: checking whether C compiler accepts -Wunused-result
configure:27470: checking whether C compiler accepts -Wmaybe-uninitialized
```

Did you run `./configure` before calling `make`?

cloudyfly, 2018-04-03:
suse can't recognize these flags.
for example, 
SZX1000381038:~ # gcc a.c
SZX1000381038:~ # gcc a.c  -Wmaybe-uninitialized
cc1: error: unrecognized command line option "-Wmaybe-uninitialized"

SZX1000381038:~ # cat /proc/version 
Linux version 3.0.76-0.11-default (geeko@buildhost) (gcc version 4.3.4 [gcc-4_3-branch revision 152973] (SUSE Linux) ) #1 SMP Fri Jun 14 08:21:43 UTC 2013 (ccab990)

kaleb-himes, 2018-05-24:
@cloudyfly,

I have tested your configuration:

```
./configure --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib64 --libexecdir=/usr/lib64 --enable-languages=c,c++,objc,fortran,obj-c++,java,ada --enable-checking=release --with-gxx-include-dir=/usr/include/c++/4.3 --enable-ssp --disable-libssp --with-bugurl=http://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --disable-libgcj --disable-libmudflap --with-slibdir=/lib64 --with-system-zlib --enable-__cxa_atexit --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-version-specific-runtime-libs --program-suffix=-4.3 --enable-linux-futex --without-system-libunwind --with-cpu=generic --build=x86_64-suse-linux
```

Using gcc:

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib64/gcc/x86_64-suse-linux/4.8/lto-wrapper
Target: x86_64-suse-linux
Configured with: ../configure --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib64 --libexecdir=/usr/lib64 --enable-languages=c,c++,objc,fortran,obj-c++,java,ada --enable-checking=release --with-gxx-include-dir=/usr/include/c++/4.8 --enable-ssp --disable-libssp --disable-plugin --with-bugurl=http://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --disable-libgcj --disable-libmudflap --with-slibdir=/lib64 --with-system-zlib --enable-__cxa_atexit --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-version-specific-runtime-libs --enable-linker-build-id --enable-linux-futex --program-suffix=-4.8 --without-system-libunwind --with-arch-32=i586 --with-tune=generic --build=x86_64-suse-linux --host=x86_64-suse-linux
Thread model: posix
gcc version 4.8.5 (SUSE Linux)
```

```
uname -a
Linux linux-h3lz 4.4.92-6.30-default #1 SMP Mon Dec 4 08:08:30 UTC 2017 (1fb0e00) x86_64 x86_64 x86_64 GNU/Linux
```

Thank you for bringing this to our attention but we are unable to reproduce this report. Could you make sure you have all the correct dependencies installed for using autoconf and auto build IE:

```
linux-headers-generic
build-essential
libtool
```

We have tried to reproduce with both our master branch and our 3.14.0 release. There is no issue on SUSE Linux x86_64 that we can reproduce.

The one thing I did notice was you had ```../configure``` are you attempting a shadow build from a sub-directory like you would if using cmake?

If so that functionality is not supported at this time.

Warm Regards,

Kaleb

closed 2018-05-24 by kaleb-himes