wolfSSL/wolfssl #1367: Build: Make fails with the error "invalid for movq" and the warning "implicit conversion shortens 64-bit value". Workaround: Use clang instead of lvm-gcc.

dgdy9hvz, 2018-02-12:
I have found that `llvm-gcc` on Mac OS X fails to compile wolfssl-3.13.0.  If gcc is a symlink to llvm-gcc, then, by adding the line `export CC=clang` before `./configure`, `clang` must be forced to be used.

### I did this

The current directory is "wolfssl-3.13.0", to which "wolfssl-3.13.0.zip" expanded.

    ./configure --enable-static --disable-shared
    make

### I expected the following
The "make" completes the compilation and link successfully.

### what happened
The "make" terminates with errors.  The following is the log that "make" printed.  I replaced some paths with "[omitted]" in the log for security and privacy reasons.  According to the "config.log", `CC='gcc'`.

    make -j3  all-am
      CC       wolfcrypt/test/testsuite_testsuite_test-test.o
      CC       examples/echoclient/testsuite_testsuite_test-echoclient.o
      CC       examples/client/testsuite_testsuite_test-client.o
    wolfcrypt/test/test.c: In function 'rsa_decode_test':
    wolfcrypt/test/test.c:7513: warning: implicit conversion shortens 64-bit value into a 32-bit value
    wolfcrypt/test/test.c:7520: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       examples/echoserver/testsuite_testsuite_test-echoserver.o
      CC       examples/server/testsuite_testsuite_test-server.o
      CC       examples/client/tests_unit_test-client.o
      CC       examples/server/tests_unit_test-server.o
      CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
      CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
      CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
    In file included from wolfcrypt/src/hmac.c:38:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-random.lo
      CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
      CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
    In file included from wolfcrypt/src/random.c:96:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/sha256.c:95:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/rsa.c:178:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
      CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
      CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
    In file included from wolfcrypt/src/aes.c:243:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/sha.c:87:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/sha512.c:126:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
      CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
    In file included from wolfcrypt/src/sha3.c:160:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo
      CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
      CC       wolfcrypt/src/src_libwolfssl_la-error.lo
      CC       wolfcrypt/src/src_libwolfssl_la-signature.lo
      CC       wolfcrypt/src/src_libwolfssl_la-wolfmath.lo
      CC       wolfcrypt/src/src_libwolfssl_la-memory.lo
    In file included from wolfcrypt/src/wolfmath.c:51:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-dh.lo
      CC       wolfcrypt/src/src_libwolfssl_la-asn.lo
      CC       wolfcrypt/src/src_libwolfssl_la-coding.lo
    In file included from wolfcrypt/src/asn.c:71:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/dh.c:43:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-poly1305.lo
      CC       wolfcrypt/src/src_libwolfssl_la-md5.lo
    In file included from wolfcrypt/src/poly1305.c:42:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/md5.c:45:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-chacha.lo
      CC       wolfcrypt/src/src_libwolfssl_la-chacha20_poly1305.lo
      CC       wolfcrypt/src/src_libwolfssl_la-tfm.lo
    In file included from wolfcrypt/src/chacha.c:45:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    wolfcrypt/src/chacha.c: In function 'wc_Chacha_encrypt_bytes':
    wolfcrypt/src/chacha.c:991: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/chacha20_poly1305.c:42:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
      CC       wolfcrypt/src/src_libwolfssl_la-ecc.lo
      CC       src/src_libwolfssl_la-internal.lo
    In file included from wolfcrypt/src/tfm.c:44:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    wolfcrypt/src/tfm.c: In function 'fp_forcezero':
    wolfcrypt/src/tfm.c:2266: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from wolfcrypt/src/ecc.c:119:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    wolfcrypt/src/ecc.c: In function 'wc_ecc_mulmod_ex':
    wolfcrypt/src/ecc.c:2221: warning: implicit conversion shortens 64-bit value into a 32-bit value
    In file included from src/internal.c:40:
    ./wolfcrypt/src/misc.c: In function 'ForceZero':
    ./wolfcrypt/src/misc.c:198: warning: implicit conversion shortens 64-bit value into a 32-bit value
    src/internal.c: In function 'SendServerHello':
    src/internal.c:20346: warning: implicit conversion shortens 64-bit value into a 32-bit value
    /var/[omitted]/ccvDgLmH.s:15:suffix or operands invalid for `movq'
    /var/[omitted]/ccvDgLmH.s:58:suffix or operands invalid for `movq'
    /var/[omitted]/ccvDgLmH.s:116:suffix or operands invalid for `movq'
    /var/[omitted]/ccvDgLmH.s:143:suffix or operands invalid for `movq'
    /var/[omitted]/ccvDgLmH.s:3359:suffix or operands invalid for `movq'
    make[1]: *** [wolfcrypt/src/src_libwolfssl_la-tfm.lo] Error 1
    make[1]: *** Waiting for unfinished jobs....
    make: *** [all] Error 2

### workaround

This problem can be avoided by adding the line `export CC=clang` before `./configure`.  Then, the `make` successfully completes.

    export CC=clang
    ./configure --enable-static --disable-shared
    make

### environment

wolfssl-3.13.0
Mac OS X

kaleb-himes, 2018-02-12:
Hi @dgdy9hvz,

Could you type echo $CC in a terminal where you have not exported CC=clang. Then could you tell us the version of that compiler (For example gcc -v if $CC is default set to gcc).

Also is your Mac a 64-bit or 32-bit arch?

Have you replaced the default gcc (which on a Mac usually just uses clang anyway)? It seems odd that using CC=clang on a Mac would change much unless using an alternate gcc for example on my Mac I see:

```
gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.39.2)
```

Thus my request to see which version and which tool is being used by default. Thanks!

- KH

kaleb-himes, 2018-02-12:
@dgdy9hvz,

Also as a note, the reason I asked you to ```echo $CC``` even though it is not usually set is that our configure will "DEFAULT" to gcc iff $CC is not already set. I am wondering trying to check two things here:

#1: has $CC been set to something already and if so what and what version
#2: If not set then our configure script will default to gcc but on a standard MAC this should not be any different then setting to clang so.... if it is defaulting to gcc what version of gcc is that?

Thanks!

- KH

dgdy9hvz, 2018-02-20:
By default, the environment variable CC is empty, as `echo $CC` and `printenv CC` return nothing.  The architecture on which this problem occurs is 64-bit.

Newer versions of Xcode come with only clang; hence, gcc invokes clang, as you point out.

However, older versions of Xcode came with both llvm-gcc and clang.  gcc was a symlink to llvm-gcc, and cc was a symlink to clang.  gcc did not invoke clang, as `gcc -v` showed.

kaleb-himes, 2018-04-30:
This should be resolved in recent updates to autoconf system. Let us know if the issue recurs.

K

closed 2018-04-30 by kaleb-himes