wolfSSL/wolfssl #1398: compile error when use --enable-32bit

yesme2000, 2018-02-24:
./configure --enable-aesni  --enable-32bit --enable-xts --enable-debug  --enable-aesccm --enable-aesctr

yesme2000, 2018-02-24:
wolfcrypt/src/aes.c: In function ‘AES_GCM_encrypt’:
wolfcrypt/src/aes.c:4522:5: error: unknown register name ‘r13’ in ‘asm’
     __asm__ __volatile__ (
     ^~~~~~~
wolfcrypt/src/aes.c:4522:5: error: unknown register name ‘xmm11’ in ‘asm’
wolfcrypt/src/aes.c:4522:5: error: unknown register name ‘xmm10’ in ‘asm’
wolfcrypt/src/aes.c:4522:5: error: unknown register name ‘xmm9’ in ‘asm’
wolfcrypt/src/aes.c:4522:5: error: unknown register name ‘xmm8’ in ‘asm’

kaleb-himes, 2018-03-05:
Hi yesme2000,

Could you try adding --disable-asm (disable assembly optimizations) along with --enable-32bit and let me know your results?

By default our assembly optimizations are using registers that are not available on a 32-bit architecture thus the errors you are seeing. I've added a note to have one of our developers look at updating --enable-32bit to default disable assembly.

Warm Regards,

Kaleb

kaleb-himes, 2018-04-30:
Hi @yesme2000,

Just wanted to confirm this resolved your issues before closing, could you provide an update on this issue when you get a chance?

Thanks!

K

closed 2018-05-24 by kaleb-himes