wolfSSL/wolfssl #1415: 3.14.0 build failure No rule to make target `wolfssl/wolfcrypt/fips.h', needed by `all-am'.  Stop.

ilovezfs, 2018-03-03:
```
make[1]: *** No rule to make target `wolfssl/wolfcrypt/fips.h', needed by `all-am'.  Stop.
```
https://jenkins.brew.sh/job/Homebrew%20Core%20Pull%20Requests/19634/version=high_sierra/console

kaleb-himes, 2018-03-05:
Hi @ilovezfs,

I suspect your Jenkins system must be combing through our configure options that are available and simply testing them all or something to that effect?

We do have a --enable-fips option but that will fail 100% of the time if not used with our commercial FIPS bundle as we do not distribute the FIPS sources in our open-source packages. I would recommend adding an exception to the jerkins test to skip that configure option.

Warmest Regards,

Kaleb

kaleb-himes, 2018-03-05:
Hi @ilovezfs,

I did a little more digging and I think I see what's happening, our include.am file references that header but the header does not exist by default. However if you run ```./autogen.sh``` then the fips.h header will at least be "touched" so that it exists even if not configuring with --enable-fips.

Could you tell me how your system is pulling in our library and testing? Are you grabbing a tarball from GitHub or are you cloning and then checking out a tag? Are you running ```./autogen.sh``` before running ```./configure``` in the test?

Warm Regards,

Kaleb

ilovezfs, 2018-03-05:
@kaleb-himes the formula is here https://github.com/Homebrew/homebrew-core/blob/master/Formula/wolfssl.rb

kaleb-himes, 2018-03-05:
@ilovezfs,

Thank you so much for sharing the build script! That was most helpful. We found the issue. Our autogen.sh is ONLY touching the fips.h header IF the directory .git exists. Since you are checking out the tarball ```url "https://github.com/wolfSSL/wolfssl/archive/v3.13.0-stable.tar.gz"``` there is no .git directory so the header is not touched. We will take steps to fix and I will get you an update ASAP!

Cheers,

Kaleb

dgarske, 2018-03-14:
Fix was committed in PR #1423. Marking this issue resolved. Thanks ilovezfs.

closed 2018-03-14 by dgarske

yarf, 2018-11-30:
I ran into this issue with 3.15.5 ... where wolfssl/ is hosted within a subtree of a larger git repo (and didn't have a local .git file in the wolfssl/ directory). 

Hacked it by explicitly creating the .git dir before running ./autogen.sh, yet I thought this issue had been resolved prior to the 3.15.5 build?