wolfSSL/wolfssl #1435: Issues while trying to build with WOLFSSL_ATMEL and WOLFSSL_ATECC508A

duhampler, 2018-03-14:
Hello,
i'm trying to build wolfssl for an ARM-Platform and with activated support for MicroChips ATECC508a. The Version is wolfssl 3.14.0
Unfortunatly I'm running in some issues.
First, "socket/include/socket.h" included by wolfio.c isn't found. As it is included with quotes I assume that it have to be some where in the source tree? As it is only include when WOLFSSL_ATMEL is active it looks to me that socket/include/socket.h provides some ATMEL/ATECC508 specific implementations, right?I didn't found it. The Makefile neither. So how I provide this include file to wolfssl?
Second issue was that the "struct tm" implemented by wolfssl collides with the implementation of time.h. I solved it by passing -DNO_ASN_TIME.. I don't know if this appsoach is correct.

duhampler, 2018-03-14:
okay. I don't know if this is the right solution, but I removed WOLFSSL_ATMEL as additional CFLAG and now it compiles. What is the reason for having WOLFSSL_ATMEL and WOLFSSL_ATECC508A? Is it the right way to do the build without WOLFSSL_ATMEL?

dgarske, 2018-03-19:
Hi Duhampler,

That is a great question! For your situation I recommend not defining either of these and just using `HAVE_PK_CALLBACKS` to define your own ECC callbacks like is done for the ATECC508A TLS example here:
https://github.com/dgarske/atmel/blob/master/tls_demo/tls_client.c#L302

For your questions:
1. The socket.h referenced in wolfio.h when `WOLFSSL_ATMEL` is part of the WINC1500 sources. Which network stack are you using? We support most stacks or you can define your own read/write callbacks using `wolfSSL_SetIORecv` and `wolfSSL_SetIOSend`. Good example here:
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-callback.c#L193

2. If you are having a `struct tm` collision with time.h then you'll need to setup your time correctly. If you see the `WOLFSSL_ATMEL` defines enables the wolf implementation of `struct tm` here:
https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/wc_port.h#L392
You'll need to figure out which time implementation best suites your environment to get time. If you define `NO_ASN_TIME` you will be skipping all certificate date checking. That is not recommended in a production environment.

The `WOLFSSL_ATMEL` define does the following:
1. Sets the ASN time to use the `atmel_get_curr_time_and_date` function.
2. Sets the RNG to use the `atmel_get_random_number` function.
3. On wolfCrypt_init calls `atmel_init`.
These functions are in the `wolfcrypt/src/port/atmel/atmel.c` file.
4. Uses a BSD socket interface at `WOLFSSL_ATMEL`, which is part of the WINC1500.

The `WOLFSSL_ATECC508A` define does the following:
1. Enables hardware based ECC P256 operations in wolfCrypt for wc_ecc_* API's.
2. In the atmel.c port it enables calls to `atcatls_random` (in Atmel's CryptoAuthLib) and `rtc_calendar_get_time`.

David Garske, wolfSSL

duhampler, 2018-03-21:
Hello David,
thanks for your fast response. Maybe I have to be more specific on what system I'm using. I'm building the wolfssl library for an embedded linux OS with the help of the yocto build system. So my network stack is the linux network stack.
Why is the library coupled to code of an Atmel Wifi-Controller? Is there an easy/short way to change the stack to the linux network stack?
And it is not clear to me why I shouldn't define the two switches but register own ECC Callbacks? Maybe I didn't understand what your library is capable of?

dgarske, 2018-03-21:
Hi duhampler,

Thanks for the details! If you avoid defining `WOLFSSL_ATMEL` it will default to linux style BSD interfaces for the network interface, DRBG seed (dev/random) and time. In that case you will not need to use the read/write callbacks.

You will need to define the PK callbacks (HAVE_PK_CALLBACKS) for using the ECC508A with TLS to intercept the ECC sign, verify and shared secret options.

David Garske, wolfSSL

dgarske, 2018-03-21:
Hi duhampler,

A fix for building without WOLFSSL_ATMEL defined has been pushed here:
https://github.com/wolfSSL/wolfssl/pull/1453

Thanks,
David Garske, wolfSSL

duhampler, 2018-03-22:
Hi dgarkse,

when I try to build I've got the error that PUB_KEY_SIZE is not defined in ecc.h

 In file included from ../git/wolfssl/openssl/ec.h:28:0,
                  from ../git/wolfssl/openssl/evp.h:49,
                  from ../git/wolfssl/openssl/hmac.h:38,
                  from ../git/wolfssl/ssl.h:91,
                  from ../git/cyassl/ssl.h:33,
                  from ../git/examples/echoserver/echoserver.c:27:
 ../git/wolfssl/wolfcrypt/ecc.h:310:21: error: 'PUB_KEY_SIZE' undeclared here (not in a function)
      byte pubkey_raw[PUB_KEY_SIZE];

Disabling building of examples (--disable-examples) led to the following error:
 make[1]: *** No rule to make target `cyassl/options.h', needed by `all-am'.  Stop.

What I found is a define named #define ED25519_PUB_KEY_SIZE 32. Is there maybe a typo or has PUB_KEY_SIZE to be defined elsewhere?

Before that I used wolfssl from the download site, version 3.13, and the configure optione --enable-opensslextra.
Thanks

dgarske, 2018-03-22:
Hi duhampler,

The `PUB_KEY_SIZE` macro is in the Atmel demo at `cryptoauthlib/tls/atcatls.h:64` as `#define PUB_KEY_SIZE ATCA_PUB_KEY_SIZE` and `ATCA_PUB_KEY_SIZE` is defined at `cryptoauthlib/atca_command.h:243` as `#define ATCA_PUB_KEY_SIZE (64) //!< size of a p256 public key`. If `WOLFSSL_ATECC508A is defined the `wolfssl/wolfcrypt/port/atmel/atmel.h` file is included, which includes the above.

In case something changes in CryptoAuthLib I went ahead and pushed a changed to PR #1453 to use an internal define.

The `./configure` command generates `wolfssl/options.h` and `cyassl/options.h`. Does that file at `../git/wolfssl/cyassl/options.h` exist for you? Its possible not configuring and building directly from your `git/wolfssl` directory is causing that error. For reference the file in cyassl and ctaocrypt are for compatibility to support our old function names. The echoserver uses those old files to test the compatibility layer.

Thanks, David Garske

duhampler, 2018-03-23:
Okay. So what version of cryptoauthlib you are using? In the version on github the line defining PUB_KEY_SIZE is disabled. 
In the log file of configure I recognized this line(s):
```
touch: cannot touch 'cyassl/options.h': No such file or directory
../git/configure: line 32184: cyassl/options.h: No such file or directory
../git/configure: line 32185: cyassl/options.h: No such file or directory
../git/configure: line 32186: cyassl/options.h: No such file or directory
```
Could it be that the directory cyassl isn't created in the build directory? wolfssl/options.h was created.
When I build wolfssl 3.13 the cyassl directory seems not to be needed as the build is successful.
Sorry I keep you busy

dgarske, 2018-03-23:
Hi duhampler,

Touch is supposed to create the file if its missing. What platform are you running ./configure on? For now you can probably get passed that by creating an empty file at `cyassl/options.h`.

Not seeing a version in CryptoAuthLib, but the readme is dated `01/08/2016`. We plan to update to the latest CryptoAuthLib V2 soon and we will be adding support for the ATECC608A as well. For now you can use the pull request #1453 to resolve this as it now uses a wolf  macro to determine HW key size.

David

closed 2018-04-06 by dgarske

ejohnstown, 2018-06-25:
Have you tried this recently? In v3.15.3, we updated the configure to handle the cyassl/options.h the same way wolfssl/options.h is handled.