wolfSSL/wolfssl #1436: logging.c:(.text+0xda): undefined reference to `sprintf'

daichunkai, 2018-03-14:
hello,When I use SGX to call aes function of Wolfssl,an error happened when I link enclave.so as following:
```
home/dck/SGX-develop/wolfssl/wolfssl-master/IDE/LINUX-SGX/libwolfssl.sgx.static.lib.a(logging.o): In function `WOLFSSL_BUFFER':
logging.c:(.text+0xda): undefined reference to `sprintf'
logging.c:(.text+0x139): undefined reference to `sprintf'
collect2: error: ld returned 1 exit status
Makefile:267: recipe for target 'enclave.so' failed
make: *** [enclave.so] Error 1
```
In the example of SGX-LINUX in Wolfssl the link flags as following:
```
-m64 -O0 -g -DSGX_DEBUG -Wl,--no-undefined -nostdlib -nodefaultlibs -nostartfiles -L/opt/intel/sgxsdk/lib64 -L../../wolfssl-master/IDE/LINUX-SGX/ -lwolfssl.sgx.static.lib -Wl,--whole-archive -lsgx_trts -Wl,--no-whole-archive -Wl,--start-group -lsgx_tstdc -lsgx_tstdcxx -lsgx_tcrypto -lsgx_tservice -Wl,--end-group -Wl,-Bstatic -Wl,-Bsymbolic -Wl,--no-undefined -Wl,-pie,-eenclave_entry -Wl,--export-dynamic -Wl,--defsym,__ImageBase=0 -Wl,--version-script=trusted/Wolfssl_Enclave.lds@
LINK =>  Wolfssl_Enclave.so
```

and In my enclave,the link flags as following:
```
-m64 -O0 -g -DSGX_DEBUG -Wl,--no-undefined -nostdlib -nodefaultlibs -nostartfiles -L/opt/intel/sgxsdk/lib64 -L/home/dck/SGX-develop/wolfssl/wolfssl-master/IDE/LINUX-SGX -lwolfssl.sgx.static.lib -Wl,--whole-archive -lsgx_trts -Wl,--no-whole-archive -Wl,--start-group -lsgx_tstdc -lsgx_tcxx -lsgx_tcrypto -lsgx_tservice -Wl,--end-group -Wl,-Bstatic -Wl,-Bsymbolic -Wl,--no-undefined -Wl,-pie,-eenclave_entry -Wl,--export-dynamic -Wl,--defsym,__ImageBase=0 -Wl,--gc-sections -Wl,--version-script=Enclave/Enclave.lds@
```
the link flags is same  with above flags, here is my follow edl file and enclave file:
edl file:
```
/*
 * Copyright (C) 2011-2017 Intel Corporation. All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 *
 *   * Redistributions of source code must retain the above copyright
 *     notice, this list of conditions and the following disclaimer.
 *   * Redistributions in binary form must reproduce the above copyright
 *     notice, this list of conditions and the following disclaimer in
 *     the documentation and/or other materials provided with the
 *     distribution.
 *   * Neither the name of Intel Corporation nor the names of its
 *     contributors may be used to endorse or promote products derived
 *     from this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 * "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
 * A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
 * OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
 * SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
 * LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
 * DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
 * THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
 * (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
 * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 */

/* Enclave.edl - Top EDL file. */

enclave {
    
    include "user_types.h" /* buffer_t */
    include "wolfssl/wolfcrypt/aes.h"

    /* Import ECALL/OCALL from sub-directory EDLs.
     *  [from]: specifies the location of EDL file. 
     *  [import]: specifies the functions to import, 
     *  [*]: implies to import all functions.
     */
    
    from "Edger8rSyntax/Types.edl" import *;
    from "Edger8rSyntax/Pointers.edl" import *;
    from "Edger8rSyntax/Arrays.edl" import *;
    from "Edger8rSyntax/Functions.edl" import *;

    from "TrustedLibrary/Libc.edl" import *;
    from "TrustedLibrary/Libcxx.edl" import ecall_exception, ecall_map;
    from "TrustedLibrary/Thread.edl" import *;

    /* 
     * ocall_print_string - invokes OCALL to display string buffer inside the enclave.
     *  [in]: copy the string buffer to App outside.
     *  [string]: specifies 'str' is a NULL terminated buffer.
     */
    untrusted {
        void ocall_print_string([in, string] const char *str);
    };

};
```

enclave.cpp
```
#include <stdarg.h>
#include <stdio.h>      /* vsnprintf */

#include "Enclave.h"
#include "Enclave_t.h"  /* print_string */
#include "sgx_trts.h"

/* 
 * printf: 
 *   Invokes OCALL to display the enclave buffer to the terminal.
 */
void test_aes (void)
{
   Aes enc;
   const byte iv[]={1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16};
   int result_iv;
   result_iv=wc_AesSetIV(&enc,iv);
   
}
void printf(const char *fmt, ...)
{
    char buf[BUFSIZ] = {'\0'};
    va_list ap;
    va_start(ap, fmt);
    vsnprintf(buf, BUFSIZ, fmt, ap);
    va_end(ap);
    test_aes();
    ocall_print_string(buf);
}
```
Thank you very much!

JacobBarthelmeh, 2018-03-14:
Hi daichunkai,

For use with debugging, sprintf will need to be implemented similar to how you have implemented printf. An example of this is located here with the wolfSSL Linux SGX build 
 (https://github.com/wolfSSL/wolfssl-examples/blob/b21f358dac4fd3bb1cea58b7ee2d49812a93c092/SGX_Linux/trusted/Wolfssl_Enclave.c#L172)

Regards,
Jacob

daichunkai, 2018-03-15:
@JacobBarthelmeh  YES,I compile it as your suggestion，it works! Thank  you very much！

JacobBarthelmeh, 2018-03-20:
Happy to help daichunkai. Am going to go ahead and mark this issue closed but feel free to contact us with any wolfSSL SGX questions that arise.

Regards,
Jacob

closed 2018-03-20 by JacobBarthelmeh