wolfSSL/wolfssl #1439: Converting project from openssl to wolfssl

lu-yong, 2018-03-15:
I get a bunch of warnings telling me:
> warning:` implicit declaration of function 'SSL_set_mode
 error: 'SSL_MODE_AUTO_RETRY' undeclared (first use in this function)
 note: each undeclared identifier is reported only once for each function it appears in
 error: 'GENERAL_NAMES' undeclared (first use in this function)

I compiled wolfssl with flags "--enable-opensslextra --disable-examples". I can't find definitions fucntion "SSL_set_mode" and structure "GENERAL_NAMES" in your wolfssl project.

JacobBarthelmeh, 2018-03-15:
Hi lu-yong,

Thanks for contacting wolfSSL! Is this an open source project that we could view? The function SSL_set_mode is currently not implemented in wolfSSL. The compatibility layer is expanding with the number of functions provided but it still remains a subset of the total number of OpenSSL API. I'll add implementing the function to the desired feature list.

Regards,
Jacob

JacobBarthelmeh, 2018-03-20:
Thank you lu-yong for the link. I took a brief look at the project and you may be able to get away with just having a stub function for SSL_set_mode(). An example of a stub function would be wolfSSL_ctrl in src/ssl.c that has the macro mapping SSL_ctrl to wolfSSL_ctrl in wolfssl/openssl/ssl.h.

Implementing the function was added to our feature request list to be tackled when an engineer has time between projects but it does not have a set timeframe yet. We could also accept a code contribution (if having a signed contributor agreement) if you wanted to go ahead and open a pull request on the wolfSSL repository with the implementation of SSL_set_mode().

If questions arise through the process of adding SSL_set_mode() or any other question with using wolfSSL, feel free to reach out to us.

Regards,
Jacob

haydenroche5, 2021-01-13:
SSL_set_mode is now supported in our OpenSSL compatibility layer.

closed 2021-01-13 by haydenroche5