wolfSSL/wolfssl #1450: Conflicting types for 'time_t' within GNU Arm Embedded Toolchain 7

ycherniavsky, 2018-03-20:
Got this error on trying to compile wolfssl bindings in ChibiOS stable_18.2.x which uses WolfSSL 3.12.2 with GNU Arm Embedded Toolchain 7 (7.2.1 20170904):
```
Compiling wolfssl_chibios.c
In file included from ../../../ext/wolfssl/wolfssl/wolfcrypt/types.h:28:0,
                 from ../../../os/various/wolfssl_bindings/wolfssl_chibios.h:42,
                 from ../../../os/various/wolfssl_bindings/wolfssl_chibios.c:41:
../../../ext/wolfssl/wolfssl/wolfcrypt/wc_port.h:442:18: error: conflicting types for 'time_t'
     typedef long time_t;
                  ^~~~~~
In file included from /Users/ych/projects/dev/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include/sys/select.h:26:0,
                 from /Users/ych/projects/dev/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include/sys/types.h:68,
                 from /Users/ych/projects/dev/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include/time.h:28,
                 from ../../../ext/wolfssl/wolfssl/wolfcrypt/wc_port.h:399,
                 from ../../../ext/wolfssl/wolfssl/wolfcrypt/types.h:28,
                 from ../../../os/various/wolfssl_bindings/wolfssl_chibios.h:42,
                 from ../../../os/various/wolfssl_bindings/wolfssl_chibios.c:41:
/Users/ych/projects/dev/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include/sys/_timeval.h:40:18: note: previous declaration of 'time_t' was here
 typedef _TIME_T_ time_t;
                  ^~~~~~
make[1]: *** [build-stm32f746_discovery/obj/wolfssl_chibios.o] Error 1
make: *** [all] Error 2
```
From error message it seams that prolem exactly in `wolfssl/wolfcrypt/wc_port.h` and `arm-none-eabi/include/sys/select.h` and not in ChibiOS wolfssl bindings.

And yes under old GNU Arm Embedded Toolchain 4 (4.9.3 20150303) the same code compiles without error.

dgarske, 2018-03-20:
Hi ycherniavsky,

That would happen if `USE_WOLF_TIME_T` is defined, which can happen if `USER_TIME` or `TIME_OVERRIDES` is defined. You can try defining `HAVE_TIME_T_TYPE` to tell our library that your system already includes the time.h time_t.

You can see a list of time configuration options here in wc_port.h:
https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/wc_port.h#L336

Can you provide a list of build defines / options you are using and which platform you are on so we can suggest the best build options?

Thanks,
David Garske, wolfSSL

ycherniavsky, 2018-03-21:
@dgarske you are absolutely right about `USE_WOLF_TIME_T` - just remove it from [wolfssl settings header](https://github.com/ChibiOS/ChibiOS/blob/master/os/various/wolfssl_bindings/user_settings.h) and all compiled without error under GNU Arm Embedded Toolchain 4 and 7.
Thanks!

closed 2018-03-21 by ycherniavsky