wolfSSL/wolfssl #1474: some APIs required for Extended Cache support requires openssl extra

hajjihraf, 2018-04-04:
Hi,

I'm using the extended cache feature (macro HAVE_EXT_CACHE) which requires the APIs:
wolfSSL_SESSION_free
wolfSSL_CTX_sess_set_get_cb
wolfSSL_CTX_sess_set_new_cb
wolfSSL_CTX_sess_set_remove_cb

The 4 APIs above requires the definition of the macro OPENSSL_EXTRA.
Is it possible to remove this dependencies so HAVE_EXT_CACHE can be used without OPENSSL_EXTRA.

Thanks in advance,
Achraf HAJJI

hajjihraf, 2018-04-05:
Hi dgarske,

I looked into your change, it resolves the build error.
However the ext_cache callbacks setters still requires the definition of the macro OPENSSL_EXTRA.

Regards,
Achraf HAJJI

dgarske, 2018-04-05:
Hi hajihraf,

Thanks for pointing that out. It looked like those were available when in-fact they were not without OPENSSL_EXTRA. I've fixed that and even added tests for the callbacks.

The change is here in pull request #1480.

Thanks,
David Garske, wolfSSL

closed 2018-04-06 by hajjihraf