wolfSSL/wolfssl #1481: AES128-SHA is broken when TLS 1.3 support is enabled

pelya, 2018-04-05:
When TLS 1.3 support is enabled, and TLS version auto-detection is used in the client, static RSA cipher suites are broken (at least AES128-SHA and AES256-SHA).

Steps to reproduce:
Start simple HTTPS server using OpenSSL commandline tool, force AES128-SHA cipher:

    openssl req -x509 -newkey rsa:4096 -keyout key.pem -out server.pem -days 365 -passout pass:12345
    openssl s_server -accept 11111 -cipher AES128-SHA -key key.pem -pass pass:12345 -www

Build WolfSSL with static RSA support:

    env CFLAGS=-DWOLFSSL_STATIC_RSA ./configure --enable-tls13 --prefix=`pwd`/build
    make -j4
    make install

Build the attached client-tls.c, it's the sample client from wolfssl-examples repository, with only two changes are to disable certificate validity checks and to use `wolfSSLv23_client_method()` instead of `wolfTLSv1_2_client_method()`:

    gcc client-tls.c -I build/include/ -L build/lib  -o client-tls -lwolfssl -lm

Run the test:

    env LD_LIBRARY_PATH=build/lib ./client-tls 127.0.0.1

The server will print the error `SSL routines:ssl3_get_record:decryption failed or bad record mac`

If you remove `--enable-tls13` from `configure` arguments or use `wolfTLSv1_2_client_method()` in client-tls.c, the error does not happen.

[client-tls.c.txt](https://github.com/wolfSSL/wolfssl/files/1880597/client-tls.c.txt)

kaleb-himes, 2018-04-05:
Hi @pelya 

Thank you so much for reporting this. I've assigned our TLS 1.3 expert to the issue for review.
Thanks for contacting wolfSSL!

- Kaleb

SparkiDev, 2018-04-09:
Hi @pelya 

First of all a quick note: TLS 1.3 does not support static RSA.
To help with your problem, OpenSSL and wolfSSL aren't supporting the same draft version of TLS 1.3 at this time.
The latest of OpenSSL master supports Draft 26 and wolfSSL supports Draft 27.
There is no significant difference in the protocol implementation, just a difference in version number.
To quickest way to fix this is to change the draft minor version in internal.h. That is, change TLS_DRAFT_MINOR to be 0x1a in wolfssl/internal.h.

The TLS 1.3 specification is approved but not yet been made an RFC. Hopefully OpenSSL will update themselves to the latest draft version number. wolfSSL will change to 28 in the very near future.

Thanks,
Sean :-)

pelya, 2018-04-10:
When I'm forcing TLS version to 1.1 or 1.2 on the server side, connection still fails, however in case of TLS 1.1 the server does not print the error. This does not look like a problem in differences between TLS 1.3 implementations.

    openssl s_server -accept 11111 -cipher AES128-SHA -key key.pem -pass pass:12345 -www -tls1_1
    openssl s_server -accept 11111 -cipher AES128-SHA -key key.pem -pass pass:12345 -www -tls1_2

kaleb-himes, 2018-04-10:
Hi @pelya,

As Sean pointed out AES128-SHA is a static RSA cipher suite so testing these with TLS v1.3 is a mute point anyway. As for testing with TLS v1.1 and static RSA you need to configure wolfSSL like so:

```
./configure --enable-tlsv10 CFLAGS=-DWOLFSSL_STATIC_RSA && make
```

Also in your openSSL command you are loading a unique key with a password but I don't see you loading the cert that corresponds to that key. Our client will reject the connection if the cert sent by the server was not signed by the key loaded by the server. You should be loading the cert that corresponds to that key with ```-cert <path-to-cert>/certName.pem``` unless you have modified their server code to load it already?

Regards,

- KH

pelya, 2018-04-10:
I have tried `--enable-tlsv10` as suggested, and it makes no difference. Full command line:

    ./configure --enable-tlsv10 CFLAGS=-DWOLFSSL_STATIC_RSA --enable-tls13 --prefix=`pwd`/build
    make
    make install
    gcc client-tls.c -I build/include/ -L build/lib  -o client-tls -lwolfssl -lm

The default certificate name for `openssl s_server` command is `server.pem`, which was generated by previous `openssl req` command from comment 0. Here I specify `server.pem` explicitly, and force TLS 1.0:

    $ openssl s_server -accept 11111 -cipher AES128-SHA -key key.pem -pass pass:12345 -www -tls1 -cert server.pem

Still the same error:

    $ env LD_LIBRARY_PATH=build/lib ./client-tls 127.0.0.1
    ERROR: failed to connect to wolfSSL

The certificate check is disabled in client-tls.c, which is attached to this bug:

    wolfSSL_CTX_set_verify(ctx, WOLFSSL_VERIFY_NONE, NULL);

pelya, 2018-04-30:
The problem is specifically with `--enable-tls13` flag, which you did not use in your test. When TLS 1.3 is enabled via `configure --enable-tls13 CFLAGS=-DWOLFSSL_STATIC_RSA` options, TLS 1.1 stops working.

kaleb-himes, 2018-05-01:
@pelya,

I apologize for prematurely closing. It's not that TLS 1.0 stops working it is that the downgrade option stops working. Here is the configure I'm using to test in light of your last:

```
./configure --enable-tls13 --enable-tlsv10 CFLAGS=-DWOLFSSL_STATIC_RSA && make
make install
```

I can confirm that using ```wolfSSLv23_client_method``` will no longer allow you to downgrade to TLSv1.0 when TLSv 1.3 is enabled. You can however successfully invoke ```wolfTLSv1_client_method``` and get a successful connection using TLSv1.0.

Thanks for pointing out the difference in my test case. I'll reach out to our TLSv1.3 developer and let him know about this issue.

-KH

kaleb-himes, 2018-05-02:
@pelya,

This has been addressed in PR #1534 

https://github.com/wolfSSL/wolfssl/pull/1534/files

Please let us know once you have an opportunity to test those changes.

Regards,

KH

closed 2018-05-24 by kaleb-himes