wolfSSL/wolfssl #1486: Missing include: <sys/time.h>

jkortge, 2018-04-06:
Compiling 3.14.0-stable for QNX Neutrino 6.6 fails due to a missing include.  

./configure CFLAGS=-Os CPPFLAGS="-DWOLFSSL_HAVE_MIN -DWOLFSSL_HAVE_MAX" --host=arm-unknown-nto-qnx6.6.0eabi --enable-opensslextra --enable-fortress --enable-aesgcm --enable-ecc --enable-eccencrypt --enable-tlsx --enable-hkdf --enable-ocsp --enable-ocspstapling --enable-ocspstapling2 --enable-sessioncerts --enable-cmac --enable-certgen --enable-certreq --disable-oldtls --enable-keygen
...
$ make
...
  CC       wolfcrypt/src/src_libwolfssl_la-ecc.lo
  CC       src/src_libwolfssl_la-internal.lo
  CC       src/src_libwolfssl_la-wolfio.lo
In file included from ./wolfssl/internal.h:112:0,
                 from src/internal.c:32:
./wolfssl/callbacks.h:60:20: error: field 'timestamp' has incomplete type
./wolfssl/callbacks.h:72:16: error: field 'timeoutValue' has incomplete type
In file included from ./wolfssl/internal.h:112:0,
                 from src/wolfio.c:37:
./wolfssl/callbacks.h:60:20: error: field 'timestamp' has incomplete type
./wolfssl/callbacks.h:72:16: error: field 'timeoutValue' has incomplete type
make[1]: *** [src/src_libwolfssl_la-wolfio.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [src/src_libwolfssl_la-internal.lo] Error 1

In wc_port.h it includes <time.h>, but "struct timeval" referenced in callbacks.h is actually from <sys/time.h> ( see http://pubs.opengroup.org/onlinepubs/7908799/xsh/systime.h.html ).  Adding that to wc_port.h allows a clean build.

kaleb-himes, 2018-04-09:
Hi @jkortge,

Thank you for letting us know about this item. When you say:

> Adding that to wc_port.h allows a clean build.

Does that mean you "REPLACED" time.h with <sys/time.h> or does that mean you included both:

```
#include <time.h>
#include <sys/time.h>
```

Regards,

Kaleb

jkortge, 2018-04-09:
Both. I assumed <time.h> would still be required for other types like strict timeval.

kaleb-himes, 2018-04-09:
@jkortge,

Thank you. I'm also working with another customer reporting missing headers for the timeval structs and we have been dong some optimization refactoring lately so it is possible the header includes may be broken in certain scenarios. <sys/time.h> won't be available on every system so we want to be careful with how and where we are to include it.

I'll talk to the team and see what we can figure out for a portable solution.

- KH

dgarske, 2018-04-23:
Hi jkortge,

Thanks for the report about missing `sys/time.h`. I agree on certain platforms this header is required. I've issued a new PR #1512 which allows you to define `HAVE_SYS_TIME_H` to add the include in `wc_port.h`. You may also find it useful to build with the `WOLF_C99` option to force using features for c99 compliance.

Specifically at line 416 the code is now:

```
#include <time.h>
    #if defined(HAVE_SYS_TIME_H) || defined(WOLF_C99)
        #include <sys/time.h>
    #endif
```

For now I recommend patching your sources to include both time.h and sys/time.h as you are now.

Let me know if you have any questions or issues.

Thanks,
David Garske, wolfSSL

kaleb-himes, 2018-04-25:
Hi jkortge,

The changes supporting this scenario have been merged. Could you let us know when you get a chance to test and ensure they work in your environment? Once we have confirmation from you that the changes have resolved the issue we can consider this item closed.

Thanks!

- K

kaleb-himes, 2018-05-24:
@jkortge,

We never heard back if you were able to test these changes. I'll close the issue for now but please re-open if issue persists.

Warm Regards,

Kaleb

closed 2018-05-24 by kaleb-himes