wolfSSL/wolfssl #1487: Compiling for Arduino Leonardo

flacidsnake, 2018-04-06:
Hi,

I know this is closed, but I'm having similar problems as @AnthonyDeroche at #539 

After doing what he posted, I'm having these compilation problems:

arduino-1.8.5\libraries\wolfssl/wolfssl/wolfcrypt/wc_port.h:469:31: error: expected unqualified-id before 'const'

     extern struct tm* XGMTIME(const time_t* timer, struct tm* tmp);

                               ^

E:\arduino-1.8.5\libraries\wolfssl/wolfssl/wolfcrypt/wc_port.h:469:31: error: expected ')' before 'const'

E:\arduino-1.8.5\libraries\wolfssl/wolfssl/wolfcrypt/wc_port.h:469:31: error: expected initializer before 'const'

exit status 1
Error compilando para la tarjeta Arduino Leonardo.

Files I think are involved are pasted below. Please, any clue?
Thanks in advance

**_My file settings.h_**
`
/* settings.h
 *
 * Copyright (C) 2006-2017 wolfSSL Inc.
 *
 * This file is part of wolfSSL.
 *
 * wolfSSL is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * wolfSSL is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1335, USA
 */

/* Place OS specific preprocessor flags, defines, includes here, will be
   included into every file because types.h includes it */
#define XTIME fnSecondsSinceEpoch
#define XGMTIME

#ifndef WOLF_CRYPT_SETTINGS_H
#define WOLF_CRYPT_SETTINGS_H

#ifdef __cplusplus
    extern "C" {
#endif

/* Uncomment next line if using IPHONE */
/* #define IPHONE */

/* Uncomment next line if using ThreadX */
/* #define THREADX */

/* Uncomment next line if using Micrium uC/OS-III */
/* #define MICRIUM */

/* Uncomment next line if using Mbed */
/* #define MBED */

/* Uncomment next line if using Microchip PIC32 ethernet starter kit */
[... 2053 lines cut ...]
#if defined(USER_TIME)
    struct tm* gmtime(const time_t* timer);
    extern time_t XTIME(time_t * timer);

    #ifdef STACK_TRAP
        /* for stack trap tracking, don't call os gmtime on OS X/linux,
           uses a lot of stack spce */
        extern time_t time(time_t * timer);
        #define XTIME(tl)  time((tl))
    #endif /* STACK_TRAP */

#elif defined(TIME_OVERRIDES)
    extern time_t XTIME(time_t * timer);
    extern struct tm* XGMTIME(const time_t* timer, struct tm* tmp);
#elif defined(WOLFSSL_GMTIME)
    struct tm* gmtime(const time_t* timer);
#endif
#endif /* NO_ASN_TIME */

#ifdef __cplusplus
    }  /* extern "C" */
#endif

#endif /* WOLF_CRYPT_PORT_H */
`

kaleb-himes, 2018-04-06:
Hi @flacidsnake,

Could you try adding the define NO_ASN_TIME to your settings and let me know if that resolved the GMTIME issues?

- KH

kaleb-himes, 2018-04-06:
Hi @flacidsnake,

I also wanted to let you know that define will disable all time checks on certificates so it should ONLY be used for testing purposes. Once you have everything working in a test environment I would highly recommend adding a solution for XGMTIME before putting anything out in the wild.

This can impact security in that you could end up trusting expired or bad certificates when connecting to peers.

- KH

flacidsnake, 2018-04-07:
Hi @kaleb-himes 

First of all. Thank you very much for your fast reply.

Indeed, defining NO_ASN_TIME at settings.h an uncommenting it at wc_port.h solved the issue and I could finally compile the example wolfssl_client.ino.

But, when compilation ends, I get this message telling the sketch is too big for Leonardo:

El Sketch usa 131036 bytes (457%) del espacio de almacenamiento de programa. El máximo es 28672 bytes.
Las variables Globales usan 11352 bytes (443%) de la memoria dinámica, dejando -8792 bytes para las variables locales. El máximo es 2560 bytes.

Programa muy grande: visite http://www.arduino.cc/en/Guide/Troubleshooting#size para ver cómo reducirlo.
Error compilando para la tarjeta Arduino Leonardo.

Basically, the example sketch is too big. Any idea on how to make it work?

On the other hand, what I want to do with wolf_ssl is:
I just want to receive HTTPS requests and parse the JSON inside them to know if the Leonardo's LED should be ON or OFF.

How can I achieve this? Maybe disabling some cipher options to reduce the size of the sketch?

Thanks in advance!
Regards.

kaleb-himes, 2018-04-09:
@flacidsnake,

Glad to hear that got you past the build issues. To customize the build see in that same "settings.h" header where we check ```#ifdef WOLFSSL_USER_SETTINGS``` and then include the header ```user_settings.h``` if that is defined?

The easiest way to track changes is to add the file ```user_settings.h``` to the library root for Arduino and then define WOLFSSL_USER_SETTINGS in settings.h so your new header gets pulled in.

Then [chapter 2](https://www.wolfssl.com/docs/wolfssl-manual/ch2/) of our manual has a section "2.4.1 REMOVING FEATURES" which covers the features that can be disabled to reduce build size.

If your device really only has 28k available, the smallest build of wolfSSL to date is 23k and that's with the lean-psk option. It's pretty uncommon to come across servers in the wild that support PSK. 

If you configure wolfSSL to use ECC cipher suites at a minimum you'll need roughly 60K of flash available and to use RSA cipher suites you'll need closer to 80K of flash. With such a limited flash space I am not sure you will be able to get SSL/TLS running on your device. Do you have an alternate platform you can use with more flash space?

- KH

kaleb-himes, 2018-04-30:
@flacidsnake,

I never heard back from you on this item, is your device really limited to just 28K of flash? If so were you able to evaluate using a device with more available flash?

Warm Regards,

K

closed 2018-05-24 by kaleb-himes