wolfSSL/wolfssl #1489: Make check - Fail: testsuite/testsuite.test (Alpine Linux - Musl Libc)

TBK, 2018-04-09:
```
======================================
   wolfssl 3.14.0: ./test-suite.log
======================================

# TOTAL: 7
# PASS:  6
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: testsuite/testsuite
=========================

error    test passed!
base64   test passed!
asn      test passed!
MD5      test passed!
SHA      test passed!
SHA-224  test passed!
SHA-256  test passed!
SHA-384  test passed!
SHA-512  test passed!
SHA-3    test passed!
Hash     test passed!
HMAC-MD5 test passed!
HMAC-SHA test passed!
HMAC-SHA224 test passed!
HMAC-SHA256 test passed!
HMAC-SHA384 test passed!
HMAC-SHA512 test passed!
HMAC-SHA3   test passed!
GMAC     test passed!
Chacha   test passed!
POLY1305 test passed!
ChaCha20-Poly1305 AEAD test passed!
AES      test passed!
AES192   test passed!
AES256   test passed!
AES-GCM  test passed!
RANDOM   test passed!
RSA      test passed!
DH       test passed!
PWDBASED test passed!
OPENSSL  test passed!
OPENSSL (EVP MD) passed!
OPENSSL (PKEY0) passed!
OPENSSL (PKEY1) passed!
OPENSSL (EVP Sign/Verify) passed!
ECC      test passed!
logging  test passed!
mutex    test failed!
 error = -8003
FAIL testsuite/testsuite.test (exit status: 255)
```

I have attached the Alpine Linux build file - [APKBUILD.zip](https://github.com/wolfSSL/wolfssl/files/1892325/APKBUILD.zip)

kaleb-himes, 2018-04-11:
Hi @TBK,

That error code (-8003) is a check to make sure that the mutex can not be free'd while it is locked. You'll notice in the test case we do:

```
init mutex, should return 0 success
lock mutex, should return 0 success
free mutex, should return BAD_MUTEX_E (-106)
unlock mutex, should return 0 success
free mutex, NOW it should return 0 success
```

I threw a few print statements in wolfcrypt/test/test.c and can confirm what you are seeing on alpine linux.

```
{ // begin local scope
    int ret;
    ret = wc_FreeMutex(&m);
    printf("Expecting a return of BAD_MUTEX_E which would be %d\n", BAD_MUTEX_E);
    printf("Instead seeing %d\n", ret);
    if (ret != BAD_MUTEX_E)
        return -8003;
} // end local scope, pop ret
```

Here is my system info:

```
khimes-alpn:~/wolfssl# cat /etc/alpine-release
3.7.0
khimes-alpn:~/wolfssl# name -r
4.9.65-1-virthardened
khimes-alpn:~/wolfssl# 
```

Have you contacted alpine Linux dev team about this issue? Just a quick google search turns up a lot of bugs and error reports with the key words "pthreads" and "Alpine Linux". Looks like there are some known issues there.

If you don't need threading you can do the following to resolve this issue:

```
khimes-alpn:~/wolfssl# ./configure --enable-singlethreaded
khimes-alpn:~/wolfssl# make check
```

kaleb-himes, 2018-04-25:
Hi @TBK,

I left this open in case you had any followup commentary but as two weeks has passed with no update I will go ahead and close it for now. If anything else comes up please open a new issue anytime or feel free to shoot us an email at support@wolfssl.com.

Thanks for contacting us about this issue, let us know if you find anything out from the alpine linux team.

Warm Regards,

- K

closed 2018-04-25 by kaleb-himes