wolfSSL/wolfssl #1506: undefined reference to 'wc_Des3_SetKey'

daichunkai, 2018-04-18:
hello, I am a fresh man of wolfssl.
When I use wolfssl to build a application with des3, I have include "des3.h" and link the library `libwolfssl.so` in `/usr/local/lib`,it also has error:
```
/home/dck/cpp-ethereum/ethvm/main.cpp:168: error: undefined reference to 'wc_PBKDF2'
/home/dck/cpp-ethereum/ethvm/main.cpp:238: error: undefined reference to 'wc_Des3_SetKey'
/home/dck/cpp-ethereum/ethvm/main.cpp:246: error: undefined reference to 'wc_Des3_CbcEncrypt'
```
it seems like I miss a linked library, so,what should I do to solve this problem, is there any documents about  compiling the code?
Thank you very much!

dgarske, 2018-04-18:
Hi daichukai,

When configuring you'll need to include `./configure --enable-des3 --enable-pwdbased && make`. This will enable the DES3/DES and PWDBASED functionality. Those features are disabled by default.

Thanks,
David Garske, wolfSSL

daichunkai, 2018-04-19:
@dgarske  yes, it works! thank you very much!

closed 2018-04-19 by daichunkai