wolfSSL/wolfssl #1529: wrong subjectAltName but right CN

bagder, 2018-04-30:
Hey WolfSSLers,

I'm working on curl and since we can build curl with WolfSSL as a backend, I've tasked myself to [add a curl CI build using WolfSSL](https://github.com/curl/curl/pull/2528).

This task not only highlight some weaknesses in our wolfssl integration, but it also points to a weirdness in WolfSSL: `CyaSSL_check_domain_name` does not seem to properly reject certificates with wrong subjectAltName but right CN. The SAN should according to all spec be used if present so in this case, the verification *should* fail but it doesn't.

This is easily reproduced using curl test 311 built to use WolfSSL 3.14.0. The CI build fail we get on travis looks as mundane as this: https://travis-ci.org/curl/curl/jobs/370979870#L6362

Thoughts?

kaleb-himes, 2018-04-30:
Hi @bagder,

Thanks for contacting us with regards to this item. Can you point me to the test-case source code that was running at the time? I'm curious if check_domain_name was invoked prior to or post call to wolfSSL_connect.

Warm Regards,

K

bagder, 2018-04-30:
It is called before. Here's the code:

https://github.com/curl/curl/blob/master/lib/vtls/cyassl.c#L482-L489

kaleb-himes, 2018-04-30:
Thanks! Reviewing now to see what might be the cause of the domain name check not properly failing.

K

kaleb-himes, 2018-04-30:
Ok, so in review I see the step to either verify the peer or not is conditional on the configuration:

https://github.com/curl/curl/blob/master/lib/vtls/cyassl.c#L340-L343

The config switch being:
https://github.com/curl/curl/blob/master/lib/urldata.h#L205

And checked by this macro-mapped function:
https://github.com/curl/curl/blob/master/lib/vtls/vtls.h#L135-L136

I tried to read through the job configuration but I am unfamiliar with the travis setup. With your understanding of the travis job configuration can you tell me if this configure setup is turning on the verifyPeer option noted above? https://travis-ci.org/curl/curl/jobs/370979870/config

kaleb-himes, 2018-04-30:
> Ok, so in review I see the step to either verify the peer or not is conditional on the configuration:
>
> https://github.com/curl/curl/blob/master/lib/vtls/cyassl.c#L340-L343
>
> The config switch being: [...]

The reason I bring this up is due to the way our internals work. Even though the comment states the result is checked later on in the cyassl.c file, our internals will return a success on the mismatch if verifyNone is set. See these lines from wolfssl/src/internal.c

https://github.com/wolfSSL/wolfssl/blob/master/src/internal.c#L8996-L9009

kaleb-himes, 2018-04-30:
@bagder,

>  /* SSL always tries to verify the peer, this only says whether it should
>   * fail to connect if the verification fails, or if it should continue
>   * anyway. In the latter case the result of the verification is checked with
>   * SSL_get_verify_result() below. */

https://github.com/curl/curl/blob/master/lib/vtls/cyassl.c#L340-L343

Just to be clear on the comment noted above, this is not how our library works, if you pass in VERIFY_NONE unlike with OpenSSL you will effectively disable all checks. My recommendation here would be to remove the option to pass VERIFY_NONE if you expect checks to take place.

Unlike OpenSSL and other similar SSL implementations, wolfSSL does not offload the responsibility of checking/verifying things themselves, we either do it all or we do nothing if the user asks us not to by setting VERIFY_NONE. We try to make it easier for the end user and reduce potential for implementation mistakes this way!

bagder, 2018-04-30:
No, the problematic case here is when it sends `SSL_VERIFY_PEER` in the argument (we do not disable the certificate checks for this test) and WolfSLL does its check as requested, *but* it thinks the certificate is fine and valid while in fact it is not. In this test, the subjectAltName field of the certificate is carefully crafted to fail to match while the CN field matches the hostname, and from where I sit that looks like a WolfSSL bug. Ie, WolfSSL should return an error for this certificate but it doesn't.

Alternatively, and I'm certainly not ruling that out. we're not using WolfSSL correctly.

kaleb-himes, 2018-05-01:
Daniel,

Can you provide the test certificate so we can reproduce this for testing and implementation of a fix if it is the case there is a bug? Also since it is a test cert would it be possible to also share the associated test private key for testing in an SSL/TLS scenario? If the private key is being used in any way in an end products though please do not send publicly as we do not want in-use private keys posted publicly.

KH

bagder, 2018-05-01:
The exact pem file used for the test server is here: https://github.com/curl/curl/blob/master/tests/certs/Server-localhost0h-sv.pem and in that directory you'll find the key and other related files.

If you build curl from the git source with wolfssl, you can run this specific test with:
```
$ configure --with-wolfssl --without-ssl
$ make
$ cd tests
$ ./runtests 311
```

(and it'll fail the test since it returns 52 instead of 51...)

kaleb-himes, 2018-05-01:
Perfect! Thank you for providing steps to reproduce this, I’ll get you an update as soon as possible.

kaleb-himes, 2018-05-01:
Daniel,

Can you tell me if this is the expected response for the noted test case:

```
 ./runtests.pl 311
********* System characteristics ******** 
* curl 7.60.0-DEV (x86_64-apple-darwin17.4.0) 
* libcurl/7.60.0-DEV wolfSSL/3.14.0 zlib/1.2.11 libidn2/2.0.4
* Features: AsynchDNS IDN IPv6 Largefile SSL libz UnixSockets 
* Host: kalebs-MBP
* System: Darwin kalebs-MBP 17.4.0 Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64 x86_64
* Servers: SSL HTTP-IPv6 HTTP-unix FTP-IPv6 
* Env: 
***************************************** 
RUN: Unknown server on our https port: 8991 (55)
RUN: HTTPS server failed verification
== Contents of files in the log/ dir after test 311
=== Start of file curlverout.log
 curl 7.60.0-DEV (x86_64-apple-darwin17.4.0) libcurl/7.60.0-DEV wolfSSL/3.14.0 zlib/1.2.11 libidn2/2.0.4
 Release-Date: [unreleased]
 Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtsp smtp smtps telnet tftp 
 Features: AsynchDNS IDN IPv6 Largefile SSL libz UnixSockets 
=== End of file curlverout.log
=== Start of file http_server.log
 13:42:19.459721 Running HTTP IPv4 version on port 8990
 13:42:19.460987 Wrote pid 43636 to .http_server.pid
 13:42:20.497917 ====> Client connect
 13:42:20.498170 accept_connection 3 returned 4
 13:42:20.504049 accept_connection 3 returned 0
 13:42:20.504387 Read 96 bytes
 13:42:20.504517 Process 96 bytes request
 13:42:20.504824 Got request: GET /verifiedserver HTTP/1.1
 13:42:20.505026 Are-we-friendly question received
 13:42:20.505276 Wrote request (96 bytes) input to log/server.input
 13:42:20.505568 Identifying ourselves as friends
 13:42:20.506079 Response sent (56 bytes) and written to log/server.response
 13:42:20.506217 special request received, no persistency
 13:42:20.506327 ====> Client disconnect 0
 13:42:22.581388 select() failed with error: (4) Interrupted system call
 13:42:22.581540 signalled to die
 13:42:22.586017 ========> IPv4 sws (port 8990 pid: 43636) exits with signal (15)
=== End of file http_server.log
=== Start of file http_verify.log
 *   Trying 127.0.0.1...
 * TCP_NODELAY set
 * Connected to 127.0.0.1 (127.0.0.1) port 8990 (#0)
 > GET /verifiedserver HTTP/1.1
 > Host: 127.0.0.1:8990
 > User-Agent: curl/7.60.0-DEV
 > Accept: */*
 > 
 < HTTP/1.1 200 OK
 < Content-Length: 17
 < 
 { [17 bytes data]
 * Connection #0 to host 127.0.0.1 left intact
=== End of file http_verify.log
=== Start of file http_verify.out
 WE ROOLZ: 43636
=== End of file http_verify.out
=== Start of file https_stunnel.log
 2018.05.01 13:42:21 LOG5[ui]: stunnel 5.44 on x86_64-apple-darwin17.2.0 platform
 2018.05.01 13:42:21 LOG5[ui]: Compiled with OpenSSL 1.0.2m  2 Nov 2017
 2018.05.01 13:42:21 LOG5[ui]: Running  with OpenSSL 1.0.2o  27 Mar 2018
 2018.05.01 13:42:21 LOG5[ui]: Threading:PTHREAD Sockets:POLL,IPv6 TLS:ENGINE,FIPS,OCSP,PSK,SNI
 2018.05.01 13:42:21 LOG5[ui]: Reading configuration from file /Users/kalebhimes/curl/tests/https_stunnel.conf
 2018.05.01 13:42:21 LOG5[ui]: UTF-8 byte order mark not detected
 2018.05.01 13:42:21 LOG5[ui]: FIPS mode disabled
 2018.05.01 13:42:21 LOG5[ui]: Failed to get compression methods
 2018.05.01 13:42:21 LOG5[ui]: Configuration successful
 2018.05.01 13:42:22 LOG5[0]: Service [curltest] accepted connection from 127.0.0.1:51273
 2018.05.01 13:42:22 LOG3[0]: SSL_accept: Peer suddenly disconnected
 2018.05.01 13:42:22 LOG5[0]: Connection reset: 0 byte(s) sent to TLS, 0 byte(s) sent to socket
 2018.05.01 13:42:22 LOG5[ui]: Terminated
=== End of file https_stunnel.log
=== Start of file https_verify.log
 *   Trying 127.0.0.1...
 * TCP_NODELAY set
 * Connected to 127.0.0.1 (127.0.0.1) port 8991 (#0)
 * successfully set certificate verify locations:
 *   CAfile: /etc/ssl/cert.pem
   CApath: none
 * CA signer not available for verification, continuing anyway
 * SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
 * SSL write: ASN no signer error to confirm failure, errno 0
 * Failed sending HTTP request
 * Connection #0 to host 127.0.0.1 left intact
=== End of file https_verify.log
=== Start of file server.input
 GET /verifiedserver HTTP/1.1
 Host: 127.0.0.1:8990
 User-Agent: curl/7.60.0-DEV
 Accept: */*
=== End of file server.input
=== Start of file server.response
 HTTP/1.1 200 OK
 Content-Length: 17
 WE ROOLZ: 43636
=== End of file server.response
test 0311 SKIPPED: failed starting HTTPS server (stunnel)
TESTFAIL: No tests were performed
TESTDONE: 1 tests were considered during 8 seconds.
TESTINFO: 1 tests were skipped due to these restraints:
TESTINFO: "failed starting HTTPS server (stunnel)" 1 times (311)
```

kaleb-himes, 2018-05-01:
I think I have identified the issue I just want to confirm the above is the correct or expected output.

bagder, 2018-05-02:
No, it's not the expected output:

> test 0311 SKIPPED: failed starting HTTPS server (stunnel)

I failed to mention that the test needs **stunnel** to be installed, so that it can run that as a front to the "normal" HTTP test server.

kaleb-himes, 2018-05-02:
Hi Daniel,

Thank you for confirming, I suspected as much. Here is why I ask.

A. Test case assumes that there is an option to allow a connection through even though there is an issue and that a specific error code can be checked later after the connection is established.
 
B. There is no such option in wolfSSL, we are very strict in all our checks unlike other SSL implementations. There are legitimate use-cases for invoking SSL_VERIFY_NONE when surrounding architectural security design will provide sufficient countermeasures to allow wolfSSL to skip the verification process. There are only two options:

Option 1: wolfSSL will check everything and reject the connection completely if there is any issue (SSL_VERIFY_PEER).

Option 2: wolfSSL will do no checks at all and will not store or report any issues before, during, or after the connection. Assumption is that the calling application will handle any errors expected to be encountered apart from wolfSSL.

C. This print statement shows that in the reported error case SSL_VERIFY_NONE is set:

```
//Starts at line 336 of curl/lib/vtls/cyassl.c
/* SSL always tries to verify the peer, this only says whether it should       
 * fail to connect if the verification fails, or if it should continue         
  * anyway. In the latter case the result of the verification is checked with   
  * SSL_get_verify_result() below. */                                           
SSL_CTX_set_verify(BACKEND->ctx,                                               
                   SSL_CONN_CONFIG(verifypeer)?SSL_VERIFY_PEER:                
                                               SSL_VERIFY_NONE,                
                   NULL);

if (SSL_CONN_CONFIG(verifypeer) == 0x0) 
    printf("Using SSL_VERIFY_NONE\n");
else
    printf("Using SSL_VERIFY_PEER");

// and just to be sure:
printf("SSL_CONN_CONFIG(verifypeer) %p\n", SSL_CONN_CONFIG(verifypeer));
```

OUTPUT:

```
Using SSL_VERIFY_NONE
SSL_CONN_CONFIG(verifypeer) (nil)

test 0311...[HTTPS wrong subjectAltName but right CN]

curl returned 52, when expecting 51
 exit FAILED

 - abort tests
TESTDONE: 0 tests out of 1 reported OK: 0%
TESTFAIL: These test cases failed: 311 
TESTDONE: 1 tests were considered during 4 seconds.
```

If I modify these lines in curl-root/lib/vtls/cyassl.c:

```
 336   /* SSL always tries to verify the peer, this only says whether it should       
 337    * fail to connect if the verification fails, or if it should continue         
 338    * anyway. In the latter case the result of the verification is checked with   
 339    * SSL_get_verify_result() below. */                                           
 340   SSL_CTX_set_verify(BACKEND->ctx,                                               
 341                      SSL_CONN_CONFIG(verifypeer)?SSL_VERIFY_PEER:                
 342                                                  SSL_VERIFY_PEER,                
 343                      NULL); 
```

 To force SSL_VERIFY_PEER regardless of what ```SSL_CONN_CONFIG(verifypeer)``` returns we get the results I sent you yesterday.

OUTPUT:

```
./runtests.pl 311
********* System characteristics ******** 
* curl 7.60.0-DEV (x86_64-apple-darwin17.4.0) 
* libcurl/7.60.0-DEV wolfSSL/3.14.0 zlib/1.2.11 libidn2/2.0.4
* Features: AsynchDNS IDN IPv6 Largefile SSL libz UnixSockets 
* Host: kalebs-MBP
* System: Darwin kalebs-MBP 17.4.0 Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64 x86_64
* Servers: SSL HTTP-IPv6 HTTP-unix FTP-IPv6 
* Env: 
***************************************** 
Using SSL_VERIFY_NONE
SSL_CONN_CONFIG(verifypeer) (nil)

...

test 0311 SKIPPED: failed starting HTTPS server (stunnel)
TESTFAIL: No tests were performed
TESTDONE: 1 tests were considered during 8 seconds.
TESTINFO: 1 tests were skipped due to these restraints:
TESTINFO: "failed starting HTTPS server (stunnel)" 1 times (311)
```

Note the SSL_VERIFY_NONE is still reporting it will be set but I changed the lines to force SSL_VERIFY_PEER so now the connection is being rejected due to strict checking in wolfSSL.

The test case is not able to get to the ```CyaSSL_check_domain_name``` check because with the certs being used it can not get past the initial connection anyway.

I hope I am conveying this clearly, please let me know if there is any confusion as to how our internal checks work.

Warm Regards,

KH

kaleb-himes, 2018-05-02:
Hi Daniel,

Thank you for confirming, I suspected as much. Here is why I ask.

A. Test case assumes that there is an option to allow a connection to proceed to a given point bypassing certain checks but executing other checks.
 
B. There is no such option in wolfSSL, we are very strict in all our checks or we do not do any checks (SSL_VERIFY_NONE) even if ```CyaSSL_check_domain_name``` is invoked.  unlike other SSL implementations. There are legitimate use-cases for invoking SSL_VERIFY_NONE when surrounding architectural security design will provide sufficient countermeasures to allow wolfSSL to skip the verification process. There are only two options:

Option 1: wolfSSL will check everything and reject the connection completely if there is any issue (SSL_VERIFY_PEER).

Option 2: wolfSSL will do no checks at all and will not store or report any issues before, during, or after the connection. Assumption is that the calling application will handle any errors expected to be encountered apart from wolfSSL.

C. This print statement shows that in the reported error case SSL_VERIFY_NONE is set:

```
//Starts at line 336 of curl/lib/vtls/cyassl.c
/* SSL always tries to verify the peer, this only says whether it should       
 * fail to connect if the verification fails, or if it should continue         
  * anyway. In the latter case the result of the verification is checked with   
  * SSL_get_verify_result() below. */                                           
SSL_CTX_set_verify(BACKEND->ctx,                                               
                   SSL_CONN_CONFIG(verifypeer)?SSL_VERIFY_PEER:                
                                               SSL_VERIFY_NONE,                
                   NULL);

if (SSL_CONN_CONFIG(verifypeer) == 0x0) 
    printf("Using SSL_VERIFY_NONE\n");
else
    printf("Using SSL_VERIFY_PEER");

// and just to be sure:
printf("SSL_CONN_CONFIG(verifypeer) %p\n", SSL_CONN_CONFIG(verifypeer));
```

OUTPUT:

```
Using SSL_VERIFY_NONE
SSL_CONN_CONFIG(verifypeer) (nil)

test 0311...[HTTPS wrong subjectAltName but right CN]

curl returned 52, when expecting 51
 exit FAILED

 - abort tests
TESTDONE: 0 tests out of 1 reported OK: 0%
TESTFAIL: These test cases failed: 311 
TESTDONE: 1 tests were considered during 4 seconds.
```

If I modify these lines in curl-root/lib/vtls/cyassl.c:

```
 336   /* SSL always tries to verify the peer, this only says whether it should       
 337    * fail to connect if the verification fails, or if it should continue         
 338    * anyway. In the latter case the result of the verification is checked with   
 339    * SSL_get_verify_result() below. */                                           
 340   SSL_CTX_set_verify(BACKEND->ctx,                                               
 341                      SSL_CONN_CONFIG(verifypeer)?SSL_VERIFY_PEER:                
 342                                                  SSL_VERIFY_PEER,                
 343                      NULL); 
```

 To force SSL_VERIFY_PEER regardless of what ```SSL_CONN_CONFIG(verifypeer)``` returns we get the results I sent you yesterday.

OUTPUT:

```
./runtests.pl 311
********* System characteristics ******** 
* curl 7.60.0-DEV (x86_64-apple-darwin17.4.0) 
* libcurl/7.60.0-DEV wolfSSL/3.14.0 zlib/1.2.11 libidn2/2.0.4
* Features: AsynchDNS IDN IPv6 Largefile SSL libz UnixSockets 
* Host: kalebs-MBP
* System: Darwin kalebs-MBP 17.4.0 Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64 x86_64
* Servers: SSL HTTP-IPv6 HTTP-unix FTP-IPv6 
* Env: 
***************************************** 
Using SSL_VERIFY_NONE
SSL_CONN_CONFIG(verifypeer) (nil)

...

test 0311 SKIPPED: failed starting HTTPS server (stunnel)
TESTFAIL: No tests were performed
TESTDONE: 1 tests were considered during 8 seconds.
TESTINFO: 1 tests were skipped due to these restraints:
TESTINFO: "failed starting HTTPS server (stunnel)" 1 times (311)
```

Note the SSL_VERIFY_NONE is still reporting it will be set but I changed the lines to force SSL_VERIFY_PEER so now the connection is being rejected due to strict checking in wolfSSL.

The certs used in this test case can not pass a traditional verify in order to get to the ```CyaSSL_check_domain_name``` check but even if they did the connection would still fail just with a different error code being reported from wolfSSL. Right now a -188 error is reported because the root CA loaded can't pass the validation in order to get to the check_domain_name case.

When SSL_VERIFY_NONE is used the ```CyaSSL_check_domain_name``` check will not fire because the calling application asked wolfSSL not to do any checks during the connection.

I hope I am conveying this clearly, please let me know if there is any confusion as to how our internal checks work.

Warm Regards,

KH

kaleb-himes, 2018-05-02:
Daniel,

Turning on debugging in wolfSSL we can see the check_domain_name is not reached due to an earlier failure (Inability to find a CA in verify_locations to validate this cert with):

```
No CA signer to verify with
 Failed to verify Peer's cert
 	No callback override available, fatal
 wolfSSL Leaving ProcessPeerCerts, return -188
 wolfSSL Leaving DoCertificateVerify, return -188
 wolfSSL Leaving DoHandShakeMsgType(), return -188
 wolfSSL Leaving DoHandShakeMsg(), return -188
 wolfSSL error occurred, error = -188
 wolfSSL Entering SSL_get_error
 wolfSSL Leaving SSL_get_error, return -188
 * CA signer not available for verification, continuing anyway
 wolfSSL Entering SSL_get_version
 wolfSSL Entering SSL_CIPHER_get_name
 wolfSSL Entering wolfSSL_get_cipher_name_from_suite
 * SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
 wolfSSL Entering SSL_get_session
 wolfSSL Entering SSL_write()
 handshake not complete, trying to finish
 wolfSSL Entering wolfSSL_negotiate
 wolfSSL Entering SSL_connect()
 ProcessReply retry in error state, not allowed
 wolfSSL error occurred, error = -188
 wolfSSL Leaving wolfSSL_negotiate, return -1
 wolfSSL Leaving SSL_write(), return -1
 wolfSSL Entering SSL_get_error
 wolfSSL Leaving SSL_get_error, return -188
 wolfSSL Entering ERR_error_string
```

So I updated curl-root/lib/vtls/cyassl.com to load the ca to verify that cert:

```
wolfSSL_Debugging_ON();
if (SSL_CTX_load_verify_locations(BACKEND->ctx, "certs/EdelCurlRoot-ca.crt",      NULL) != SSL_SUCCESS)
         printf("Test load of Server-localhost0h-sv.pem failed\n");
```

OK now I am past the verify (-188 error) we were experiencing previously.

I can now reproduce the case you are reporting, when SSL_VERIFY_PEER is set and the correct CA is loaded I can now produce ```curl returned 52, when expecting 51```.

I'll be working to check your initial concerns:

> CyaSSL_check_domain_name does not seem to properly reject certificates with wrong subjectAltName but right CN. The SAN should according to all spec be used if present so in this case, the verification should fail but it doesn't.

bagder, 2018-05-02:
You're clearly having trouble to run the test the way I can, which seems to make you draw the wrong conclusions.

I'm telling you: in test 311 we're calling wolfSSL like this `SSL_CTX_set_verify(SSL_VERIFY_PEER. NULL)` but it will not reject this connection, presumably (my assumption) because it matches the CN field.

We run this exact same test with numerous other TLS libraries and they all detect this error - making the test return error 51 (ssl cert error) instead of error 52 (there's no content returned).

bagder, 2018-05-03:
I don't want to pile on your burden but that sounds like a security flaw and one that we've had all over in the past. Blogged about it in 2009: https://daniel.haxx.se/blog/2009/10/19/null-prefix-domino/

kaleb-himes, 2018-05-03:
Daniel,

Thank you for pointing out the research. We will be reviewing in coming days the fix and security impact in altname/CN comparison. I will keep you informed of the conversation!

Regards,

KH

ejohnstown, 2018-06-27:
I think I have this fixed, beyond having a null in the middle of a name. I have a test certificate with the CN set to "localhost" and the only subject alt name being "garbage", and it is getting rejected.

bagder, 2018-06-28:
I presume this fix is included in the latest release?

ejohnstown, 2018-06-28:
No. This is still a pull request.

ejohnstown, 2018-07-17:
See commit adb3cc5 for the actual fix.

closed 2018-07-17 by ejohnstown