wolfSSL/wolfssl #1530: missing fips.h in tarball on github

bagder, 2018-04-30:
Shouldn't `./wolfssl/wolfcrypt/fips.`h be present in the tarball of the source archive we can download from github? 

In the "official" 3.14.0 download, that file is present and zero bytes:

```
$ ls -l wolfssl/wolfcrypt/fips.
-rw-r--r-- 1 daniel daniel    0 mar  2 22:40 wolfssl/wolfcrypt/fips.h
```

Detected while [adding a CI build for curl with WolfSSL](https://github.com/curl/curl/pull/2528).

I want to download the release archive automatically using curl, and I found no way to do that from the wolfssl.com site.

ejohnstown, 2018-04-30:
That was an issue discovered and fixed with commit a7fe5e3 after the v3.14.0 release. Your solution to touch the file is correct at this time.

As to the file being empty, that is intentional. The FIPS build is licensed separately, and that gives one the missing files.

bagder, 2018-04-30:
Great, thanks!

closed 2018-04-30 by bagder