wolfSSL/wolfssl #1535: Shadow build does not create cyassl/ folder

denravonska, 2018-05-02:
If you try to do a shadow build outside of the source directory it will try to create `cyassl/options.h` without first creating the folder. This causes the configure command to fail:

```
$ ../wolfSSL/configure
...
../wolfSSL/configure: line 32529: cyassl/options.h: No such file or directory
../wolfSSL/configure: line 32529: cyassl/options.h: No such file or directory
../wolfSSL/configure: line 32529: cyassl/options.h: No such file or directory
../wolfSSL/configure: line 32529: cyassl/options.h: No such file or directory
sed: can't read cyassl/options.h: No such file or directory
rm: cannot remove 'cyassl/options.h.bak': No such file or directory
```

If you manually create the folder in your shadow build directory it works as expected:

```
$ mkdir cyassl
$ ../wolfSSL/configure
...
Generating user options header...
option w/o begin -D is -fvisibility=hidden, not saving to wolfssl/options.h
not outputting (N)DEBUG to wolfssl/options.h
option w/o begin -D is -pthread, not saving to wolfssl/options.h
option w/o begin -D is -Wall, not saving to wolfssl/options.h
option w/o begin -D is -Wno-unused, not saving to wolfssl/options.h
option w/o begin -D is -O2, not saving to wolfssl/options.h
option w/o begin -D is -fomit-frame-pointer, not saving to wolfssl/options.h
...

$ ls -la cyassl/options.h 
-rw-rw-r-- 1 marco marco 1929 maj  2 10:47 cyassl/options.h
```

denravonska, 2018-05-03:
This seems to be a duplicate of #1001 which has been closed. The issue remains.

denravonska, 2018-05-04:
Besides build segregation between platforms there probably is no need for shadow builds. I used it with qmake since it was the only sane way to build for different platforms without them affecting each others build process. I have set my build script to create the `cyassl` directory before I launch configure and it works perfectly.

I think the change required is trivial as the `wolfssl` folder is already created in the shadow build directory today.

ejohnstown, 2018-06-25:
Have you tried this recently? In v3.15.3, we updated the configure to handle the cyassl/options.h the same way wolfssl/options.h is handled.

denravonska, 2018-07-06:
@ejohnstown I just tried this with 9bf8122af7c5d9f164ee92ed28fb20530ea57035 and it seems to work as expected now. Thanks!

closed 2018-07-06 by denravonska