wolfSSL/wolfssl #1536: Link error when using user I/O

denravonska, 2018-05-02:
I have an AVR32 platform where I want to enable user I/O in wolfSSL to use in combination with wolfMQTT.  When looking at the configure script it seems like the only way to get to the `WOLFSSL_USER_IO` define is to enable leanpsk which I've done. wolfSSL builds but linking the final binary fails with:

```
mqtt_socket.c:(.text+0x317): undefined reference to `wolfSSL_CTX_SetIORecv'
mqtt_socket.c:(.text+0x327): undefined reference to `wolfSSL_CTX_SetIOSend'
mqtt_socket.c:(.text+0x425): undefined reference to `wolfSSL_SetIOReadCtx'
mqtt_socket.c:(.text+0x431): undefined reference to `wolfSSL_SetIOWriteCtx'
mqtt_socket.c:(.text+0x43f): undefined reference to `wolfSSL_set_using_nonblock'
mqtt_socket.c:(.text+0x498): undefined reference to `wolfSSL_CTX_set_verify'
```

where mqtt_socket.c comes from wolfMQTT. Is there a better/proper way of enabling user I/O functions that I should be using?

dgarske, 2018-05-02:
Hi denravonska,

Thanks for the question and interest in wolfSSL/wolfMQTT!

You do need those API's and they should be available by default if building wolfSSL correctly. In fact `WOLFSSL_LEANPSK` will remove `wolfSSL_set_using_nonblock`. Are you including `src/*.c` files in your build?

If you are building these sources yourself and not using the autoconf ./configure then you should make sure and define the pre-processor macro `WOLFSSL_USER_SETTINGS` and include a user_settings.h file. This file allows you to setup your library configuration in a single place. We have templates such as this one you can use: https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h

Another possibility is the `wolfssl/wolfcrypt/visibility.h` section, which sets the API's that are made visible and hidden. You might try adding the preprocessor macro `BUILDING_WOLFSSL`.

If you are building with ./configure and cross-compiling please let me know the command you are using, so we can look into it.

Thanks,
David Garske, wolfSSL

denravonska, 2018-05-02:
I wanted to avoid `user_settings.h` when possible, so I am using currently using configure using the following options:

```
OPTS = \
   --prefix=$${DESTDIR} \
   --enable-static \
   --enable-leanpsk \
   --enable-smallstack \
   --enable-ecc \
   --disable-shared \
   --disable-examples \
   --disable-fips \
   --disable-selftest \
   --disable-dh \
   --disable-harden \
   --enable-tlsv10 \
   CFLAGS="\"$${QMAKE_CFLAGS}\""
```

I build it the same way for PC and AVR32, so on the PC where I'm currently able to test the CFLAGS variable is just "-pipe -std=gnu99".

I'm wondering if it's just a matter of link order though as libwolfssl.a is linked before libwolfmqtt.a in the Makefile qmake generates.

dgarske, 2018-05-02:
Hi denravonska,

That sounds like an issue with the rename we did in this commit:
https://github.com/wolfSSL/wolfssl/commit/040e0ab752fc080ca9408f5d1d570736acc43b02

The correct function name is `wolfSSL_CTX_SetIORecv`, but we have a define to map the old name as `#define wolfSSL_SetIOSend wolfSSL_CTX_SetIOSend`.

Are you using an older version of wolfSSL that still has the name as `wolfSSL_SetIOSend`? If so you'll need to remap those manually to accommodate the older wolfSSL with the newer wolfMQTT.

David