wolfSSL/wolfssl #1546: CRL and curl's test 313

bagder, 2018-05-05:
Hi again,

I'm still on my task to add a wolfSSL build to curl's CI build setup and I have one final nit left (after #1529 is fixed and I can use a wolfssl version with that patch bundled).

Our [test 313](https://github.com/curl/curl/blob/master/tests/data/test313) provides a CRL file on the curl command line and makes sure it works fine (it should still return an error about the CA cert). I've read up on the wolfSSL API and I'm calling `wolfSSL_LoadCRL` the way I thought I'm supposed to but it doesn't seem to work as I only get `-362`back! (`CRL_MISSING`) but the PEM is file there and seems to be loaded when I try with other TLS backends.

My basic patch to add CRL support to the wolfSSL backend is here right now in a separate dev branch: https://github.com/curl/curl/commit/d4ca1bc2f8313064f0455c93fcd32ccdb7aafa62 (a complete version might need some #ifdefs and what not, but I'm now primarily looking to get this to work with the latest wolfssl version).

Would you be interested in giving me a hand with this? The docs for the CRL functions are a bit sparse so I'm not entirely sure of I'm supposed to do something else/different...

dgarske, 2018-05-11:
Hi bagder,

Thanks for your work on updating the curl wolfSSL support! For the CRL there are a few things I found when testing and reproducing the issues you reported.

1. Loading the CRL file. The function `wolfSSL_LoadCRL` takes a directory as the argument. With the `WOLFSSL_FILETYPE_PEM` option it expects the file to end with `.pem` and be X509 CRL format. If you use `WOLFSSL_FILETYPE_ASN1` it can be named `.crl` or `.der`, but must be binary ASN.1 (DER) format.

2. The CRL file has an expired timestamp for validity. `Next Update: Apr 20 15:07:11 2015 GMT`, which results in the following logs:

```
wolfSSL Entering GetBasicDate
Date AFTER check failed
CRL after date is no longer valid
ParseCRL error
FreeDecodedCRL
CRL file load failed, continuing
...
wolfSSL_connect error -362, CRL missing, not loaded
```

Your changes to support CRL look good otherwise.

If you want there are a few other options you can use for CRL checking:

1. Set a CRL missing callback and use it to load the latest CRL from the URL. The original connect operation will fail with `CRL_MISSING`, then call `wolfSSL_Connect` against.

```
void CRL_CallBack(const char* url) {
    printf("CRL callback url = %s\n", url);
}

wolfSSL_SetCRL_Cb(ssl, CRL_CallBack);
```

2. Enable the build option `HAVE_CRL_IO`. This option will enable an HTTP call to the certificates defined CRL URL, retrieve the CRL fie, load all CRL's and try to validate the certificate against it. The HTTP request is blocking, but a timeout can be set using `wolfIO_SetTimeout(DEFAULT_TIMEOUT_SEC);`. The loaded CRL's will be cached inside the WOLFSSL_CTX, so if you reuse this for future connections against the same server the CRL will still be loaded/available.

Thanks,
David Garske, wolfSSL

ejohnstown, 2018-06-26:
@bagder:

Did David's suggestion help? Do you need any more assistance?

--John

bagder, 2018-06-27:
Thanks and sorry for dropping the ball. The suggestion was awesome and looks exactly what I need. The problem is me. I haven't gotten back to the finishing the task I started. I ended up disabling the two failing curl-wolfssl tests in the curl travis build and then I haven't raised the prio of that job in my TODO list again and thus it still lingers...

I will not feel hurt if you close this issue since it doesn't seem to be a wolfssl problem!

ejohnstown, 2018-06-27:
That's good to hear. So, this issue may be good. Your other issue, in #1529 might be fixed, as well.

closed 2018-06-27 by ejohnstown