wolfSSL/wolfssl #1561: Read on socket with timeout returns WANT_READ instead of TIMEOUT

drbild, 2018-05-16:
A read on blocking socket with a timeout set returns `WANT_READ` instead of `TIMEOUT`. 

This commit e5c04e70a7a1cbdb3f5c4e75b537d93fad3933c2 seems to be the culprit.  Prior to it,
`EmbedReceive` on a blocking socket with a timeout set would return `TIMEOUT`.

Perhaps the `||` should be an `&&`?

dgarske, 2018-05-16:
Hi drbild,

Thanks for the report. I agree, that looks wrong to me. I am assigning this ticket over to John, who is most familiar with the DTLS logic and can address this fix.

John, the offending code in current master is here:
https://github.com/wolfSSL/wolfssl/blob/master/src/wolfio.c#L224

David Garske

dgarske, 2018-05-18:
Hi drbild,

For reference a fix for this was pushed to this PR:
https://github.com/wolfSSL/wolfssl/pull/1566

Thanks,
David Garske, wolfSSL

ejohnstown, 2018-05-23:
@drbild What environment are you using? Do you expect the TIMEOUT to be a fatal error or did you just want to know it timed out? I'd make it a fatal error.

drbild, 2018-05-23:
@ejohnstown This is on OSX and Linux.

Not a fatal error, just to know it timed out, same as a plain socket.  The application might decide to try the read again.

With a plain blocking TCP socket with a timeout set, `recv(sock, ...)` will return `EAGAIN` or `EWOULDBLOCK` (those are `#define`d to the same value on most modern OSes) on timeout.  The same error code returned by a non-blocking socket.

My impression is that most libraries (e.g., OpenSSL) only translate `EAGAIN` to a `WANT_READ` (or `WANT_WRITE`) is the socket is in non-blocking mode.  For blocking sockets`, `EAGAIN` is passed through unchanged or translated to a `TIMEOUT` error.

ejohnstown, 2018-05-23:
@drbild Thanks.

We try to remain agnostic to a socket being blocking or non-blocking, or that you are actually using a socket. (I think) We should return a WANT_READ on a blocked socket timeout. Is this complicating your application logic?

Some of what is in the function `EmbedReceive()`, and the functions `wolfSSL_set_using_nonblock()` and `wolfSSL_get_using_nonblock()`, was there to support DTLS and calling `recvfrom()`. Later we made the `EmbedRecvFrom()` to handle that, but left `EmbedReceive()` alone. At this point I'm going to restore `EmbedRecv()` to how it used to be.

For DTLS we use socket timeouts for timing the handshake retries for blocking sockets with the default `EmbedReceiveFrom()`.